realsenseai/realsense-ros #1229: Not all tests pass in rs2_test.py

timeamagyar, 2020-06-04:
Hello, 

when I execute the tests in rs2_test.py some of the tests fail.
My output:

                      TEST RESULTS                       
---------------------------------------------------------
index test name                      score : message        
----- ----------------------------  ------   --------------
0     accel_up_1                        OK : 
1     non_existent_file                 OK : 'num_channels'
2     align_depth_color_1           FAILED : 'num_channels'
3     align_depth_ir1_1                 OK : 
4     depth_w_cloud_1                   OK : 
5     points_cloud_1                FAILED : 'width'
6     vis_avg_2                         OK : 
7     depth_avg_1                       OK : 
8     depth_avg_decimation_1            OK : 
9     align_depth_ir1_decimation_1      OK : 

exit (-1)

I have the following system settings/parameters:

rosdistro:melodic
rosversion: 1.14.3
python: 2.7
Realsense ROS Wrapper 2.2.13
Realsense SDK 2.33.1
Camera: D435
D435 Firmware 5.12.3.0

I have downloaded the test files as suggested in .travis.yml:

  - bag_filename="http://realsense-hw-public.s3.amazonaws.com/rs-tests/TestData/outdoors_1color.bag";
  - wget $bag_filename -P "records/"
  - bag_filename="http://realsense-hw-public.s3-eu-west-1.amazonaws.com/rs-tests/D435i_Depth_and_IMU_Stands_still.bag";
  - wget $bag_filename -P "records/"

Thanks,

Timea

MartyG-RealSense, 2020-06-04:
When you say that you downloaded the test files, do you mean that you downloaded them to your computer please?  I have not personally used the **rs2_test.py** script that I would think that the full internet paths to the files are provided so that the script downloads them automatically when the script is run if the computer has an internet connection.

I would speculate that you could make the files local by downloading them to the same folder as the rs2_test.py script is in and removing the paths:

**bag_filename="outdoors_1color.bag";
wget $bag_filename -P "records/"
bag_filename="D435i_Depth_and_IMU_Stands_still.bag";
wget $bag_filename -P "records/"**

timeamagyar, 2020-06-04:
Hi,

i have downloaded the files on my local computer and placed them in the exact same folder as the rs2_test.py script.
So that can't be it i guess.

Regards,

Timea

MartyG-RealSense, 2020-06-04:
If the rs2_test.py script file is not edited to remove the paths then it would still be looking for the files on the internet and ignoring the locally stored ones though.

MartyG-RealSense, 2020-06-04:
Does the failure of those two aspects of the test (**align_depth_color_1** and **points_cloud_1**)  negatively affect what you want to do with the camera, please, or does it work fine for you?

timeamagyar, 2020-06-05:
I would put it differently, how can i trust a piece of software where 2 out of 8 unit tests are failing:) What does that say about software quality? I mean we talk about a commercial product  (realsense D400 depth cameras for instance) that ought to rely on the realsense ros wrapper.

MartyG-RealSense, 2020-06-05:
May I confirm please that when you ran the test script, you used the **all** condition:

**python src/realsense/realsense2_camera/scripts/rs2_test.py --all**

timeamagyar, 2020-06-05:
yes this is exactly the way i have launched the script

MartyG-RealSense, 2020-06-05:
It is difficult for me to diagnose what might cause a failed state, as I did not write **rs2_test.py**.  Looking through the source code though, I can see that nearly all of the tests seem to utilise only the '**outdoors_1color**' file, whilst the **D435i_Depth_and_IMU_Stands_still** bag file is just used for the **accel_up_1** IMU test for D435i cameras.

The **depth_w_cloud_1** and **points_cloud_1** both seem to rely on 'pointcloud' being True.  depth_w_cloud_1 passed as OK and points_cloud_1 Failed, suggesting that the test program was able to create the point cloud successfully.

I would recommend judging the stability of the RealSense ROS wrappers by practical tests such as visualising point clouds in Rviz over extended periods.  That is likely to give you a better picture of whether Librealsense's ROS wrapper support meets your commercial grade needs.

MartyG-RealSense, 2020-06-18:
Hi @timeamagyar  Do you still require assistance with this case please?   Thanks!

timeamagyar, 2020-06-18:
From my side we can close the issue. Thanks

closed 2020-06-19 by MartyG-RealSense