realsenseai/realsense-ros #1231: get_device_time_ms() took too long (more then 2 mSecs)

comwise, 2020-06-05:
PARAMETERS 
 * /front/realsense2_camera/accel_fps: 250 
 * /front/realsense2_camera/accel_frame_id: front_accel_frame 
 * /front/realsense2_camera/accel_optical_frame_id: front_accel_optic... 
 * /front/realsense2_camera/align_depth: False 
 * /front/realsense2_camera/aligned_depth_to_color_frame_id: front_aligned_dep... 
 * /front/realsense2_camera/aligned_depth_to_fisheye1_frame_id: front_aligned_dep... 
 * /front/realsense2_camera/aligned_depth_to_fisheye2_frame_id: front_aligned_dep... 
 * /front/realsense2_camera/aligned_depth_to_fisheye_frame_id: front_aligned_dep... 
 * /front/realsense2_camera/aligned_depth_to_infra1_frame_id: front_aligned_dep... 
 * /front/realsense2_camera/aligned_depth_to_infra2_frame_id: front_aligned_dep... 
 * /front/realsense2_camera/allow_no_texture_points: False 
 * /front/realsense2_camera/base_frame_id: base_footprint 
 * /front/realsense2_camera/calib_odom_file:  
 * /front/realsense2_camera/clip_distance: -2.0 
 * /front/realsense2_camera/color_fps: 30 
 * /front/realsense2_camera/color_frame_id: front_color_frame 
 * /front/realsense2_camera/color_height: 720 
 * /front/realsense2_camera/color_optical_frame_id: front_color_optic... 
 * /front/realsense2_camera/color_width: 1280 
 * /front/realsense2_camera/depth_fps: 90 
 * /front/realsense2_camera/depth_frame_id: front_depth_frame 
 * /front/realsense2_camera/depth_height: 240 
 * /front/realsense2_camera/depth_optical_frame_id: camera_link3 
 * /front/realsense2_camera/depth_width: 424 
 * /front/realsense2_camera/device_type:  
 * /front/realsense2_camera/enable_accel: False 
 * /front/realsense2_camera/enable_color: True 
 * /front/realsense2_camera/enable_depth: True 
 * /front/realsense2_camera/enable_fisheye1: True 
 * /front/realsense2_camera/enable_fisheye2: True 
 * /front/realsense2_camera/enable_fisheye: False 
 * /front/realsense2_camera/enable_gyro: False 
 * /front/realsense2_camera/enable_infra1: False 
 * /front/realsense2_camera/enable_infra2: False 
 * /front/realsense2_camera/enable_pointcloud: True 
 * /front/realsense2_camera/enable_sync: False 
 * /front/realsense2_camera/filters:  
 * /front/realsense2_camera/fisheye1_frame_id: front_fisheye1_frame 
 * /front/realsense2_camera/fisheye1_optical_frame_id: front_fisheye1_op... 
 * /front/realsense2_camera/fisheye2_frame_id: front_fisheye2_frame 
 * /front/realsense2_camera/fisheye2_optical_frame_id: front_fisheye2_op... 
 * /front/realsense2_camera/fisheye_fps: 30 
 * /front/realsense2_camera/fisheye_frame_id: front_fisheye_frame 
 * /front/realsense2_camera/fisheye_height: 480 
 * /front/realsense2_camera/fisheye_optical_frame_id: front_fisheye_opt... 
 * /front/realsense2_camera/fisheye_width: 640 
 * /front/realsense2_camera/gyro_fps: 400 
 * /front/realsense2_camera/gyro_frame_id: front_gyro_frame 
 * /front/realsense2_camera/gyro_optical_frame_id: front_gyro_optica... 
 * /front/realsense2_camera/imu_optical_frame_id: front_imu_optical... 
 * /front/realsense2_camera/infra1_frame_id: front_infra1_frame 
 * /front/realsense2_camera/infra1_optical_frame_id: front_infra1_opti... 
 * /front/realsense2_camera/infra2_frame_id: front_infra2_frame 
 * /front/realsense2_camera/infra2_optical_frame_id: front_infra2_opti... 
 * /front/realsense2_camera/infra_fps: 30 
 * /front/realsense2_camera/infra_height: 720 
 * /front/realsense2_camera/infra_width: 1280 
 * /front/realsense2_camera/initial_reset: False 
 * /front/realsense2_camera/json_file_path:  
 * /front/realsense2_camera/linear_accel_cov: 0.01 
 * /front/realsense2_camera/odom_frame_id: front_odom_frame 
 * /front/realsense2_camera/pointcloud_texture_index: 0 
 * /front/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_ANY 
 * /front/realsense2_camera/pose_frame_id: front_pose_frame 
 * /front/realsense2_camera/pose_optical_frame_id: front_pose_optica... 
 * /front/realsense2_camera/publish_odom_tf: True 
 * /front/realsense2_camera/publish_tf: True 
 * /front/realsense2_camera/rosbag_filename:  
 * /front/realsense2_camera/serial_no:  
[... 66 lines cut ...]
 05/06 16:37:40,288 WARNING [139942968239872] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details. 
 05/06 16:37:40,415 ERROR [139943203604224] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs) 
 05/06 16:37:41,723 ERROR [139943203604224] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs) 
[ERROR] [1591346281.124088576]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
 05/06 16:38:21,094 ERROR [139943203604224] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs) 
 05/06 16:38:41,156 ERROR [139943203604224] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs) 
 05/06 16:38:47,208 ERROR [139943203604224] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs) 
 05/06 16:40:19,245 ERROR [139943203604224] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs) 
 05/06 16:40:50,262 ERROR [139943203604224] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs) 
 05/06 16:41:10,273 ERROR [139943203604224] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs) 
 05/06 16:41:12,281 ERROR [139943203604224] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs) 
 [ERROR] [1591346479.413763025]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
 [ERROR] [1591346488.346784044]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
 [ERROR] [1591346556.552866367]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
 [ERROR] [1591346566.551524715]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info

----
get_device_time_ms() took too long (more then 2 mSecs)
 An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
--- 
What could cause this? my sdk lib ? 
ubuntu 16.04   linux kernl 4.15.xx
last version is ok?  RealSense ROS 2.2.8  and SDK 2.26.0
current version is  RealSense ROS 2.2.13 and SDk 2.34.0
can you tell me why?  I'm looking forward to your reply soon

MartyG-RealSense, 2020-06-05:
Looking through your log, the initial errors - **requires patch for fourcc code RW16!** and **UVC metadata payloads not available** - suggest that there is a problem related to patching.  Did you compile Librealsense from source code please, and did you apply a patch script?

comwise, 2020-06-08:
@MartyG-RealSense i just downloaded these packages by "apt-get install", then i install libs to other pc by the following scripts: 
 sudo apt install -y ros-kinetic-cv-bridge \
    ros-kinetic-image-transport \
    ros-kinetic-tf \
    ros-kinetic-diagnostic-updater \
    dkms || { echo -e "${RED}deps install realsense d435i runtime lib failed ${NC}"; exit 7; }
download_rs_pkgs=(
    librealsense2-udev-rules_2.34.0-0~realsense0.2250_amd64.deb
    librealsense2-dkms_1.3.11-0ubuntu1_all.deb
    librealsense2_2.34.0-0~realsense0.2250_amd64.deb
    librealsense2-dev_2.34.0-0~realsense0.2250_amd64.deb
    )
for pkg in ${download_rs_pkgs[@]}; do
    [ -f ${debs_dir}/${pkg} ] || ${download_tool} -P ${debs_dir} ${server}/third_parties/all_debs/${pkg}
    sudo dpkg -E -i ${debs_dir}/${pkg}
done

is there anything wrong?

MartyG-RealSense, 2020-06-08:
@comwise    Installing files in this way is outside of my ROS knowledge unfortunately, so I cannot offer advice about it.  The best way to do a Librealsense installation in Ubuntu is with a fresh Ubuntu install or if that is not convenient, by removing all previous Librealsense installations before doing a new installation.

**Debian package uninstall**

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

**Clean-up when compiling from source code**

sudo make uninstall && make clean && make && sudo make install

MartyG-RealSense, 2020-06-18:
@comwise  Do you still require assistance with this case please?   Thanks!

MartyG-RealSense, 2020-06-25:
Case closed due to no futher comments received.

closed 2020-06-25 by MartyG-RealSense