realsenseai/realsense-ros #1232: Intel D4XX series depth camera filter installation problem

dingbaiyin, 2020-06-08:
In the experimental test, the HWB830 infrared high-transmittance filter is selected, and the filter is 90% high-transparent in the 850 nm band. The filter is installed in front of the two cameras without high-transparency glass, and the filter is not installed in front of the Projector, the test effect is very good. Now the company wants to integrate into the product and encounter several problems
1. Can filters be added in front of Projector
2. Is it possible to use high-transparent glass with filters in front of the camera?

MartyG-RealSense, 2020-06-08:
1.  Section 4.2 of Intel's paper about optical filters discusses the advantages and disadvantages of how adding a filter to the projector can affect the camera's performance.

For example, a long-pass filter on the projector may let IR light pass through but attenuate some of the visible light and so make the image darker.  Ambient light can also help the camera to see over a greater distance range.  An advantage though of a long-pass filter is that it may make distant white walls more visible to the camera.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters

2.  The Chief Trechnical Officer of the RealSense Group at Intel (agrunnet) offers advice about choosing filters in the links below:

https://github.com/IntelRealSense/librealsense/issues/2566#issuecomment-431852571

https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446793681

dingbaiyin, 2020-06-08:
I am surprised to receive your reply so soon. Thank you very much.

dingbaiyin, 2020-06-09:
Thank you very much for the continuous and efficient technical support provided by the Intel team
I see that many users have encountered the same or similar problems as me. At present, I am trying to use long pass filters to solve or reduce the effect of light on depth images. The current test has already played a certain effect. The main scene I tested was matt, light and daylight and used a variety of filters for comparison. The final test selected the HWB infrared high-transmission filter and conducted a second test to test several specifications of the HWB series for comparison. Let me share my test effect picture below: I use Intel D430 camera
[IntelD430深度摄像头滤光片测试完整版.zip](https://github.com/IntelRealSense/realsense-ros/files/4749119/IntelD430.zip)
[北京永兴感知4款HWB红外滤光片测试.zip](https://github.com/IntelRealSense/realsense-ros/files/4749122/4.HWB.zip)
Intel recommends adding filters in front of the two cameras, but because the product is difficult to meet such requirements,we choose to place the entire filter in front of the camera (including the projector), and the experimental results were made during the day and night. as follows
day picture:
[HWB830dayand night test.zip](https://github.com/IntelRealSense/realsense-ros/files/4749958/HWB830dayand.night.test.zip)
I hope it can help you, and I hope I can learn more valuable experience with you

MartyG-RealSense, 2020-06-09:
Thanks so much for your filter test feedback.  I will pass it on personally to Intel.

MartyG-RealSense, 2020-06-18:
I will close the case now.  Please feel free to create new issues for further questions.  Thanks!

closed 2020-06-18 by MartyG-RealSense