realsenseai/realsense-ros #1233: SLAM with D435i - No light scenarios

shirbarzel, 2020-06-08:
Do you think this is possible where some of the rooms are without any light source?
Currently, with the given configurations I'm getting "Registration failed" when entering a dark room.
Do you have any recommendations on different configurations where this can work? 
For example, I have tried ICP odometry but it seems that the performance of the illuminated rooms is worst.
Thanks

MartyG-RealSense, 2020-06-08:
If the problem consistently occurs when entering a dark room, you could perhaps try giving the camera some light-source to work with by having the IR Emitter and its dot pattern active if it is not active already.   That will give the camera the ability to analyse depth detail in the absence of an ambient light source.

shirbarzel, 2020-06-09:
> If the problem consistently occurs when entering a dark room, you could perhaps try giving the camera some light-source to work with by having the IR Emitter and its dot pattern projection active if it is not active already. The camera can analyse depth detail from dots cast over the surfaces of the objects in a location in the absence of an ambient light source.
> 
> [image]

1. The IR emitter is active already, but I found that there is an option to increase the laser power, I will try that.
2. I think that the problem is that the rtabmap gives high weight to feature extraction from the rgb image.

MartyG-RealSense, 2020-06-09:
1.  Increasing Laser Power increases the visibility of the dot pattern but may also increase the camera's operating temperature, so be careful that it doesn't become excessive or problems in the image could occur.  

2.  The RTABMAP wiki offers some advice on tuning the odometry:

http://wiki.ros.org/rtabmap_ros/Tutorials/Advanced%20Parameter%20Tuning#Visual_Odometry

shirbarzel, 2020-06-16:
> 1. Increasing Laser Power increases the visibility of the dot pattern but may also increase the camera's operating temperature, so be careful that it doesn't become excessive or problems in the image could occur if the temperature exceeds 42 degrees C.  You also want to avoid the Projector temperature passing 60 degrees C so that the camera's laser safety shut-off is not triggered.
> 2. The RTABMAP wiki offers some advice on tuning the odometry:
> 
> http://wiki.ros.org/rtabmap_ros/Tutorials/Advanced%20Parameter%20Tuning#Visual_Odometry

1. Tried a lot of parameters combination but still having the same problem.
2. Also I got problems on texture-less scenes like white walls where the odometry get lost easily

MartyG-RealSense, 2020-06-16:
Section 4.2.1 of Intel's white-paper guide to optical filters for 400 Series cameras states that "Long-pass filters can help in some indoor scenes.  Specifically, they will help make distant white walls more visible by increasing the relative brightness of the IR projector".

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

shirbarzel, 2020-06-17:
How can I apply these filters with the realsense-ros?

MartyG-RealSense, 2020-06-17:
The filters discussed are not software filters like post-processing ones.  They are physical filters that you purchase and put in front of the camera's lens.

shirbarzel, 2020-06-17:
Thanks, can I purchase these filters from Intel?

MartyG-RealSense, 2020-06-17:
The Chief Technical Officer of the RealSense Group at Intel (agrunnet) recommends a company called Astra Products as a source for purchasing optical filters.

https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446793681

MartyG-RealSense, 2020-07-05:
Case closed due to no further comments received.

closed 2020-07-05 by MartyG-RealSense

shirbarzel, 2020-07-07:
Hi, what about the new LIDAR L515 is it going to perform better in such scenarios?

MartyG-RealSense, 2020-07-07:
@shirbarzel   Can you post your question as a new issue please? Thanks!