realsenseai/realsense-ros #1234: Jetson Nano "Failed to meet update rate!" with D435i

gotros, 2020-06-09:
Hello,
I am new to ROS and I keep getting full cpu load on all cores with following command:

`roslaunch realsense2_camera opensource_tracking.launch`

>  * /rosdistro: melodic
>  * /rosversion: 1.14.5

Firmware of the D435i is 05.12.03.00

rviz is very laggy and it's impossible to record a point cloud. Pointcloud builts up very slow and laggy. Positioning starts floating into nirvana by moving the camera by any speed and the cloudmapping stops. Realsense-viewer works with no errors.

I am using a jetson nano and a D435i under Ubuntu 18.04 as platform. I did not built ros from source. I followed these steps https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

I attached my log. This is kind of the cleanest session I can get by now. Additional errors coming up by killing (STRG+C) roslaunch and restarting. For example the topic "map" is not published and other. The only solution I found is to reconnect the camera to the USB 3 port.

I hope this is the right place to ask for advice.

Cheerz,
gotros

This is the log.. 

`ros@ros-desktop:/opt/ros/melodic/share/realsense2_camera/launch$ roslaunch realsense2_camera opensource_tracking.launch
... logging to /home/ros/.ros/log/b394d6d2-aa86-11ea-80e4-7cb27d196fae/roslaunch-ros-desktop-9006.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ros-desktop:43619/

SUMMARY
========

CLEAR PARAMETERS
 * /ukf_se/

PARAMETERS
 * /ImuFilter/_publish_tf: False
 * /ImuFilter/_world_frame: enu
 * /ImuFilter/use_mag: False
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
[... 690 lines cut ...]
 09/06 21:23:49,807 ERROR [546299687296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 09/06 21:23:49,844 WARNING [546299687296] (types.cpp:49) Out of frame resources!
 09/06 21:23:49,844 ERROR [546299687296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
[ WARN] [1591730629.864103383]: Failed to meet update rate! Took 0.0085166560000000009201
[ WARN] [1591730629.866479679]: Failed to meet update rate! Took 0.00757425400000000075
[ WARN] [1591730629.872932250]: Failed to meet update rate! Took 0.01064385600000000015
 09/06 21:23:49,874 WARNING [546299687296] (types.cpp:49) Out of frame resources!
 09/06 21:23:49,876 ERROR [546299687296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
[ WARN] [1591730629.874990102]: Failed to meet update rate! Took 0.0093995220000000004323
[ WARN] [1591730629.924705845]: Failed to meet update rate! Took 0.0073817040000000002167
[ WARN] [1591730629.927323760]: Failed to meet update rate! Took 0.0067253500000000005984
[ WARN] [1591730629.984183546]: Failed to meet update rate! Took 0.0069384750000000003714
[ WARN] [1591730629.992349640]: Failed to meet update rate! Took 0.011140912000000001078
[ WARN] [1591730629.997410518]: Failed to meet update rate! Took 0.0073834230000000000693
[ INFO] [1591730629.998467205]: rtabmap (2): Rate=1.00s, Limit=0.000s, RTAB-Map=0.7983s, Maps update=0.0161s pub=0.0000s (local map=1, WM=1)
[ WARN] [1591730630.036859034]: Failed to meet update rate! Took 0.0067462120000000005623
[ WARN] [1591730630.041994184]: Failed to meet update rate! Took 0.0084114120000000000554
[ WARN] [1591730630.044759758]: Failed to meet update rate! Took 0.0080632400000000010315
[ WARN] [1591730630.053974882]: Failed to meet update rate! Took 0.0095699750000000013972
[ WARN] [1591730630.055588819]: Failed to meet update rate! Took 0.0078526630000000010451
[ WARN] [1591730630.074208863]: Failed to meet update rate! Took 0.012004420000000000021
[ WARN] [1591730630.077121992]: Failed to meet update rate! Took 0.011615311000000001426
[ WARN] [1591730630.171587446]: Failed to meet update rate! Took 0.012366683000000001577
 09/06 21:23:50,174 WARNING [546299687296] (types.cpp:49) Out of frame resources!
 09/06 21:23:50,175 ERROR [546299687296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!`

MartyG-RealSense, 2020-06-10:
Hi @gotros   The large number of continuously generating errors makes me think that the lag that you are experiencing may be at least partially related to the processing overhead added by ROS having to handle these messages.

The ROS wrapper developer @doronhi has said that **interrupt event received** warnings can be ignored.

https://github.com/IntelRealSense/realsense-ros/issues/1089#issuecomment-597074650

In regard to the other main error of **Failed to meet update rate!**, a search on the page for this term came up with 357 occurrences in the log, and you said that this was the cleanest log that you could achieve.  This adds to the theory that it is occurring at such a regular rate that it is slowing up the processing.

There is very little documentation available on this error.  When ROS does generate this error though, it sometimes provides this advice:

**Try decreasing the rate, limiting sensor output frequency, or limiting the number of sensors.**

I believe that 'sensor output frequency' is referring to the IMU frequency.

**Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21)** can mean that the user has not performed a calibration on the IMU motion module component of their D435i.

https://www.intelrealsense.com/wp-content/uploads/2019/07/Intel_RealSense_Depth_D435i_IMU_Calibration.pdf

gotros, 2020-06-13:
I calibrated the IMU with the python script and there was no change in the update rate message. I will try to reduce the frequency of the IMU.

MartyG-RealSense, 2020-06-18:
Hi @gotros  Do you have an update on this case that you can provide please?   Thanks!

closed 2020-07-05 by MartyG-RealSense