realsenseai/realsense-ros #1235: How to transfer realsense data via ROS to other Device(s)

0x7c1, 2020-06-10:
I want to make the program under
" catkin_ws/src/realsense-ros-3.2.13/realsense2_camera "
directory from the "ros wrapper site" transfer realsense data to other devices.
I've chosen to use ".cpp file" since I've already made the program which uses C++ and realsense camera.
I want to know the relationships of the files inside " realsense2_camera/src " directory
and how and where the executable files appears after catkin_make etc., because I want to remodel that file(s),
if it's able to transfer the realsense data to other Devices to do it.
I think it's related with roscpp_tutorials, rospy_tutorials and beginner_tutorials' Publisher and Subscriber Programs and I was able to make Publisher and Subscriber Programs communicating with different Devices, though I don't know about the theory of why they run. (how and where the executable files appears)
---
Environment

Device OS: ubuntu 18.04
Device HW: Jetson nano
rosdistro: melodic
python: Python 2.7.17
Realsense ROS Wrapper: 2.2.13
Realsense Viewer Version: 2.34.0
Camera: D435
D435 Firmware: 5.12.03.00

MartyG-RealSense, 2020-06-10:
Hi @0x7c1    This topic goes beyond my current level of ROS knowledge, but I researched the subject extensively to try to provide a useful starting point for the discussion so that others may be able to fill in the knowledge gaps.

In the link below, you can find a **class hierarchy** diagram of realsense2_camera:

http://docs.ros.org/kinetic/api/realsense2_camera/html/inherits.html

If you browse around the options tabs on this site then you can search for details of realsense2_camera from other perspectives, such as file, namespace and class lists.

For example, if you bring up the files list with this link:

http://docs.ros.org/kinetic/api/realsense2_camera/html/files.html

Then by clicking on a file name, you can bring up a **dependency graph** for that file.

[image]

0x7c1, 2020-06-12:
I want to get "only" the RGB Camera and Depth Data via ROS and get them to the other Device by "remodeling" the C++ and/or Python program in realsense2_camera directory,
because I think it'll be easier to make those Data Process and Transfer Program.

Is it possible, technically? And is that difficult? How can I make that?

Please advise, as soon as possible.

"I think" realsense camera's default programs are not so easy to remodel.

Best regards.

MartyG-RealSense, 2020-06-12:
Apologies for the delay in responding further, as I was considering your project carefully.

Am I correct in thinking that you want a device to be able to read a topic published by another device?

If so, there is a ROS multiple machine tutorial where a **talker** and a **listener** can be set up.

http://wiki.ros.org/ROS/Tutorials/MultipleMachines

So you could have a single **Master** device that publishes the color and depth topics, and multiple "Listener" devices that can subscribe to those topics.

Potentially you could change which device is the Master publisher, and make the previous Master a Listener.

Alternatively, if file transfer is a requirement of your project, you could consider setting up an ethernet network, with all the camera-equipped devices connected to a master **Host** computer.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

0x7c1, 2020-06-17:
Thank you for your information.
I've been wandering why I couldn't get the D435's Data, but I solved it.
Starting "realsense-viewer" when "roslaunch" the ".launch file" and restarting the "Terminal" SEVERAL TIME can make "default .launch file" launch, so re-writing the "launch file" might be able to move the C++/Python file(s).
And I think your talker & listener idea will realize the system.

closed 2020-06-17 by 0x7c1