realsenseai/realsense-ros #1243: T265 pose drift

bkitchen1015, 2020-06-11:
I am using a T265 and Structure Core to build a map with a handheld device where I also require the pose to manipulate other data.  In general I have all of this working well together, but I have this one lingering issue with the pose drifting upon startup.  I have looked at these issues [t265: pose_frame drifts significantly](https://github.com/IntelRealSense/realsense-ros/issues/793), and [T265 pose drift](https://github.com/IntelRealSense/realsense-ros/issues/828) and both were solved seemingly by updating the librealsense2 version.  I have updated to the latest repo on apt for librealsense2 and pulled the latest realsense-ros.  The roslaunch output shows:
[ INFO] [1591895097.446298449]: RealSense ROS v2.2.13
[ INFO] [1591895097.446368788]: Running with LibRealSense v2.32.1

When starting the pose data show the pitch and y-axis position change significantly over about 30 seconds before settling down.  After that everything seems to work fine, but that initial drift causes everything else to be misaligned. 
[image]
[image]

bkitchen1015, 2020-06-11:
I forgot to add that I also looked at rs-pose and similar to the two previous issues it seems to give better answers with no drift.  I am not sure how to ask rs-pose for its versions, but when cat'ing the binary I do see "librealsense2.so.32" so it does appear to be using the same version as what I am using in ROS.

RealSenseSupport, 2021-03-18:
Thank you for highlighting this issue. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265. There are some known drift that happens with the product. It is worth reviewing the specification update as outlined on the following page: https://dev.intelrealsense.com/docs/tracking-camera-t265

Also to add, the T265 requires some initial movement in order for the system to begin tracking features.

closed 2021-04-06 by RealSenseSupport