realsenseai/realsense-ros #1244: the test bag

yangqun-yangrong, 2020-06-12:
Can you share a bag with  D435i camera data for slam about https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i?
Thanks.

MartyG-RealSense, 2020-06-12:
The RealSense sample data page has a zip-file download of two D435i bag files ('walking' and 'walk around') that contain IMU motion module data.  Would this meet your needs please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

yangqun-yangrong, 2020-06-12:
Thanks for your reply.But I want to test the result of indoor D435i data.
Can you help me ?

MartyG-RealSense, 2020-06-12:
I did a careful research for available indoor D435i bag files that could be downloaded but could not find any.  I am sorry.

yangqun-yangrong, 2020-06-12:
OK. Thank you very much.

MartyG-RealSense, 2020-06-18:
Case closed due to non-availability of D435i indoor bag file.

closed 2020-06-18 by MartyG-RealSense