realsenseai/realsense-ros #1245: D435i : unable to set manual exposure

echoGee, 2020-06-12:
I am trying to disable autoexposure on D435i camera. But there is no parameter for that when I run `rosrun dynamic_reconfigure dynparam` 

Any reference for this ?

MartyG-RealSense, 2020-06-12:
I could not find definite information on doing this for D435i.  On T265 though it can be done in the launch file with **rosparam**.

https://github.com/IntelRealSense/realsense-ros/issues/1049

If it is possible for D435i, I would speculate (not certain) that an adaptation might look like this:

<rosparam>
  /camera/rgb_camera/enable_auto_exposure: false
  /camera/stereo_module/enable_auto_exposure: false
</rosparam>

MartyG-RealSense, 2020-06-25:
Hi @echoGee   Do you still require assistance on this case please, or can it be closed?   Thanks!

echoGee, 2020-06-26:
Found out a method
`rosrun dynamic_reconfigure dynparam set /camera/rgb_camera exposure 100`

To see what values of parameters are already set, use `rosrun dynamic_reconfigure dynparam get /node` which in an example would be `rosrun dynamic_reconfigure dynparam get /camera/rgb_camera`

closed 2020-06-26 by echoGee

MartyG-RealSense, 2020-06-26:
Great news - thanks for sharing your solution!