realsenseai/realsense-ros #1246: Support for L515 LiDAR camera

caochao39, 2020-06-14:
When I do `roslaunch realsense2_camera rs_camera.launch` with the L515 model, the nodes failed to launch with the error `Unsupported device! Product ID: 0x0B64`. I have to manually change [RS_L515_PID](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/include/constants.h) to `0x0B64` to make it launch. 

With that hack, the nodes launched with another error `An error has occurred during frame callback: Time is out of dual 32-bit range`.

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud` works without errors and publishes the depth image. But there are still a couple of warnings:
1. Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 1, Width: 640, Height: 480, FPS: 30
2. No mathcing profile found for accel with fps=250
3. Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
4. No stream match for pointcloud chosen texture Process - Color

I haven't looked deep into the code. But it seems that the driver is not in full support for the L515 model. The firmware version I am using is `01.04.01.00`

doronhi, 2020-06-14:
The latest librealsense2 version (2.35.2) together with firmware version 01.04.01.00 introduced a new PID for the L515 device.
The plan is to release a new realsense2_camera version to include that new PID later this week.
As for the other warnings you mention:
"Time is out of dual 32-bit range" - as far as I remember this is not a realsense2_camera error. You should look it up.
As for the other issues I'll have to check when I'll have a device at hand. In the meanwhile, regarding the "No matching profile..." issues, do the configurations you ask for, including fps, appear when you run `rs-enumerate-devices`?

caochao39, 2020-06-14:
Great! I'll wait for the update then. For the "no matching profile...", `rs-enumerate-devices` shows that the fps options for Accel are  400, 200, and 100Hz. I think that's why it's complaining about the 250Hz. I will take a deeper look into the other error. Thanks!

nkhedekar, 2020-06-17:
On launching (`roslaunch realsense2_camera rs_camera.launch`), I continuously get the warning on the terminal: `No RealSense devices were found!` even after changing the PID. The device does get enumerated and is able to show the sensor streams in the `realsense-viewer`. Are there any other changes that need to be made from the current version of the development branch to launch the driver with the L515?

Thanks!

caochao39, 2020-06-17:
Having the same problem here.

doronhi, 2020-06-18:
This is fixed in PR #1248.

doronhi, 2020-06-18:
PR #1248 which adds 0x0B64 PID is merged to the new release: version 2.2.14.
It also takes care of the issue you mentioned:
`Given stream configuration is not supported by the device! Stream: Infrared, Stream Index: 1, Width: 640, Height: 480, FPS: 30`
Issue 3. `Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)` is fixed by updating package librealsense2-dkms to version 1.3.12.
Please let me know if the issue 4 still occurs.

Le-Li-Franka, 2020-06-19:
With the newest release v2.2.14, the issue regarding `RS_L515_PID` is resolved. However, there are still some warnings:
1. `Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 640, Height: 480, FPS: 30`.
2. `No mathcing profile found for accel with fps=250` (I guess there is a typo with `matching`) and `Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)` are still there. The version of the package `librealsense2-dkms` on our setup is 1.3.13.
3. `No stream match for pointcloud chosen texture Process - Color` still occurs, and no colored pointcloud is published. Only when we run the command `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY`, the warning doesn't appear and the pointcloud is published. But the pointcloud is still not colored.

nkhedekar, 2020-06-20:
@doronhi Thanks for the update. The L515 does get launched now, however, there doesn't seem to be a pointcloud topic (colored or otherwise)

spgitmonish, 2020-06-23:
I am seeing the same issue as reported by @nkhedekar in this thread. Not able to view the point cloud using Rviz

doronhi, 2020-06-23:
I can't reproduce. I tested L515 with the following command: `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud` and besides the obvious warning "No mathcing profile found for accel with fps=250" I got no message regarding the color stream nor did I see any issue with the pointcloud. Checking with rqt_reconfigure; camera->pointcloud->stream_filter="Process - Color(2)".
I use ubuntu 16.04 with LRS v2.35.2, realsense2_camera version 2.2.14 and L515 with firmware 01.04.01.00 connected though usb 3.2.

Le-Li-Franka, 2020-06-23:
> I can't reproduce. I tested L515 with the following command: `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud` and besides the obvious warning "No mathcing profile found for accel with fps=250" I got no message regarding the color stream nor did I see any issue with the pointcloud. Checking with rqt_reconfigure; camera->pointcloud->stream_filter="Process - Color(2)".
> I use ubuntu 16.04 with LRS v2.35.2, realsense2_camera version 2.2.14 and L515 with firmware 01.04.01.00 connected though usb 3.2.
> @RealSenseCustomerSupport , did you manage to reproduce?

Thanks for updating. We are using ubuntu 18.04 with exactly the same librealsense, realsense_camera and firmware. Could you test it again on a machine with ubuntu 18.04? @doronhi @RealSenseCustomerSupport

nkhedekar, 2020-06-25:
Config: 
Ubuntu: 18.04
librealsense: 2.35.2
firmware: 01.04.01.00
rqt_reconfigure: camera->pointcloud->stream_filter="Process - Color(2)".

The topic `/camera/depth/color/points` exists but nothing is published on it. 

Output of `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud` 

```bash
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
[... 109 lines cut ...]
/camera/infra/image_raw
/camera/infra/image_raw/compressed
/camera/infra/image_raw/compressed/parameter_descriptions
/camera/infra/image_raw/compressed/parameter_updates
/camera/infra/image_raw/compressedDepth
/camera/infra/image_raw/compressedDepth/parameter_descriptions
/camera/infra/image_raw/compressedDepth/parameter_updates
/camera/infra/image_raw/theora
/camera/infra/image_raw/theora/parameter_descriptions
/camera/infra/image_raw/theora/parameter_updates
/camera/l500_depth_sensor/parameter_descriptions
/camera/l500_depth_sensor/parameter_updates
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/pointcloud/parameter_descriptions
/camera/pointcloud/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
```

pauldeee, 2020-06-25:
@nkhedekar according to the output it seems the default settings are not valid for the color stream...

`[ WARN] [1593124727.454867835]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 640, Height: 480, FPS: 30`

Try setting the RGB stream to 1280x720 or 1920x1080.

nkhedekar, 2020-06-25:
@pauldeee Thank you! That was the issue :sweat_smile: 

I was also curious about this line: 

```bash
25/06 15:38:47,623 WARNING [139987242583808] (types.cpp:49) Requested extrinsics are not available!
[ WARN] [1593124727.623920993]: (Gyro, 0) -> (Depth, 0): Requested extrinsics are not available! : using unity as default.
```

Aren't the extrinsics between the gyro and depth frames supposed to be calibrated out of the box?

nkhedekar, 2020-06-25:
There seem to be a few messages at the end of the output:

```bash
25/06 16:26:45,953 WARNING [140254882772736] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.7/0003:8086:0B64.0009/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable
[ INFO] [1593127605.957366407]: SELECTED BASE:Depth, 0
 25/06 16:26:46,068 WARNING [140255123457792] (types.cpp:49) Requested extrinsics are not available!
[ WARN] [1593127606.069341075]: (Gyro, 0) -> (Depth, 0): Requested extrinsics are not available! : using unity as default.
 25/06 16:26:46,069 WARNING [140255123457792] (types.cpp:49) Requested extrinsics are not available!
[ WARN] [1593127606.069574713]: (Accel, 0) -> (Depth, 0): Requested extrinsics are not available! : using unity as default.
```

I'm not sure what the first one is but for the second and third, aren't these supposed to be calibrated out of the box or are the users supposed to calibrate these?

doronhi, 2020-06-28:
@nkhedekar , @pauldeee ,
I'm happy that you managed to work out the color stream as well. I am a little concerned though that 640x480 resolution wasn't available. Could you check it out sung realsense-viewer? Can you check and post the available resolutions as seen in `rs-enumerate-devices` please?
As for the warnings, the "HID set_power 1 failed..." is still under investigation. Theoretically it could mean that the IMU is not working. Do you see anything like that? Do you still have IMU information?
As for the extrinsics for the IMU, they are expected to be available in the upcoming librealsense2 version (2.36.0).

nkhedekar, 2020-06-28:
Here is the output of `rs-enumerate-devices`:

```bash
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	00000000f0140727
    Firmware Version              : 	01.04.01.00
    Recommended Firmware Version  : 	01.04.01.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video4
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	L500
    Asic Serial Number            : 	00000003a8a873dd
    Firmware Update Id            : 	00000003a8a873dd

Stream Profiles supported by L500 Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Infrared	  1024x768	@ 30Hz	   Y8
    Infrared	  768x1024	@ 30Hz	   Y8
    Infrared	  640x480	@ 30Hz	   Y8
    Infrared	  480x640	@ 30Hz	   Y8
    Depth	  1024x768	@ 30Hz	   Z16
    Depth	  768x1024	@ 30Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  480x640	@ 30Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1920x1080	@ 30Hz	   RGB8
    Color	  1920x1080	@ 30Hz	   Y16
    Color	  1920x1080	@ 30Hz	   BGRA8
    Color	  1920x1080	@ 30Hz	   RGBA8
    Color	  1920x1080	@ 30Hz	   BGR8
    Color	  1920x1080	@ 30Hz	   YUYV
    Color	  1920x1080	@ 15Hz	   RGB8
    Color	  1920x1080	@ 15Hz	   Y16
    Color	  1920x1080	@ 15Hz	   BGRA8
    Color	  1920x1080	@ 15Hz	   RGBA8
    Color	  1920x1080	@ 15Hz	   BGR8
    Color	  1920x1080	@ 15Hz	   YUYV
    Color	  1920x1080	@ 6Hz	   RGB8
    Color	  1920x1080	@ 6Hz	   Y16
    Color	  1920x1080	@ 6Hz	   BGRA8
    Color	  1920x1080	@ 6Hz	   RGBA8
    Color	  1920x1080	@ 6Hz	   BGR8
    Color	  1920x1080	@ 6Hz	   YUYV
    Color	  1280x720	@ 60Hz	   RGB8
    Color	  1280x720	@ 60Hz	   Y16
    Color	  1280x720	@ 60Hz	   BGRA8
    Color	  1280x720	@ 60Hz	   RGBA8
    Color	  1280x720	@ 60Hz	   BGR8
    Color	  1280x720	@ 60Hz	   YUYV
    Color	  1280x720	@ 30Hz	   RGB8
    Color	  1280x720	@ 30Hz	   Y16
    Color	  1280x720	@ 30Hz	   BGRA8
    Color	  1280x720	@ 30Hz	   RGBA8
    Color	  1280x720	@ 30Hz	   BGR8
    Color	  1280x720	@ 30Hz	   YUYV
    Color	  1280x720	@ 15Hz	   RGB8
    Color	  1280x720	@ 15Hz	   Y16
    Color	  1280x720	@ 15Hz	   BGRA8
    Color	  1280x720	@ 15Hz	   RGBA8
    Color	  1280x720	@ 15Hz	   BGR8
    Color	  1280x720	@ 15Hz	   YUYV
    Color	  1280x720	@ 6Hz	   RGB8
    Color	  1280x720	@ 6Hz	   Y16
    Color	  1280x720	@ 6Hz	   BGRA8
    Color	  1280x720	@ 6Hz	   RGBA8
    Color	  1280x720	@ 6Hz	   BGR8
    Color	  1280x720	@ 6Hz	   YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 400Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 100Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 100Hz	   MOTION_XYZ32F
```

The output of `rostopic list`:

```bash
/camera/accel/imu_info
/camera/accel/sample
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/gyro/imu_info
/camera/gyro/sample
/camera/infra/camera_info
/camera/infra/image_raw
/camera/infra/image_raw/compressed
/camera/infra/image_raw/compressed/parameter_descriptions
/camera/infra/image_raw/compressed/parameter_updates
/camera/infra/image_raw/compressedDepth
/camera/infra/image_raw/compressedDepth/parameter_descriptions
/camera/infra/image_raw/compressedDepth/parameter_updates
/camera/infra/image_raw/theora
/camera/infra/image_raw/theora/parameter_descriptions
/camera/infra/image_raw/theora/parameter_updates
/camera/l500_depth_sensor/parameter_descriptions
/camera/l500_depth_sensor/parameter_updates
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/pointcloud/parameter_descriptions
/camera/pointcloud/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
```

Both the accelerometer and gyroscope seem to be publishing normally on the topics `/camera/accel/sample` and `/camera/gyro/sample`. Is there any particular reason for them to be published on different topics instead of one single combined message?

nkhedekar, 2020-06-28:
With regards to the  `HID set_power 1 failed...` issue, that seems to be a bit sporadic

doronhi, 2020-06-29:
@nkhedekar , thanks for the info. Looking into it and will keep you posted.

nkhedekar, 2020-06-29:
There also seems to be another largely sporadic bug. The L515 causes the orientation of the desktop to change. I noticed this on two occasions while using the realsense-viewer and today after killing the ros node.

RealSenseSupport, 2020-06-30:
@doronhi FYI we are also unable to reproduce these issues. 640x480 resolution is avail and working as well as pointclouds. We don't see the HID set_power 1 failed error nor the desktop orientation change.

I do however see the...

"No matching profile found for accel with fps=250
profiles found for accel:
fps: 400, format: MOTION_XYZ32F
fps: 200, format: MOTION_XYZ32F
fps: 100, format: MOTION_XYZ32F"

...warning and the "Requested extrinsics are not available!" warning.

Using Ubuntu 18.04 and ROS  Melodic

Thanks

spgitmonish, 2020-07-13:
> @pauldeee Thank you! That was the issue 
> 
> Just to recap: The launch file should be launched with `filters:=pointcloud` and the `color_width` and `color_height` set to an allowed resolution (eg. 1280x720, 1920x1080)

This worked for me I am able to see the output point cloud.

RealSenseSupport, 2020-07-13:
Hi @caochao39 

Will you be needing further help with this? Did the latest librealsense and suggestions above resolve your issue?

Thanks

nkhedekar, 2020-07-13:
I think the major issue of not being able to see the pointcloud output is solved. There are two possible bugs that have come up in this discussion:

> With regards to the `HID set_power 1 failed...` message, that seems to be a bit sporadic

> There also seems to be another largely sporadic bug. The L515 causes the orientation of the desktop to change. I noticed this on two occasions while using the realsense-viewer and today after killing the ros node.

@RealSenseSupport I can open a new issue for each of these since this issue can be marked as solved.

RealSenseSupport, 2020-07-13:
That works @nkhedekar and thank you! It's easier to track em that way too. 

We'll go ahead and close this issue then.

Thanks again.

closed 2020-07-13 by RealSenseSupport