realsenseai/realsense-ros #1247: The distance value is very different from the actual value

dingbaiyin, 2020-06-14:
I encountered a problem when using the Intel D435. The distance between the camera and the target object measured on the depth map is quite different from the actual distance. The error will become smaller after dynamic calibration. What is the difference between Distance and actual distance?
[image]

MartyG-RealSense, 2020-06-14:
Hi @dingbaiyin  The subject of measured distance versus actual distance was covered recently in the link below.

https://github.com/IntelRealSense/librealsense/issues/6562

dingbaiyin, 2020-06-15:
Thank you very much for receiving your reply. I read the Depth RMS Error (mm) curveI.
The actual test distance of the test camera in the previous figure has been dynamically calibrated is 1500mm, but the software test result is 1585.37mm, this difference greatly exceeds Depth RMS Error (mm). I don't know why. In practical applications, we installed the camera at a height of 1700mm from the ground. After the camera was applied for a period of time, we used the PhotoShop software to use 16-bit channels. The RGB mode to view the depth image. The difference between the camera and the object is too large, as shown in Figure 1.
[image]

After the dynamic calibration, the distance information is also checked through the PhotoShop software to show that the actual value is significantly reduced from the measured value. Figure 2
[image]

However, viewing the distance information of each point through PhotoShop is not uniform, and the distance information of each point is different. Sometimes the difference between points is also very large. Through this test, I want to know the possible causes of excessive Depth RMS Error (mm) and the ways to avoid it. At the same time, use PhotShop software to test whether the depth image Depth RMS Error (mm) is appropriate.

MartyG-RealSense, 2020-06-15:
If you are reading depth in Photoshop through a saved image file such as jpeg or raw then you may not get accurate results, because most of the depth information gets lost when saving.  Instead, the **.raw** format should be used in order to retain the depth information in the file.

https://github.com/IntelRealSense/librealsense/issues/2348

dingbaiyin, 2020-06-15:
Thank you.I tried to use the Depth.Quality.Tool software to save the raw format image and then use Photoshop to view the Depth RMS Error (mm) of the image. The result is that the Depth RMS Error value is very large, which is not as accurate as the depth map. And the value of any two points is also very large, I don’t know why? After the camera is used for a period of time, the Depth RMS Error (mm) needs to be calibrated to get better. If there is no external force that causes the camera to deform physically, is it possible that the change in the camera temperature has caused the Depth RMS Error (mm) to deteriorate? Or are there other reasons?

MartyG-RealSense, 2020-06-15:
Is the room that the camera is used in lit all the time or is it sometimes dark?   If there is light all the time then you could try improving RMS by reducing the Laser Power setting to '1' (keeping the projector on but making the dot pattern invisible).  This should also help to avoid the black dots ("laser speckle") that appear on the depth image.

dingbaiyin, 2020-06-15:
The projector of the camera is turn on all the time（laster power 150mw)and camera is install in a sealed box.As shown in Figure 1
![install](https://user-images.githubusercontent.com/40271689/84671264-bdd4a000-af59-11ea-8ff1-39bd0d8b0215.jpg
From these two factors, can it be determined that the camera temperature is too high for a long time, causing the RMS to be too large? If this is the case, I still don’t understand why the higher camera temperature will cause the RMS to increase. At this time, will the internal parameters of the camera change so that it can only be restored by recalibrating the camera?

dingbaiyin, 2020-06-15:
The  projector of the camera is turn on all the time(laster power：150mw) and camera is installed in a sealed box.as the picture shows：
[image]

From these two factors, can it be determined that the camera temperature is too high for a long time, causing the RMS to be too large? If this is the case, I still don’t understand why the higher camera temperature will cause the RMS to increase. At this time, will the internal parameters of the camera change so that it can only be restored by recalibrating the camera?

MartyG-RealSense, 2020-06-15:
Cased USB RealSense cameras have a means of venting heat away from the circuit board inside.  The original generation of RealSense cameras dissipated heat with "fins" on the rear of the case, whilst the modern 400 Series cameras have a conventional ventilation grille.  Without a means of heat dissipation, the heat created by the camera would have nowhere to go inside the casing and it would become like an oven inside.

You can monitor the camera temperature in real-time in the "Controls" section of the RealSense Viewer's options side-panel.  If this is not possible, the current camera temperature can also be read with script code. 

As a simple rule though, if you put your hand on the casing after the camera has been streaming for a few minutes and it is red hot then the temperature inside is too high.

Temperature does not have to become hot enough to mis-calibrate the camera before problems with the image can start to occur.  Once the camera's operating temperature exceeds around 44 degrees then past that level, the chances of the image being disrupted become greater.  This is probably due to the ASIC chip on the camera board not processing as effectively when very hot.

dingbaiyin, 2020-06-15:
Thank you very much for your patience to answer this question, which has always troubled us. I'm glad to know the cause of the problem. I will test the temperature value of the camera ASIC and continue to feedback.

dingbaiyin, 2020-06-15:
Thank you very much for your patience to answer this question, which has always troubled us. I'm glad to know the cause of the problem. I will test the temperature value of the camera ASIC and continue to feedback.
In our current application scenario, the projector cannot be turned off when collecting an image, which will cause a poor depth image effect. We have a function to collect an image in 10 seconds. Can I switch the Projector once when collecting each image? Will this make the ASCI temperature better? Is it desirable? At the same time, frequent switching of the Projector will affect the service life of the camera?

MartyG-RealSense, 2020-06-16:
The laser may degrade over time but the camera would still be able to function, though with a reduction in depth quality.

https://github.com/IntelRealSense/librealsense/issues/3452

If you want to turn the projector on and off, you can either toggle the projector one-off, or turn it off by setting the Laser Power option to a value of '0'.  Example code for both methods is in the link below, in the section headed 'Librealsense2'.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

As the sparseness of the depth image is related to Laser Power (a higher laser power value = less sparseness), the best approach may be to set laser power to maximum before doing the capture and then setting it to '0' for the next 10 seconds.  This should give some time for heat in the case to be drawn away.

I would think that if you are only toggling the projector every 10 seconds instead of continuously, it should not have a significant effect on projector lifespan.

MartyG-RealSense, 2020-07-05:
Hi @dingbaiyin   Do you require further assistance with this case please, or can it be closed?   Thanks!

dingbaiyin, 2020-07-06:
sorry,I did not reply to you in time.Thank you for your professional answer. I will close it.

closed 2020-07-06 by dingbaiyin

MartyG-RealSense, 2020-07-06:
Thanks again!