realsenseai/realsense-ros #1251: Can't get to 60HZ with realsense-ros - D435i and Jetson TX2

shirbarzel, 2020-06-17:
Hi,
I'm trying to set the cameras fps to 60Hz.

I run this:
`roslaunch realsense2_camera rs_camera.launch align_depth:=true unite_imu_method:="linear_interpolation" depth_fps:=60 infra_fps:=60 color_fps:=60`

And the output of: rostopic echo /diagnostics:

`  name: "camera/realsense2_camera_manager_aligned_depth_to_color: Frequency Status"
    message: "Frequency too low."
    hardware_id: "845112071678"
    values: 
      - 
        key: "Events in window"
        value: "152"
      - 
        key: "Events since startup"
        value: "5525"
      - 
        key: "Duration of window (s)"
        value: "5.094850"
      - 
        key: "Actual frequency (Hz)"
        value: "29.834048"
      - 
        key: "Target frequency (Hz)"
        value: "60.000000"
      - 
        key: "Minimum acceptable frequency (Hz)"
        value: "54.000000"
      - 
        key: "Maximum acceptable frequency (Hz)"
        value: "66.000000"
---
header: 
  seq: 400
  stamp: 
    secs: 1592399920
    nsecs: 375811486
  frame_id: ''
status: 
  - 
    level: 1
    name: "camera/realsense2_camera_manager_color: Frequency Status"
    message: "Frequency too low."
    hardware_id: "845112071678"
    values: 
      - 
        key: "Events in window"
        value: "150"
      - 
        key: "Events since startup"
        value: "5125"
      - 
        key: "Duration of window (s)"
        value: "5.022633"
      - 
        key: "Actual frequency (Hz)"
        value: "29.864817"
      - 
        key: "Target frequency (Hz)"
        value: "60.000000"
      - 
        key: "Minimum acceptable frequency (Hz)"
        value: "54.000000"
      - 
        key: "Maximum acceptable frequency (Hz)"
        value: "66.000000"
---
`

I will note that in the realsense-viewer I managed to get to 60 and even 90 fps
Thanks

MartyG-RealSense, 2020-06-17:
Hi @shirbarzel  Frame drops have been known to occur when depth to color alignment is performed because it is processing-intensive.  The RealSense Viewer does not have a depth to color alignment function in 2D mode, which may be why you are getting faster FPS speeds in the Viewer if you are using 2D mode.

As you have an Nvidia Jetson, you should be able to take advantage of CUDA graphics acceleration.  If you used the Debian packages to install Librealsense on your TX2 then CUDA support should already be included.  If you built Librealsense from source code then CUDA support has to be set to True in the CMake build statement to enable CUDA in Librealsense.

shirbarzel, 2020-06-18:
> Hi @shirbarzel I note that you have set **align_depth:=true**. Could you see if you get better performance if you do not include this instruction please?
> 
> As you have an Nvidia Jetson, you should be able to take advantage of CUDA graphics acceleration. If you used packages to install Librealsense on your TX2 then CUDA support should already be included in Librealsense. If you built Librealsense from source code then CUDA support has to be set to True in the CMake build statement to enable CUDA in Librealsense.

1. Yes, actually not including the align_depth option produce this results:
`---
header: 
  seq: 115
  stamp: 
    secs: 1592462490
    nsecs: 652702479
  frame_id: ''
status: 
  - 
    level: 0
    name: "camera/realsense2_camera_manager_depth: Frequency Status"
    message: "Desired frequency met"
    hardware_id: "845112071678"
    values: 
      - 
        key: "Events in window"
        value: "300"
      - 
        key: "Events since startup"
        value: "2824"
      - 
        key: "Duration of window (s)"
        value: "5.007564"
      - 
        key: "Actual frequency (Hz)"
        value: "59.909364"
      - 
        key: "Target frequency (Hz)"
        value: "60.000000"
      - 
        key: "Minimum acceptable frequency (Hz)"
        value: "54.000000"
      - 
        key: "Maximum acceptable frequency (Hz)"
        value: "66.000000"
---
header: 
  seq: 116
  stamp: 
    secs: 1592462490
    nsecs: 832433355
  frame_id: ''
status: 
  - 
    level: 0
    name: "camera/realsense2_camera_manager_color: Frequency Status"
    message: "Desired frequency met"
    hardware_id: "845112071678"
    values: 
      - 
        key: "Events in window"
        value: "300"
      - 
        key: "Events since startup"
        value: "2101"
      - 
        key: "Duration of window (s)"
        value: "5.035990"
      - 
        key: "Actual frequency (Hz)"
        value: "59.571201"
      - 
        key: "Target frequency (Hz)"
        value: "60.000000"
      - 
        key: "Minimum acceptable frequency (Hz)"
        value: "54.000000"
      - 
        key: "Maximum acceptable frequency (Hz)"
        value: "66.000000"
`
But I need to align the depth for the usage or rtabmap odometry which uses both depth and color images.

2. I compiled the librealsense with cuda so I'm taking advantage of the CUDA graphics acceleration.

MartyG-RealSense, 2020-06-18:
FPS rate can be tied to exposure, so it is possible to try to enforce a constant FPS through exposure settings.  On the librealsense SDK there are two approaches to this:

- If auto-exposure is enabled, disable an option called **Auto-Exposure Priority**.

-. If manual exposure is being used, set the exposure value within a certain range.

https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

On the Nvidia forums, a Jetson TX2 user with ROS approached this issue by changing settings with rqt-reconfigure:

https://forums.developer.nvidia.com/t/ros-realsense-d435-on-jetson-tx2-only-15fps-fhd-availlable/62420/7

closed 2020-07-05 by MartyG-RealSense