realsenseai/realsense-ros #1252: Modify the position when using the T265 tracker

stevemartinov, 2020-06-20:
Hi,

Due to the fact that my T265 camera is located infront of my robot and the tracker uses the camera as a center point, I have decided to simply substract the distance (X and Y) from the camera center to my robot center thus obtaining the correct robot position (without modifying my TF).

So my code looks like this (base_realsense_node.cpp):

```cpp
odom_msg.header.stamp = t;
        odom_msg.header.seq = _seq[stream_index];
        odom_msg.pose.pose = pose_msg.pose;
        odom_msg.pose.pose.position.x = pose_msg.pose.position.x + x_to_footprint;
        odom_msg.pose.pose.position.y = pose_msg.pose.position.y + y_to_footprint;
```

Is that okay? And I also want to include the wheel odometry and thus modify correctly the JSON file, will that interfere with my position or it only reads the velocity?

RealSenseSupport, 2021-03-18:
Regarding wheel odometry, you should have a look at the calibration configuration for it:  https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/t265_wheel_odometry

Also this readme is a good outline of setting up for T265 - https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md

stevemartinov, 2021-03-23:
Oh wow, thanks for replying after 1 year. I am not using this shitty product already but thanks

closed 2021-03-23 by stevemartinov