realsenseai/realsense-ros #1253: D435 & T265 Doesn't working suddenly!!

Eslamrady42, 2020-06-21:
I have a problem in D435 & T265 they worked with me perfectly once on my laptop then I shutdown my laptop then tried in the next day but it's not working by this error:

eslam@root:~$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/eslam/.ros/log/83eda49a-b3af-11ea-aac9-64006afb8a06/roslaunch-root-11466.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://root:34765/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 83 lines cut ...]
[ INFO] [1592737736.974401737]: RealSense Node Is Up!
 21/06 13:08:57,038 WARNING [139829572531968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/06 13:08:57,099 WARNING [139829572531968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/06 13:08:57,181 WARNING [139829572531968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/06 13:08:57,264 WARNING [139829572531968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/06 13:08:57,345 WARNING [139829572531968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/06 13:08:57,406 WARNING [139829572531968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ WARN] [1592737737.432580112]: Hardware Notification:USB SCP overflow,1.59274e+12,Error,Hardware Error
 21/06 13:08:57,467 WARNING [139829572531968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/06 13:08:58,123 WARNING [139829538961152] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 23, 
[ WARN] [1592737738.453935939]: Hardware Notification:IR stream start failure,1.59274e+12,Error,Hardware Error
 21/06 13:08:58,943 WARNING [139829538961152] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 12, 
[ WARN] [1592737739.475562008]: Hardware Notification:USB REC overflow,1.59274e+12,Error,Hardware Error
 21/06 13:08:59,505 ERROR [139828750440192] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
[camera/realsense2_camera_manager-2] process has died [pid 11502, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/eslam/.ros/log/83eda49a-b3af-11ea-aac9-64006afb8a06/camera-realsense2_camera_manager-2.log].
log file: /home/eslam/.ros/log/83eda49a-b3af-11ea-aac9-64006afb8a06/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/eslam/.ros/log/83eda49a-b3af-11ea-aac9-64006afb8a06/camera-realsense2_camera-3*.log
^C[rosout-1] killing on exit
[rosout-1] escalating to SIGTERM
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
eslam@root:~$

Eslamrady42, 2020-06-21:
@doronhi  all I want to know what does it mean??!

[camera/realsense2_camera_manager-1] process has died [pid 6305, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/eslam/.ros/log/d76349ea-b3b4-11ea-a4de-64006afb8a06/camera-realsense2_camera_manager-1.log].
log file: /home/eslam/.ros/log/d76349ea-b3b4-11ea-a4de-64006afb8a06/camera-realsense2_camera_manager-1*.log

Alex-Beh, 2020-06-30:
@Eslamrady42 Hello, I am facing the same issue with you. May I know did you solve that?

Eslamrady42, 2020-06-30:
> @Eslamrady42 Hello, I am facing the same issue with you. May I know did you solve that?

Didn’t solve it yet I’m waiting for the solution from @RealSenseCustomerSupport 
@MartyG-RealSense

Alex-Beh, 2020-06-30:
Set `enable_infra2` to false in `rs_rgbd.launch` seem will stop the error. However, I don't know whether this is a proper solution or not.

Eslamrady42, 2020-06-30:
> Set `enable_infra2` to false in `rs_rgbd.launch` seem will stop the error. However, I don't know whether this is a proper solution or not.

It’s the same result nothing changed.

Alex-Beh, 2020-07-01:
Maybe you can try run `realsense-viewer` and update to the latest firmware.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

Eslamrady42, 2020-07-01:
> Maybe you can try run `realsense-viewer` and update to the latest firmware.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

That what I have SDK 2.35.2

Steven-ZY, 2020-12-01:
@Eslamrady42 I have the same question, how do you solve it? Thanks in advance. [Hardware Notification:Depth stream start ]failure,1.60683e+12,Error,Hardware Error