realsenseai/realsense-ros #1254: Pointcloud filter provides 0 z value

Joechencc, 2020-06-22:
I launch the camera with following command:
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud 

But when I transfer the point cloud obtained to x,y,z coordinate. I will always have 0 z value. I wonder why this happens

Sample Output:
[ INFO] [1592848575.149751606]: out_x:-0.102187
[ INFO] [1592848575.150255193]: out_y:413.696472
[ INFO] [1592848575.150657276]: out_z:0.000000

Code Sample:
sensor_msgs::PointCloud2Iterator<double> out_x(*outMsg, "x");
for (size_t i = 0; i < msg->height * msg->width; ++i, ++out_x, ++out_y, ++out_z, ++out_r, ++out_g, ++out_b, ++in_x, ++in_y, ++in_z)
	{
		*out_x = *in_x;
		*out_y = *in_y;
		*out_z = *in_z;
		if (out_z[0] != 0.0){
			ROS_INFO("out_x:%lf",out_x[0]);
			ROS_INFO("out_y:%lf",out_y[0]);
			ROS_INFO("out_z:%lf",out_z[0]);
		}	
 

		// color lasers with the rainbow array
		//int color = *in_ring % N_COLORS;
		//*out_r = (rainbow[color] >> 16) & 0x0000ff;
		//*out_g = (rainbow[color] >> 8) & 0x0000ff;
		//*out_b = rainbow[color] & 0x0000ff;
	}

And fixed frame is set related to the origin:
#setup the fixed frame
rosrun tf static_transform_publisher 0 0 0 0 0 0 1 map my_frame 10
# create the relative framework between camera and fix_frame
rosrun tf static_transform_publisher 0.0 0.0 0.0 0.0 0.0 0.0 camera_depth_optical_frame map 100

MartyG-RealSense, 2020-06-23:
Hi @Joechencc   do you get any depth if you leave out your fixed-frame statement and just do this:

**rosrun tf static_transform_publisher 0 0 0 0 0 0 0 map camera_depth_optical_frame 100**

Joechencc, 2020-06-23:
Yeah. That is what I got before. Is this the 3D point cloud?

[image]

Because once I subscribe the pointcloud2 , and I print out x,y,z coordinates of the pointcloud.  Z= 0

MartyG-RealSense, 2020-06-23:
Yes, that is a 3D point cloud that is displayed in the center panel.

You do not need to use Librealsense SDK programs such as **rs-pointcloud** to obtain the pointcloud2 coordinate.  Librealsense applications and applications running in the RealSense ROS wrapper are separate and do not affect each other.

It looks as though the publishing of pointcloud2 and the Image are fine.  

The link below looks like a good reference in regard to getting XYZ from pointcloud2:

https://answers.ros.org/question/310996/how-to-get-xyz-and-rgb-of-each-pixel-from-a-sensormsg-image/

Joechencc, 2020-06-23:
With the link you provided,  fixed-frame setup has error popped up actually. 

IntelRealSense/librealsense#6451 (comment)

[ERROR] [1592942324.062232575]: Ignoring transform for child_frame_id "camera_depth_optical_frame" from authority "unknown_publisher" because of a nan value in the transform (0.000000 0.000000 0.000000) (nan nan nan nan)

MartyG-RealSense, 2020-06-23:
There was a previous RealSense ROS case where this error occurred.

https://github.com/IntelRealSense/realsense-ros/issues/1153#issuecomment-615204193

Joechencc, 2020-06-23:
Is that the order wrong?

The example said "rosrun tf static_transform_publisher 0 0 0 0 0 0 0 map camera_depth_optical_frame 100" which links the map to camera_depth_optical_frame. However, map is fixed frame, it already has a parent base_link in this case.

Should it be the other way round 
"rosrun tf static_transform_publisher 0.0 0.0 0.0 0.0 0.0 0.0 camera_depth_optical_frame map 100" which the camera linked to the fixed frame.

MartyG-RealSense, 2020-06-23:
The question is somewhat beyond my current ROS knowledge, as I am a ROS generalist rather than a specialist.  I wonder though whether fixed_frame should be set in RViz to **camera_link** instead of map, like in the RealSense ROS documentation's example use of **roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**?

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

Joechencc, 2020-06-24:
Thank you. It helps a lot! I have changed the relative fix frame .  It is kind of working.

MartyG-RealSense, 2020-06-24:
Great news - thanks for the update!

MartyG-RealSense, 2020-07-05:
Hi @Joechencc   Do you require further assistance with this case please, or can it be closed?  Thanks!

closed 2020-07-18 by MartyG-RealSense