realsenseai/realsense-ros #1255: how can I  run rtabmap with AGX GPU

weseu, 2020-06-23:
enviroment : AGX Jetpack 4.3
ros : melodic
camera : D435i

when I run the comment
#roslaunch realsense2_camera opensource_tracking.launch
I check the publishing rate of topic
#rostopic hz /rtabmap/odom
I get about 10hz, and I want to let it be better.  So,I think maybe I can let it work by GPU to let it work be faster, but I don't know how to do....

have any solution for my information ?

MartyG-RealSense, 2020-06-23:
I researched your question carefully, and 10hz seems to be the widely used default value for RTABMAP odometry (less than 10hz is undesirable).  

Some have suggested using PX4 with RTAB to exceed a 10 hz publishing rate.  

https://answers.ros.org/question/317419/realsense-d435-rtabmap-pixhawk-imu-robot_localization/

https://px4.io/

weseu, 2020-06-29:
So...It didn't have cuda code now?

well.....did you know have any tools can let the code become cuda code...

MartyG-RealSense, 2020-06-29:
My understanding is that the odometry speed is affected by the number of features being extracted.  If it is above 10hz then it is suggested that the number of extracted features is increased (implying that being at 10 Hz is desirable).  You could read from this that if increasing extracted features reduces Hz, then reducing the value for extracted features might increase Hz.  This is just personal speculation and not certainty though.

http://wiki.ros.org/rtabmap_ros/Tutorials/Advanced%20Parameter%20Tuning#Reduce_the_Maximum_Features

closed 2020-07-18 by MartyG-RealSense