realsenseai/realsense-ros #1260: Frame Rate fluctuating

Tal-seven, 2020-06-27:
Hey, 

Following are the details of my setup:

```
Device: Intel Realsense D415
Device Fw Version: 05.12.05.00
Realsense ROS: v2.2.14
LibRealSense: v2.33.1

```
output of lsusb -v:

```
Bus 004 Device 002: ID 8086:0ad3 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
  idProduct          0x0ad3 
  bcdDevice           50.c5
  iManufacturer           1 Intel(R) RealSense(TM) Depth Camera 415 
  iProduct                2 Intel(R) RealSense(TM) Depth Camera 415 
  iSerial                 3 928223020142
  bNumConfigurations      1
```
In realsense-viewer i get 30 fps without any throttling or fluctuations. But when i launch realsense2_camera rs_camera.launch. The output of /diagnostics is as below: 

```
header: 
  seq: 54
  stamp: 
    secs: 1593238538
    nsecs: 325000661
  frame_id: ''
status: 
  - 
    level: 1
    name: "camera/realsense2_camera_manager_color: Frequency Status"
    message: "Frequency too low."
    hardware_id: "925322061947"
    values: 
      - 
        key: "Events in window"
        value: "85"
      - 
        key: "Events since startup"
        value: "505"
      - 
        key: "Duration of window (s)"
        value: "5.120828"
      - 
        key: "Actual frequency (Hz)"
        value: "16.598878"
      - 
        key: "Target frequency (Hz)"
        value: "30.000000"
      - 
        key: "Minimum acceptable frequency (Hz)"
        value: "27.000000"
      - 
        key: "Maximum acceptable frequency (Hz)"
        value: "33.000000"
---
header: 
  seq: 55
  stamp: 
    secs: 1593238538
    nsecs: 500725947
  frame_id: ''
status: 
  - 
    level: 0
    name: "camera/realsense2_camera_manager_depth: Frequency Status"
    message: "Desired frequency met"
    hardware_id: "925322061947"
    values: 
      - 
        key: "Events in window"
        value: "154"
      - 
        key: "Events since startup"
        value: "702"
      - 
        key: "Duration of window (s)"
        value: "5.131224"
      - 
        key: "Actual frequency (Hz)"
        value: "30.012335"
      - 
        key: "Target frequency (Hz)"
        value: "30.000000"
      - 
        key: "Minimum acceptable frequency (Hz)"
        value: "27.000000"
      - 
        key: "Maximum acceptable frequency (Hz)"
        value: "33.000000"
---
```

And this continues as long as the node is up. For the first 5-6 seconds the framerate is maintained at 30 fps and then it starts falling; until eventually hitting the 14-15 fps mark. The depth and IR stream framerates are consistent. I've tried changing the camera resolution parameters (color_width, color_height), but this doesnt affect the problem at hand.

MartyG-RealSense, 2020-06-27:
Hi @Tal-seven  There was a similar case recently in which a RealSense ROS user's frame rate was being halved but they did not experience this in the RealSense Viewer.  In that case they were using **align_depth:=true** in their roslaunch statement.  Are you using align_depth in your roslaunch too, please?

https://github.com/IntelRealSense/realsense-ros/issues/1251

Tal-seven, 2020-06-27:
Hey @MartyG-RealSense, yes that was the case. Problem solved. Thanks.

closed 2020-06-27 by Tal-seven

MartyG-RealSense, 2020-06-27:
You are very welcome.   Thanks for the update!