realsenseai/realsense-ros #1262: Get_device_time_ms()took too long (more then 2 mScecs)

Jenny0420, 2020-06-28:
camera : intel realsense D435
ros kinetic + ubuntu 16.04 LTS
my librealsense already done ,when i run "realsense-viewer",i can get the RGB Picture and Depth
**And my realsense- ros already done yet,in rviz i can get the image like this**
[image]

**but when i run "roslaunch realsense2_camera rs_rgbd.launch"
i have got an error,like this:**

[image]

**when i run the "raulmur/ORB_SLAM2",i can't get the image,like this**

[image]
 
**Thanks for any help !!**

MartyG-RealSense, 2020-06-28:
Please make sure you have the **rgbd** package for ROS Kinetic installed by using the instruction below:

**sudo apt install ros-kinetic-rgbd-launch**

Jenny0420, 2020-06-29:
@MartyG-RealSense 
thanks,and i'm sure my '**ros-kinetic-rgbdd-launch**' was  the last vision
when i run '**realsense-viewer**',i got a warn 
[image]
Maybe my two rules caused this problem.

MartyG-RealSense, 2020-06-29:
If you have multiple udev rules then guidance is provided in the link below about which one is safe to remove.

https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

Jenny0420, 2020-06-29:
i  followed the guidence and remove the rule"**60-librealsense2-udev-rules.rules**"
and then in "realsense-viewer" is ok ,no error,no warn.
but when i run "**roslaunch realsense2_camera rs_rgbd.launch**"
i got a new error
[image]

MartyG-RealSense, 2020-06-29:
@Jenny0420   The guidance was that the rule under 60-librealsense2-udev-rules.rules was the one that should **not** have been removed because it breaks Debian.  The one that should have been removed is the one under the **etc** folder.

Jenny0420, 2020-06-29:
oh...sorry .
it means that i need to clean all and rebuild from scratch.

MartyG-RealSense, 2020-06-29:
Before doing that, you could try rebuilding the udv rules with this:

**sudo apt-get install librealsense2-dkms**

Jenny0420, 2020-06-29:
the 'librealsense-dkms' was the last vision
maybe i can use
'**sudo cp config/60-librealsense2-udev-rules.rules /lib/udev/rules.d/**'
and remove the another one ??

Jenny0420, 2020-06-29:
the 'librealsense2-dkms'was the last vision

MartyG-RealSense, 2020-06-29:
Your knowledge of udev rules installation is probably better than mine!  If you are going to have to do a fresh install anyway if it does not work, you may as well try it.

Jenny0420, 2020-06-29:
i have tired and it doesn't work .i try to remove all (realsense SDK , ros ,realsense-ros)
and do a freash install .
thanks

MartyG-RealSense, 2020-06-29:
Good luck!

Jenny0420, 2020-07-01:
@MartyG-RealSense 
i have already rebuild the **realsense SDK2** and **ros kinetic**
And i want to see some image by **rviz** and when i run 
**'roslaunch realsense2_camera rs_camera.launch'**
i got an error,like this:
[image]

this is a new warn i've never met before.

and in rviz when i choose the topic ,i found that the **"depth_registered"** is gone,i can't see the pointcloud2 image
[image]

MartyG-RealSense, 2020-07-01:
Hi @Jenny0420   On Ubuntu, support for frame metadata has to be patched in if you are building from source.

https://dev.intelrealsense.com/docs/frame-metadata#section-os-support

An exception to this is if you build from source using the **backend** method, which requires an internet connection.  Metadata support is then included in the build.  The backend installation method is described in the link below.

https://github.com/IntelRealSense/librealsense/issues/6710#issuecomment-650611442

Looking at your log extract though, it does not seem that an absence of metadata has "killed" the launch process.

The link below describes how to get an RGBD pointcloud with **pointcloud2** and display it in Rviz.

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

MartyG-RealSense, 2020-07-18:
Hi @Jenny0420   Do you still require assistance with this case please, or can it be closed?   Thanks!

closed 2020-07-26 by MartyG-RealSense