realsenseai/realsense-ros #1263: L515 URDF

lauesa, 2020-06-30:
Does Intel have an URDF for the L515 available for use?

MartyG-RealSense, 2020-07-04:
Apologies for the delay in responding.  I checked the URDF page of the latest RealSense ROS wrapper version at the time of writing this (version **2.2.14**) and there is not a URDF xacro file listed for the L515 yet.

https://github.com/IntelRealSense/realsense-ros/tree/development/realsense2_description/urdf

lauesa, 2020-07-06:
That was my finding as well. Are there plans to add a URDF xacro file for the L515 in the near future (2 weeks)?

MartyG-RealSense, 2020-07-06:
I am not involved in development.  As wrapper version 2.2.14 only just came out though, it seems unlikely within that time-scale.

lauesa, 2020-07-06:
Thanks. I'll check back in periodically to see if it has been added.

doronhi, 2020-07-07:
I'm looking into it.

doronhi, 2020-12-20:
Added urdf for L515 in version 3.1.2 for eloquent and in version 2.2.20 for ROS1.

closed 2020-12-20 by doronhi