realsenseai/realsense-ros #1265: D4XX firmware update method

dingbaiyin, 2020-07-06:
I want to programmatically realize the firmware upgrade of D4XX depth camera under Android system and ubuntu system, can it be achieved, how to achieve it?

MartyG-RealSense, 2020-07-06:
The SDK provides functions for interacting with the camera firmware through scripting through interfaces called **update_device** and **updatable**.  The link below discusses how to update firmware with Python scripting (I could not find a C++ version):

https://github.com/IntelRealSense/librealsense/issues/4416

I would speculate (not certain) that you could access the firmware update with scripting on Android by using **native** code to access the update mechanism with C++ (if you could get the update scripting working with C++).  A link to the SDK's native code example program is below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/native_example/readme.md

dingbaiyin, 2020-07-06:
Thank you very much for your reply，I will try it .

MartyG-RealSense, 2020-07-18:
Hi @dingbaiyin   Do you still require assistance with this case, please?   Thanks!

dingbaiyin, 2020-07-21:
No,Thanks. I will ask questions if there are any follow-up questions。

closed 2020-07-21 by dingbaiyin