realsenseai/realsense-ros #1266: D415: control_transfer returned error, index: 768, error: No data available, number: 61

FrankGaoX, 2020-07-06:
Hi, I am new to Linux : )
Device: Jetson Nano
Jetpack 4.4

realsense-viewer has no error. Everything works fine. However. realsense-ros seems not working.
rviz not showing any image or pointcloud. 

Thanks for the help. I have been working on this for 4 days. Still not working.

$ roslaunch realsense2_camera rs_camera.launch

... logging to /home/fox/.ros/log/67e2fcb6-bf56-11ea-86e0-982cbc14a1ea/roslaunch-fox-11760.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fox:37427/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 90 lines cut ...]
[ INFO] [1594018851.651237886]: insert Infrared to Stereo Module
[ INFO] [1594018851.909449488]: SELECTED BASE:Depth, 0
[ INFO] [1594018851.954622920]: RealSense Node Is Up!
 **06/07 03:00:52,034 WARNING [547516039552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/07 03:00:52,095 WARNING [547516039552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/07 03:00:52,199 WARNING [547516039552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/07 03:00:52,259 WARNING [547516039552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61**
 06/07 03:00:52,350 WARNING [547516039552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
 06/07 03:00:56,284 WARNING [546794627456] (messenger-libusb.cpp:105) usb_request_cancel returned error, endpoint: 130 error: Success, number: 0
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

$ roslaunch rtabmap_ros rtabmap.launch     rtabmap_args:="--delete_db_on_start"     depth_topic:=/camera/aligned_depth_to_color/image_raw     rgb_topic:=/camera/color/image_raw     camera_info_topic:=/camera/color/camera_info     approx_sync:=false

[ INFO] [1594018946.599734644]: rtabmap 0.19.3 started...
[ WARN] [1594018950.510592744]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rgbd_odometry subscribed to (exact sync):
   /camera/color/image_raw,
   /camera/aligned_depth_to_color/image_raw,
   /camera/color/camera_info

MartyG-RealSense, 2020-07-06:
Hi @FrankGaoX  It looks as though the USB port is having difficulty communicating with the camera.

As a first step, please could you try adding **enable_infra2:=false** to your roslaunch statement to disable infrared2 and free up USB bandwidth. For example:

**roslaunch realsense2_camera rs_camera.launch enable_infra2:=false**

Doronhi the RealSense ROS wrapper developer has said "It will have no effect on the depth quality. It only disable the infra2 images' transmission via the USB port.  They are still taken and the depth is calculated using them inside the device".

FrankGaoX, 2020-07-06:
> Hi @FrankGaoX It looks as though the USB port is having difficulty communicating with the camera.
>
> As a first step, please could you try adding **enable_infra2:=false** to your roslaunch statement [...]

Thanks for the help!

I tried again but no luck **06/07 18:30:38,250 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61**

$ roslaunch realsense2_camera rs_camera.launch enable_infra2:=false

... logging to /home/fox/.ros/log/49ec653a-bfd8-11ea-a32b-982cbc14a1ea/roslaunch-fox-7788.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fox:36215/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 77 lines cut ...]
[ INFO] [1594074637.396686225]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1594074637.457184667]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1594074637.457412854]: setupPublishers...
[ INFO] [1594074637.479238203]: Expected frequency for depth = 30.00000
[ INFO] [1594074637.602837697]: Expected frequency for infra1 = 30.00000
[ INFO] [1594074637.678358284]: Expected frequency for color = 30.00000
[ INFO] [1594074637.747322345]: setupStreams...
[ INFO] [1594074637.857575077]: insert Depth to Stereo Module
[ INFO] [1594074637.858237969]: insert Color to RGB Camera
[ INFO] [1594074637.858410998]: insert Infrared to Stereo Module
[ INFO] [1594074638.126902403]: SELECTED BASE:Depth, 0
[ INFO] [1594074638.160619361]: RealSense Node Is Up!
 06/07 18:30:38,250 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/07 18:30:38,311 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/07 18:30:38,413 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/07 18:30:38,474 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/07 18:30:38,563 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 06/07 18:30:38,749 WARNING [547465716096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

doronhi, 2020-07-07:
We are aware of the annoying error message and it should be dealt with. No doubt.
The question is, do you notice any missing streams? Or do you get the images as expected?

FrankGaoX, 2020-07-07:
> We are aware of the annoying error message and it should be dealt with. No doubt.
> The question is, do you notice any missing streams? Or do you get the images as expected?

Yeah, the problem is, there is **no data**  to rviz, no image, nothing. 
BUT, realsense-viewer is running without any issue. I can see the picture, pointcloud.

Do I need to rebuild the kernel or something like that? I have tried so many different tutorials from jetsonhack or youtube. Sorry, I am new to Linux. I am so confused about what I should do next.

MartyG-RealSense, 2020-07-07:
@FrankGaoX   Next, could you please try adding the following instruction to your roslaunch statement to perform a hardware reset of the camera at launch:

**initial_reset:=true**

FrankGaoX, 2020-07-07:
> @FrankGaoX Next, could you please try adding the following instruction to your roslaunch statement to perform a hardware reset of the camera at launch:
> 
> **initial_reset:=true**

Still not working

I tried: 

$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true

... logging to /home/fox/.ros/log/1d46f8f0-c079-11ea-a2ef-982cbc14a1ea/roslaunch-fox-7968.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fox:41881/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 94 lines cut ...]
[ INFO] [1594143718.185981509]: setupStreams...
[ INFO] [1594143718.322074745]: insert Depth to Stereo Module
[ INFO] [1594143718.322297570]: insert Color to RGB Camera
[ INFO] [1594143718.322384919]: insert Infrared to Stereo Module
[ INFO] [1594143718.322469716]: insert Infrared to Stereo Module
[ INFO] [1594143718.595984129]: SELECTED BASE:Depth, 0
[ INFO] [1594143718.646131262]: RealSense Node Is Up!
 07/07 13:41:58,723 WARNING [547156730240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 13:41:58,784 WARNING [547156730240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 13:41:58,887 WARNING [547156730240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 13:41:58,949 WARNING [547156730240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 13:41:59,034 WARNING [547156730240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 13:41:59,217 WARNING [547156730240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/07 13:41:59,398 WARNING [547156730240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/realsense2_camera_manager-2] killing on exit
[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] escalating to SIGTERM
[camera/realsense2_camera_manager-2] escalating to SIGKILL
[rosout-1] killing on exit
[master] killing on exit
Shutdown errors:
 * process[camera/realsense2_camera_manager-2, pid 8001]: required SIGKILL. May still be running.
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2020-07-07:
Let's try finding the camera by its serial number, which I got from the logs.

**roslaunch realsense2_camera rs_camera.launch camera:=cam_1 serial_no:=947122061504**

FrankGaoX, 2020-07-07:
Hi, thank you for the help. I rebuild everything. The control_transfer returned error message is still there, but I can get picture and pointcloud in my Rviz now. WOW!

closed 2020-07-07 by FrankGaoX

MartyG-RealSense, 2020-07-07:
Great news - I'm very glad you found a solution!