realsenseai/realsense-ros #1267: 435i - No RealSense devices were found, but realsense-viewer runs perfectly

brush701, 2020-07-06:
Running on Jetson Nano, Ubuntu Bionic 18.04 LTS. I built realsense-ros from source per the instructions in the Readme. I can run realsense-viewer and rtabmap with no problems, but running any of the ROS nodes (eg `roslaunch realsense2_camera rs_camera.launch`) results in `[WARN]:  No RealSense devices were found!` Below is the terminal output. I have also run `ldd` to verify that both `realsense-viewer` and `librealsense2_camera.so` are linked against the same version of the underlying librealsense, `/usr/lib/aarch64-linux-gnu/librealsense2.so.2.35`. Any thoughts about what could be causing this discrepancy would be greatly appreciated. 

```
ben@ben-jetson:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/ben/.ros/log/45b9f902-bfc7-11ea-8a0e-6ac902b36d41/roslaunch-ben-jetson-8769.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ben-jetson:45089/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [8779]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 45b9f902-bfc7-11ea-8a0e-6ac902b36d41
process[rosout-1]: started with pid [8802]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [8809]
[ INFO] [1594067321.394545101]: Initializing nodelet with 4 worker threads.
process[camera/realsense2_camera-3]: started with pid [8811]
[ INFO] [1594067322.187608036]: RealSense ROS v2.2.14
[ INFO] [1594067322.188049970]: Running with LibRealSense v2.35.2
[ WARN] [1594067322.232738607]: No RealSense devices were found!
[ WARN] [1594067328.243469427]: No RealSense devices were found!
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

doronhi, 2020-07-07:
I noticed a situation where after running the node successfully, realsense-viewer can't find the camera. After reconnecting the camera realsense-viewer finds it. I encountered no issues running the node over and over again afterwards but could not use realsense-viewer again without reconnecting the camera.
Can you run the node if that's the first thing to connect to the camera after you reconnect it?

brush701, 2020-07-07:
Unfortunately no, it never runs successfully and always throws this error.

RealSenseSupport, 2020-07-09:
Hi @brush701 ,

What backend do you use in your system? Can you please provide a link to instructions you have followed 
for kernel patching/building the SDK with libuvc?

Thank you!

brush701, 2020-07-10:
I'm on Jetpack 4.3 / L4T  32.3.1 and ROS melodic (also built from source). I installed librealsense from the Ubuntu repository and installed realsense-ros from source using the instructions on the [homepage of this repository](https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources). Starting with L4T 32.2.1 (JetPack 4.2.2) on the NVIDIA Jetsons and the Intel RealSense SDK version v2.23.0, it is now possible to do a simple install from a RealSense debian repository rather than rebuilding librealsense from source and patching the kernel. I feel confident that this is working as expected since the `realsense-viewer` app runs correctly and produces live data from the camera.

RealSenseSupport, 2020-07-10:
Since you use Jetson Nano, please follow this [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md). Also, it's highly recommended to use the latest version of JetPack (JetPack 4.4/L4T 32.4.3 as of writing of these lines).

Thank you!

RealSenseSupport, 2020-07-17:
@brush701

Do you need any further assistance with this matter?

Thank you!

closed 2020-07-20 by RealSenseSupport