realsenseai/realsense-ros #1268: git checkout issue

kc944201, 2020-07-07:
HI, I had been used this package for a few time, I found it updated D435i yesterday

so I git clone the newest branch, when I type 

**git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`**

the package branch will change to the older branch

Could you revise the instructions? Thanks you

doronhi, 2020-07-07:
I re-checked the line and got the following good results:
```
doronhi@ubuntu:~/catkin_ws/src/realsense$ git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`
Note: checking out '2.2.14'.
```
Is it possible that you misspelled a tag (`) or something?
Also, I am not sure what do you mean by "I found it updated D435i yesterday". This package wasn't updated yesterday nor was librealsense..

kc944201, 2020-07-07:
Sorry, I got it wrong.

@mcamurri added a branch for support URDF of D435i in 19 days ago,  and when I used git checkout

The code file of D435i will be disappear, I don't know why

So later I did't use git check

closed 2020-07-13 by kc944201