realsenseai/realsense-ros #1269: SLAM with D435i - No light scenarios - L515 Lidar

shirbarzel, 2020-07-07:
Continuing on the previous issue I opened:
https://github.com/IntelRealSense/realsense-ros/issues/1233

Will the L515 LIDAR perform better in such scenarios?

blakely3dnm, 2020-07-09:
I was able to do some ICP odometry with the L515. If you are having problems with featureless walls, it sounds like you are running up against an inherent limitation of ICP, in that *IF* all the normals in your scene are perpendicular do the direction of motion, it will be impossible to resolve that motion. It is only the parallel-to-motion component of the normals that contributes information for odometry. 

[image]

In my attempt here I had to choose my path carefully so that there was always a tree in frame. If I was relying just on the ground odometry would be lost.

shirbarzel, 2020-07-09:
> I was able to do some ICP odometry with the L515. If you are having problems with featureless walls, it sounds like you are running up against an inherent limitation of ICP, in that _IF_ all the [...]

Thanks for the reply, nice work!
What about featureless scenes like white walls or dark rooms would you say L515 will work better then the D455i on these situations? In case of course there are some geometric features

blakely3dnm, 2020-07-09:
I think you will always need visual features, geometry features, or both. Otherwise you are limited to IMU only, which should generally only serve to constrain visual or ICP odometry. You can use ICP for odometry in pitch black darkness, but it requires  geometric complexity. Conversely you may be able to get reliable visual odometry in a geometric-featureless (in terms of useful normals) scene, if there are enough visual features that aren't too repetitive, but that would require good lighting. 

As for visual odometry, the L515 does not have a universal shutter for the RGB camera and I suspect any camera that does have one (e.g. D455, T265) would perform better visual odometry than the L515.

peta-peta, 2020-08-23:
@blakely3dnm  can you share some details how made ICP odometry with the L515?

RealSenseSupport, 2021-01-11:
Hi @SBarzz 

Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-27 by RealSenseSupport