realsenseai/realsense-ros #1270: ROS2 Foxy proper release

doisyg, 2020-07-07:
Repost from: https://github.com/intel/ros2_intel_realsense/issues/133

Hello,
Now that ROS2 LTS Foxy is out, many companies are considering to switch for real to ROS2. What's the planned effort for the realsense wrapper ? It seems that most of the development effort is focused on this ROS1 wrapper. 
Shouldn't the ROS2 repo (https://github.com/intel/ros2_intel_realsense)  be hosted on this organization and maintained by the same people?
Is there any development timeline and a release target date ?

doronhi, 2020-07-08:
Hello,
We feel the same way. I don't have a specific target date yet but work is already underway.
Since librealsense2 still not officially supports Ubuntu 20.04, the first realsense2_wrapper for ROS2 will be for eloquent (Ubuntu 18.04).

doisyg, 2020-07-08:
Good to know.
For info, we are using librealsense2 (deb install) on Ubuntu 20.04 + the ROS1 wrapper without any specific issue

allenh1, 2020-09-14:
Any update on this? And (slightly off topic) is there a plan to get the T265 to work with ROS 2?

doronhi, 2020-09-24:
Covid19 shuffled my plans but still we managed to release a ROS2 version. Currently only for eloquent. This version include the same support for T265 as the ROS1 wrapper.
You can download it from: https://github.com/IntelRealSense/realsense-ros/tree/eloquent

allenh1, 2020-09-24:
> This version include the same support for T265 as the ROS1 wrapper.

Fantastic news! Thanks!

closed 2020-09-24 by RealSenseSupport

liushuya7, 2020-11-25:
Can this eloquent package be run on ROS2 Foxy? I built it and run the camera node, everything seems working fine except there is no image showing from the camera topic.
See [my post](https://github.com/IntelRealSense/realsense-ros/issues/1336#issuecomment-733466270) in Issue #1336