realsenseai/realsense-ros #1271: Control transfer returned error

amanalam1, 2020-07-07:
Hi everyone, 

I am reporting this issue after launching realsense2_camera node:

WARNING [281472619504000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

I am using a Raspberry Pi 4 running Ubuntu 18.04 LTS, ROS Melodic with a D435 and kernel version 5.3.0-1027-raspi2. 

Please see the following output:

ubuntu@ubuntu:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/ubuntu/.ros/log/79d7905e-c081-11ea-858e-dca63217cdd6/roslaunch-ubuntu-4047.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:41281/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 73 lines cut ...]
[ INFO] [1594147297.354121078]: num_filters: 0
[ INFO] [1594147297.354321443]: Setting Dynamic reconfig parameters.
[ INFO] [1594147297.901105954]: Done Setting Dynamic reconfig parameters.
[ INFO] [1594147297.907511110]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1594147297.908232098]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1594147297.908801752]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1594147297.910212749]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1594147297.910372211]: setupPublishers...
[ INFO] [1594147297.922072064]: Expected frequency for depth = 30.00000
[ INFO] [1594147298.064061486]: Expected frequency for infra1 = 30.00000
[ INFO] [1594147298.139933817]: Expected frequency for infra2 = 30.00000
[ INFO] [1594147298.223664822]: Expected frequency for color = 30.00000
[ INFO] [1594147298.301020457]: setupStreams...
[ INFO] [1594147298.325085182]: insert Depth to Stereo Module
[ INFO] [1594147298.325474876]: insert Color to RGB Camera
[ INFO] [1594147298.325749695]: insert Infrared to Stereo Module
[ INFO] [1594147298.326009627]: insert Infrared to Stereo Module
[ INFO] [1594147298.445912656]: SELECTED BASE:Depth, 0
[ INFO] [1594147298.480118506]: RealSense Node Is Up!
 07/07 18:41:38,574 WARNING [281472619504000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 18:41:38,625 WARNING [281472619504000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 18:41:38,676 WARNING [281472619504000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 18:41:38,815 WARNING [281472619504000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 18:41:38,866 WARNING [281472619504000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 07/07 19:16:20,566 WARNING [281472619504000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2020-07-08:
Hi @amanalam1  control_transfer returned is a common error message experienced by RealSense ROS users, and a problem that the developer acknowledges needs solving.

https://github.com/IntelRealSense/realsense-ros/issues/1266#issuecomment-654691643

If the warning only occurs at the start of launch or at a frequency of around once per minute then it can be safely ignored.  

I note from the log that you posted that the launch process did not "die at the end of it, which is a positive sign.  It means that the ROS camera topics will hopefully be being published successfully in RViz despite the log messages.  Do you see the camera images published in RViz, please?

amanalam1, 2020-07-08:
Thanks @MartyG-RealSense for your reply. There are no images and no data to rviz. But I can run realsense-viewer without any problem. I had a look to #1266 (comment)and I tried to reset hardware and connect it with serial number but still doesn't work.

initial_reset:=true
roslaunch realsense2_camera rs_camera.launch camera:=cam_1 serial_no:=013222070833

amanalam1, 2020-07-08:
I used this command roslaunch realsense2_camera rs_camera.launch enable_infra2:=false  now I am getting this.

roslaunch realsense2_camera rs_camera.launch
... logging to /home/ubuntu/.ros/log/033139a8-c155-11ea-a8c9-dca63217cdd6/roslaunch-ubuntu-6537.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:43203/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 68 lines cut ...]
[ INFO] [1594238150.525792542]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1594238150.526546156]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1594238150.527175547]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1594238150.528743424]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1594238150.528903221]: setupPublishers...
[ INFO] [1594238150.535050932]: Expected frequency for depth = 30.00000
[ INFO] [1594238150.626900738]: Expected frequency for infra1 = 30.00000
[ INFO] [1594238150.701790696]: Expected frequency for infra2 = 30.00000
[ INFO] [1594238150.776639690]: Expected frequency for color = 30.00000
[ INFO] [1594238150.864756388]: setupStreams...
[ INFO] [1594238150.889058156]: insert Depth to Stereo Module
[ INFO] [1594238150.889230638]: insert Color to RGB Camera
[ INFO] [1594238150.889306990]: insert Infrared to Stereo Module
[ INFO] [1594238150.889425027]: insert Infrared to Stereo Module
[ INFO] [1594238151.010790807]: SELECTED BASE:Depth, 0
 08/07 19:55:51,026 WARNING [281472887939456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1594238151.078918070]: RealSense Node Is Up!
 08/07 19:55:51,140 WARNING [281472887939456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/07 19:55:51,191 WARNING [281472887939456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 08/07 19:55:51,378 WARNING [281472887939456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/07 19:55:51,429 WARNING [281472887939456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/07 19:55:51,479 WARNING [281472887939456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/07 19:55:51,540 WARNING [281472887939456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2020-07-08:
Can you view a point cloud in RViz if you follow this guide:

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

amanalam1, 2020-07-09:
I can not see a point cloud in RViz. There were too many errors then at the end I used ctrl+alt+ c to exit.

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/ubuntu/.ros/log/9178b7c8-c21c-11ea-b615-dca63217cdd6/roslaunch-ubuntu-4582.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:43427/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 975 lines cut ...]
 09/07 19:46:42,486 ERROR [281472509194624] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 131
 09/07 19:46:42,559 ERROR [281472560787840] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 09/07 19:46:42,964 WARNING [281472225243520] (types.cpp:49) Out of frame resources!
 09/07 19:46:42,983 ERROR [281472225243520] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 09/07 19:46:42,989 WARNING [281472225243520] (types.cpp:49) Out of frame resources!
 09/07 19:46:42,998 ERROR [281472225243520] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 09/07 19:46:43,000 WARNING [281472225243520] (types.cpp:49) Out of frame resources!
 09/07 19:46:43,006 ERROR [281472225243520] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 09/07 19:46:43,008 WARNING [281472225243520] (types.cpp:49) Out of frame resources!
 09/07 19:46:43,008 ERROR [281472225243520] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 09/07 19:46:43,010 WARNING [281472225243520] (types.cpp:49) Out of frame resources!
 09/07 19:46:43,017 ERROR [281472225243520] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 09/07 19:46:43,020 WARNING [281472225243520] (types.cpp:49) Out of frame resources!
 09/07 19:46:43,026 ERROR [281472225243520] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 09/07 19:46:43,028 WARNING [281472225243520] (types.cpp:49) Out of frame resources!
[... cut ...]
 09/07 19:46:43,057 WARNING [281472225243520] (types.cpp:49) Out of frame resources!
 09/07 19:46:43,057 ERROR [281472225243520] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
 09/07 19:46:43,085 WARNING [281472225243520] (types.cpp:49) Out of frame resources!
 09/07 19:46:43,086 ERROR [281472225243520] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2020-07-09:
Does the situation improve if you include a camera reset instruction called **initial_reset** in your launch instruction, please?

roslaunch realsense2_camera rs_camera.launch initial_reset:=true

amanalam1, 2020-07-10:
yes. Here is the result.

roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/ubuntu/.ros/log/278a527e-c305-11ea-9698-dca63217cdd6/roslaunch-ubuntu-3893.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:42745/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 72 lines cut ...]
[ INFO] [1594423759.882224769]: num_filters: 0
[ INFO] [1594423759.882294306]: Setting Dynamic reconfig parameters.
[ INFO] [1594423760.444492898]: Done Setting Dynamic reconfig parameters.
[ INFO] [1594423760.446708102]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1594423760.447872380]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1594423760.448925658]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1594423760.451473065]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1594423760.451730584]: setupPublishers...
[ INFO] [1594423760.463073435]: Expected frequency for depth = 30.00000
[ INFO] [1594423760.556228861]: Expected frequency for infra1 = 30.00000
[ INFO] [1594423760.639062880]: Expected frequency for infra2 = 30.00000
[ INFO] [1594423760.710566546]: Expected frequency for color = 30.00000
[ INFO] [1594423760.788355250]: setupStreams...
[ INFO] [1594423760.812974824]: insert Depth to Stereo Module
[ INFO] [1594423760.813180176]: insert Color to RGB Camera
[ INFO] [1594423760.813258602]: insert Infrared to Stereo Module
[ INFO] [1594423760.813373935]: insert Infrared to Stereo Module
[ INFO] [1594423760.930454231]: SELECTED BASE:Depth, 0
 10/07 23:29:20,948 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1594423761.002153102]: RealSense Node Is Up!
 10/07 23:29:21,065 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...] , error: No data available, number: 61
 10/07 23:29:21,169 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 23:29:21,300 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 23:29:21,351 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

amanalam1, 2020-07-10:
I followed the below link instruction to install realsense viewer. Maybe thats the reason I am getting this error?

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

amanalam1, 2020-07-10:
> Does the situation improve if you include a camera reset instruction called **initial_reset** in your launch instruction, please?
> 
> **roslaunch realsense2_camera rs_camera.launch initial_reset:=true

Yes. Here is the result

roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/ubuntu/.ros/log/278a527e-c305-11ea-9698-dca63217cdd6/roslaunch-ubuntu-3893.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:42745/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 78 lines cut ...]
[ INFO] [1594423760.444492898]: Done Setting Dynamic reconfig parameters.
[ INFO] [1594423760.446708102]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1594423760.447872380]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1594423760.448925658]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1594423760.451473065]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1594423760.451730584]: setupPublishers...
[ INFO] [1594423760.463073435]: Expected frequency for depth = 30.00000
[ INFO] [1594423760.556228861]: Expected frequency for infra1 = 30.00000
[ INFO] [1594423760.639062880]: Expected frequency for infra2 = 30.00000
[ INFO] [1594423760.710566546]: Expected frequency for color = 30.00000
[ INFO] [1594423760.788355250]: setupStreams...
[ INFO] [1594423760.812974824]: insert Depth to Stereo Module
[ INFO] [1594423760.813180176]: insert Color to RGB Camera
[ INFO] [1594423760.813258602]: insert Infrared to Stereo Module
[ INFO] [1594423760.813373935]: insert Infrared to Stereo Module
[ INFO] [1594423760.930454231]: SELECTED BASE:Depth, 0
 10/07 23:29:20,948 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1594423761.002153102]: RealSense Node Is Up!
 10/07 23:29:21,065 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 23:29:21,118 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...] or: No data available, number: 61
 10/07 23:29:21,300 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/07 23:29:21,351 WARNING [281472426570112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/ubuntu/.ros/log/278a527e-c305-11ea-9698-dca63217cdd6/camera-realsense2_camera-3*.log

MartyG-RealSense, 2020-07-11:
I'm very pleased that you got some improvement.  Are the camera images published in RViz now, please?

amanalam1, 2020-07-11:
Unfortunately there is no images on rviz 

[image]

MartyG-RealSense, 2020-07-11:
You could check if the topics are publishing even if they are not appearing in RViz by doing **rostopic hz**

http://wiki.ros.org/rostopic#rostopic_hz

amanalam1, 2020-07-11:
I am new with ros. Thanks for your help and here is the result as you asked.

`ubuntu@ubuntu:~$ rostopic hz /camera/color/image_raw
subscribed to [/camera/color/image_raw]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
^C^[^Xno new messages
ubuntu@ubuntu:~$ rostopic hz /camera/color/camera_info
subscribed to [/camera/color/camera_info]
no new messages
no new messages
no new messages
no new messages
^[^Cno new messages
ubuntu@ubuntu:~$ rostopic hz /camera/depth/camera_info
subscribed to [/camera/depth/camera_info]
no new messages
no new messages
no new messages
no new messages
^[^Cno new messages`

MartyG-RealSense, 2020-07-11:
You are welcome.  :)  Thank you very much for performing the hz test.  It looks as though nothing is being published by the topics.  So there is no data being published in RViz either.

I note that at the end of the most recent log, it says **[camera/realsense2_camera-3] process has finished cleanly**.  It is apparently a way of saying that the launch process has stopped and exited.  

I have researched this case again very carefully, but **control_transfer returned** warnings are very hard to diagnose and solve.  Sometimes the problem just goes away if a fresh rebuild of librealsense and the ROS wrapper is done.

amanalam1, 2020-07-12:
I tried to rebuild everything on different sd card but I got the same issue of control_transfer returned. 

In both cases I couldn't follow up the ROS wrapper because I get an error while using this command  'sudo apt-get install librealsense2-dkms'. Then I followed LibUVC-backend installation instruction. Is that a problem to install ?

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

MartyG-RealSense, 2020-07-13:
Raspberry Pi does not use the 'x' processor architecture like x86 / x64, but instead is based on Arm processors.  For devices that are not using x86 / x64, it is recommended to build Librealsense using the **backend** method.  Install librealsense first before installing the ROS wrapper.

The LibUVC backend method that you linked to works but is outdated now.  From RealSense SDK 2.30.0 onward, a new backend installation method called **RSUSB** was introduced.

I would recommend using the RSUSB backend method to build the source code for SDK **2.35.2** rather than the latest 2.36.0, as there is a matching ROS wrapper version **2.2.14** for SDK 2.35.2.

Instructions for doing a backend build of librealsense with RSUSB are in the link below:

https://github.com/IntelRealSense/librealsense/issues/6710#issuecomment-650611442

amanalam1, 2020-07-14:
I downloaded Realsense SDK2.30.0 then I extracted the source-code zip file. I used this command to creat build folder 
mkdir build && cd build

I ran the CMake command 
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

The realsense viewer doesnt install with this way. Do I missing some steps here?

`ubuntu@ubuntu:~$ cd librealsense
ubuntu@ubuntu:~/librealsense$ mkdir build && cd build
ubuntu@ubuntu:~/librealsense/build$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.36.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
-- Found usb: /usr/lib/aarch64-linux-gnu/libusb-1.0.so  
-- GLFW 3.3 not found; using internal version
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Using X11 for window creation
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/aarch64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so   
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- Building with TM2
-- Fetching recommended firmwares:
-- D4XX_FW_VERSION: 5.12.6.0
-- SR3XX_FW_VERSION: 3.26.1.0
-- T26X_FW_VERSION: 0.2.0.951
-- L5XX_FW_VERSION: 1.4.1.2
-- http://realsense-hw-public.s3.amazonaws.com/Releases/RS4xx/FW/D4XX_FW_Image-5.12.6.0.bin
-- Download firmware 0;"No error" for D4XX_FW_Image-5.12.6.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- Download firmware 0;"No error" for SR3XX_FW_Image-3.26.1.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd
-- Download firmware 0;"No error" for target-0.2.0.951.mvcmd
-- http://realsense-hw-public.s3.amazonaws.com/Releases/L5xx/FW/L5XX_FW_Image-1.4.1.2.bin
-- Download firmware 0;"No error" for L5XX_FW_Image-1.4.1.2.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/librealsense/build
ubuntu@ubuntu:~/librealsense/build$ 
`

amanalam1, 2020-07-14:
I downloaded Realsense SDK2.30.0 then I extracted the source-code zip file. I used this command to creat build folder
mkdir build && cd build

I ran the CMake command
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

The realsense viewer doesnt install with this way. Do I missing some steps here?

> ubuntu@ubuntu:~$ cd librealsense
ubuntu@ubuntu:~/librealsense$ mkdir build && cd build
ubuntu@ubuntu:~/librealsense/build$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.36.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
-- Found usb: /usr/lib/aarch64-linux-gnu/libusb-1.0.so  
-- GLFW 3.3 not found; using internal version
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Using X11 for window creation
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/aarch64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so   
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- Building with TM2
-- Fetching recommended firmwares:
-- D4XX_FW_VERSION: 5.12.6.0
-- SR3XX_FW_VERSION: 3.26.1.0
-- T26X_FW_VERSION: 0.2.0.951
-- L5XX_FW_VERSION: 1.4.1.2
-- http://realsense-hw-public.s3.amazonaws.com/Releases/RS4xx/FW/D4XX_FW_Image-5.12.6.0.bin
-- Download firmware 0;"No error" for D4XX_FW_Image-5.12.6.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- Download firmware 0;"No error" for SR3XX_FW_Image-3.26.1.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd
-- Download firmware 0;"No error" for target-0.2.0.951.mvcmd
-- http://realsense-hw-public.s3.amazonaws.com/Releases/L5xx/FW/L5XX_FW_Image-1.4.1.2.bin
-- Download firmware 0;"No error" for L5XX_FW_Image-1.4.1.2.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/librealsense/build
ubuntu@ubuntu:~/librealsense/build$

amanalam1, 2020-07-14:
I downloaded Realsense SDK2.30.0 then I extracted the source-code zip file. I used this command to creat build folder
mkdir build && cd build

I ran the CMake command
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

The realsense viewer doesnt install with this way. Do I missing some steps here?

ubuntu@ubuntu:~$ cd librealsense
ubuntu@ubuntu:~/librealsense$ mkdir build && cd build
ubuntu@ubuntu:~/librealsense/build$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.36.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
-- Found usb: /usr/lib/aarch64-linux-gnu/libusb-1.0.so  
-- GLFW 3.3 not found; using internal version
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Using X11 for window creation
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/aarch64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so   
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- Building with TM2
-- Fetching recommended firmwares:
-- D4XX_FW_VERSION: 5.12.6.0
-- SR3XX_FW_VERSION: 3.26.1.0
-- T26X_FW_VERSION: 0.2.0.951
-- L5XX_FW_VERSION: 1.4.1.2
-- http://realsense-hw-public.s3.amazonaws.com/Releases/RS4xx/FW/D4XX_FW_Image-5.12.6.0.bin
-- Download firmware 0;"No error" for D4XX_FW_Image-5.12.6.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- Download firmware 0;"No error" for SR3XX_FW_Image-3.26.1.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd
-- Download firmware 0;"No error" for target-0.2.0.951.mvcmd
-- http://realsense-hw-public.s3.amazonaws.com/Releases/L5xx/FW/L5XX_FW_Image-1.4.1.2.bin
-- Download firmware 0;"No error" for L5XX_FW_Image-1.4.1.2.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/librealsense/build
ubuntu@ubuntu:~/librealsense/build$

amanalam1, 2020-07-14:
I downloaded Realsense SDK2.30.0 then I extracted the source-code zip file. I used this command to creat build folder
mkdir build && cd build

I ran the CMake command
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

The realsense viewer doesnt install with this way. Do I missing some steps here?

<pre><font color="#8AE234"><b>ubuntu@ubuntu</b></font>:<font color="#729FCF"><b>~</b></font>$ cd librealsense
<font color="#8AE234"><b>ubuntu@ubuntu</b></font>:<font color="#729FCF"><b>~/librealsense</b></font>$ mkdir build &amp;&amp; cd build
<font color="#8AE234"><b>ubuntu@ubuntu</b></font>:<font color="#729FCF"><b>~/librealsense/build</b></font>$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.36.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
-- Found usb: /usr/lib/aarch64-linux-gnu/libusb-1.0.so  
-- GLFW 3.3 not found; using internal version
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Using X11 for window creation
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/aarch64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so   
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- Building with TM2
-- Fetching recommended firmwares:
-- D4XX_FW_VERSION: 5.12.6.0
-- SR3XX_FW_VERSION: 3.26.1.0
-- T26X_FW_VERSION: 0.2.0.951
-- L5XX_FW_VERSION: 1.4.1.2
-- http://realsense-hw-public.s3.amazonaws.com/Releases/RS4xx/FW/D4XX_FW_Image-5.12.6.0.bin
-- Download firmware 0;&quot;No error&quot; for D4XX_FW_Image-5.12.6.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- Download firmware 0;&quot;No error&quot; for SR3XX_FW_Image-3.26.1.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd
-- Download firmware 0;&quot;No error&quot; for target-0.2.0.951.mvcmd
-- http://realsense-hw-public.s3.amazonaws.com/Releases/L5xx/FW/L5XX_FW_Image-1.4.1.2.bin
-- Download firmware 0;&quot;No error&quot; for L5XX_FW_Image-1.4.1.2.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/librealsense/build
<font color="#8AE234"><b>ubuntu@ubuntu</b></font>:<font color="#729FCF"><b>~/librealsense/build</b></font>$ 
</pre>

MartyG-RealSense, 2020-07-15:
@amanalam1  The CMake instruction looks fine.  You can use it to build from source code SDK versions that are newer than 2.30.0 though, right up to the most recent version, which is what I meant by 2.30.0 **onward** (i.e ones that were released after 2.30.0).

jorgemia, 2020-07-15:
@amanalam1 whilst your problem might have to do with the topics not publishing properly just fyi (in case you were not aware...if you already know then ignore this) to visualise something in rviz you have to go to ADD and the choose the topic you want to subcribe to (in my image the camera has not launched but available topics should appear there). To view, the pointcloud I think you also have to set the Fixed Frame (top left) on the drop down list to camera_link:

[image]

amanalam1, 2020-07-15:
@jorgemia Thanks for your help. I did as what you said now I can see topics is publishing but I dont have any visualization of the camera on rviz.

[image]

<pre><font color="#8AE234"><b>ubuntu@ubuntu</b></font>:<font color="#729FCF"><b>~/catkin_ws</b></font>$ rostopic hz /camera/color/image_raw

subscribed to [/camera/color/image_raw]
average rate: 15.046
	min: 0.054s max: 0.078s std dev: 0.00821s window: 14
average rate: 14.979
	min: 0.054s max: 0.080s std dev: 0.00689s window: 29
average rate: 14.945
	min: 0.054s max: 0.080s std dev: 0.00641s window: 44
average rate: 14.938
	min: 0.054s max: 0.084s std dev: 0.00674s window: 59
average rate: 14.956
	min: 0.050s max: 0.084s std dev: 0.00727s window: 74
average rate: 14.941
	min: 0.042s max: 0.084s std dev: 0.00772s window: 89
average rate: 14.954
	min: 0.042s max: 0.084s std dev: 0.00735s window: 104
average rate: 14.957
	min: 0.042s max: 0.084s std dev: 0.00753s window: 119
average rate: 14.942
	min: 0.042s max: 0.084s std dev: 0.00749s window: 134
average rate: 14.939
	min: 0.042s max: 0.084s std dev: 0.00729s window: 148
average rate: 14.944
	min: 0.042s max: 0.084s std dev: 0.00709s window: 163
average rate: 14.942
	min: 0.042s max: 0.084s std dev: 0.00706s window: 178
^Caverage rate: 14.949
	min: 0.042s max: 0.084s std dev: 0.00714s window: 184
<font color="#8AE234"><b>ubuntu@ubuntu</b></font>:<font color="#729FCF"><b>~/catkin_ws</b></font>$ rostopic hz /camera/extrinsics/depth_to_color
subscribed to [/camera/extrinsics/depth_to_color]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
^[^Cno new messages
<font color="#8AE234"><b>ubuntu@ubuntu</b></font>:<font color="#729FCF"><b>~/catkin_ws</b></font>$ rostopic hz /camera/depth/camera_info
subscribed to [/camera/depth/camera_info]
average rate: 30.007
	min: 0.031s max: 0.036s std dev: 0.00087s window: 28
average rate: 29.954
	min: 0.030s max: 0.037s std dev: 0.00134s window: 58
average rate: 29.944
	min: 0.028s max: 0.038s std dev: 0.00177s window: 88
average rate: 29.995
	min: 0.020s max: 0.042s std dev: 0.00231s window: 118
^Caverage rate: 29.999
	min: 0.020s max: 0.045s std dev: 0.00272s window: 122
<font color="#8AE234"><b>ubuntu@ubuntu</b></font>:<font color="#729FCF"><b>~/catkin_ws</b></font>$ rostopic hz /camera/infra1/image_rect_raw
subscribed to [/camera/infra1/image_rect_raw]
average rate: 29.898
	min: 0.029s max: 0.037s std dev: 0.00209s window: 28
average rate: 30.042
	min: 0.029s max: 0.037s std dev: 0.00182s window: 58
average rate: 30.033
	min: 0.028s max: 0.039s std dev: 0.00185s window: 88
^[^Caverage rate: 30.026
	min: 0.028s max: 0.039s std dev: 0.00179s window: 109
<font color="#8AE234"><b>ubuntu@ubuntu</b></font>:<font color="#729FCF"><b>~/catkin_ws</b></font>$ </pre>

`

jorgemia, 2020-07-15:
@amanalam1 If you click on the camera->color->image_raw drop down and select **Image** do you still see nothing?

amanalam1, 2020-07-15:
yes I have visualization now.

[image]

MartyG-RealSense, 2020-08-02:
Case closed due to no further comments received.

closed 2020-08-02 by MartyG-RealSense

htawadegetample, 2021-03-22:
I am seeing the same issue for l515 camera. The "control transfer returned error" seems to be appearing every 1 second. I installed from source and used RSUSB_BACKEND flag during compilation. I have used initial_reset=true during launch.

MartyG-RealSense, 2021-03-22:
Hi @htawadegetample   Please create a new case about your problem by visiting the front page of this forum at the link below and clicking the **New Issue** button.  Please include 'L515' in the title of your message.   Thanks!

https://github.com/IntelRealSense/realsense-ros/issues

htawadegetample, 2021-03-22:
> Hi @htawadegetample Please create a new case about your problem by visiting the front page of this forum at the link below and clicking the **New Issue** button. Please include 'L515' in the title of your message. Thanks!
> 
> https://github.com/IntelRealSense/realsense-ros/issues

https://github.com/IntelRealSense/realsense-ros/issues/1771