realsenseai/realsense-ros #1273: L515 Incorrect Header Timestamps

pauldeee, 2020-07-08:
Hi,

I have librealsense built from source on 16.04 with ROS kinetic. When I use the following launch file it would appear that all the L515 topic header stamps appear to be almost exactly doubled from what their value should be. This appears to only be the case when launching both the T265 and the L515. The L515 appears to have the correct header stamps when launching it alone.

`<launch>
    
    <arg name="HD" value="false"/>
      
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="serial_no" value="00000000f0140893"/>
        <arg name="unite_imu_method" value="linear_interpolation"/>
        <arg name="camera" value="L515"/>
        <arg name="enable_gyro" value="true"/>
        <arg name="enable_accel" value="true"/>
        <arg name="accel_fps" value="400"/>
        <arg name="gyro_fps" value="400"/>
        
        
        <arg if="$(arg HD)" name="depth_width" value="1024"/>
        <arg if="$(arg HD)" name="depth_height" value="768"/>
        <arg if="$(arg HD)" name="color_width" value="1920"/>
        <arg if="$(arg HD)" name="color_height" value="1080"/>        

        
        <arg unless="$(arg HD)" name="depth_width" value="640"/>
        <arg unless="$(arg HD)" name="depth_height" value="480"/>
        <arg unless="$(arg HD)" name="color_width" value="1280"/>
        <arg unless="$(arg HD)" name="color_height" value="720"/>
    </include>

    <include file="$(find realsense2_camera)/launch/rs_t265.launch">
        <arg name="serial_no" value="929122110230"/>
        <arg name="camera" value="T265"/>
        <arg name="unite_imu_method" value="linear_interpolation"/>
        <arg name="enable_fisheye1" value="true"/>
        <arg name="enable_fisheye2" value="true"/>
    </include>
</launch>`

Here is a screenshot of rqt_bag:
[image]

I do see a few warnings from the Realsense Node:
`[ WARN] [1594234502.485392750]: (Gyro, 0) -> (Depth, 0): Requested extrinsics are not available! : using unity as default.
 08/07 13:55:02,485 WARNING [140156005811968] (types.cpp:49) Requested extrinsics are not available!
[ WARN] [1594234502.485949001]: (Accel, 0) -> (Depth, 0): Requested extrinsics are not available! : using unity as default.
[ INFO] [1594234502.493554982]: RealSense Node Is Up!
 08/07 13:55:02,530 WARNING [140155484866304] (l500-depth.cpp:205) UVC metadata payloads are not available for stream . Please refer to installation chapter for details.
 08/07 13:55:02,560 WARNING [140155503499008] (sensor.cpp:970) HID timestamp not found, switching to Host timestamps.
 08/07 13:55:03,745 WARNING [140155761407744] (l500-depth.cpp:205) UVC metadata payloads are not available for stream . Please refer to installation chapter for details.`

Also, if I attempt to launch with `roslaunch realsense2_camera rs_camera.launch accel_fps:=400 gyro_fps:=400 enable_accel:=true enable_gyro:=true enable_color:=false unite_imu_method:=linear_interpolation` or `roslaunch realsense2_camera rs_camera.launch accel_fps:=400 gyro_fps:=400 enable_accel:=true enable_gyro:=true color_width:=1280 color_height:=720` the header stamps are doubled.

However, when I launch just `roslaunch realsense2_camera rs_camera.launch` the L515 header stamps appear to be correct, but I do see this at the end:
`[ INFO] [1594234789.725904233]: RealSense Node Is Up!
 08/07 13:59:51,847 WARNING [140637728085760] (l500-depth.cpp:205) UVC metadata payloads are not available for stream . Please refer to installation chapter for details.
[ WARN] [1594234791.950745028]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)`

Furthermore, if I launch with `roslaunch realsense2_camera rs_camera.launch accel_fps:=400 gyro_fps:=400 enable_accel:=true enable_gyro:=true` the header stamps are correct.

Any ideas? It seems that passing certain parameters / args leads to somehow causing the header stamps to double?

RealSenseSupport, 2020-07-10:
Hi @pauldeee ,

We are unable to reproduce this issue using either the parameter overrides on the command line or your launch file exactly as is. Can you answer some questions so that we might be able see what's going?

1) Are you seeing this ONLY when you have the T265 and L515 plugged in together. Or does this happen if you only have the L515 plugged in and only launch with rs_camera.launch not rs_t265.launch?

2) Are you seeing the exact same numbers if you do "rostopic echo /L515/color/camera_info" as well? I just want to rule out any bag file record/replay issues.

3) What platform are you on? PC/RasberryPi/Jetson/etc?

4) The UVC metadata error suggests that there was a problem during installation. Have you tried installing from the Debian packages? Metadata is not supported natively in the standard Linux UVC driver and needs to be patched. Since hardware timestamp are metadata and you're getting this warning those might be related.

I would suggest removing everything and installing the latest packages instead of from source if you can. 

https://github.com/IntelRealSense/realsense-ros#install-from-debian-package---in-that-case-treat-yourself-as-a-developer-make-sure-you-follow-the-instructions-to-also-install-librealsense2-dev-and-librealsense-dkms-packages

Thanks

pauldeee, 2020-07-12:
Hi, @RealSenseCustomerSupport

It seems installing via the Debian packages has solved this issue.

In response to your questions and providing more detail:
1) It was happening with the L515 alone and with the L515 and T265 combined. It seemed to happen when passing certain parameters that were different from the defaults. For example, by default I was getting a warning that color stream resolution 640 x 480 was not valid. When setting the color resolution to 1280x720 the timestamps would then be doubled. Also leaving all settings default and setting the unite_imu_method to "linear_interpolation" seem to also cause the timestamps to double.

2) I believe if it was happening on one L515 topic, it was happening across all L515 topics. At least the all ones I checked.

3) Intel NUC, Ubuntu 16.04, ROS Kinetic.

4) This appears to have solved the header stamp issue.

Regards

closed 2020-07-12 by pauldeee

RealSenseSupport, 2020-07-13:
Awesome. Glad that worked and thanks for the info.

bigbellmercy, 2020-09-08:
@RealSenseSupport The same problem happened for L515. The L515 time stamp is doubled after the dynamic parameter `global_time_enabled` is changed to `true` in order to match the time stamps of depth image and imu (because I have a problem that the time stamp delay of imu is increasing of itself).

Environment:
 Ubuntu 18.04 kernel 5.4.0-47-generic
 libRealSense v2.37.0
 realsense-ros wrapper v2.2.16
 L515 fw v1.5.0

For the comment above:
> The UVC metadata error suggests that there was a problem during installation. Have you tried installing from the Debian packages? Metadata is not supported natively in the standard Linux UVC driver [...]

So I re-installed librealsense devian package and ros wrapper again as below but the problem is the same:
> I would suggest removing everything and installing the latest packages instead of from source if you can. [...]

According to [Linux Distribution](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages), my Ubuntu kernel v5.4 is not listed in the supported kernel versions. Should I down-grade to kernel v5.3?