realsenseai/realsense-ros #1274: Need help to start working with Realsense D435i and ROS Melodic

Cristian-wp, 2020-07-09:
Hello,
I do not have experience with this type of hardware so be patient please :)

I have correctly installed the realsense-ros package and the other parts. I was able to launch the example with Rviz:

_roslaunch realsense2_camera rs_camera.launch filters:=pointcloud_

Now I wish to read the pointcloud data, image, ecc...get from the camera with a topic using a subscriber in pyhon or C++(better).
With rqt_graph I see that the topic with pointcloud information is 

_/camera/depth/color/points_

Now the basic problem is that I do not know how start...can someone provide a basic example o tutorial?

Thank you in advance.

MartyG-RealSense, 2020-07-09:
Hi @Cristian-wp  If your goal is to get xyz coordinates from pointcloud2, I recommend reading through the discussion in the link below and exploring the links within it .

https://github.com/IntelRealSense/realsense-ros/issues/1254

MartyG-RealSense, 2020-07-18:
Hi @Cristian-wp   Do you still require assistance with this case, please?   Thanks!

Cristian-wp, 2020-07-20:
Hi @MartyG-RealSense yes, I need some help.

MartyG-RealSense, 2020-07-20:
It sounds as though you have successfully completed the example point cloud generation tutorial in the ROS wrapper documentation.  Is that correct, please?

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

As you have a D435i, the next point cloud tutorial that you could move on to is this one:

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

MartyG-RealSense, 2020-08-02:
Hi @Cristian-wp   Do you still require assistance with this case please, or can it be closed?   Thanks!

closed 2020-08-08 by MartyG-RealSense