realsenseai/realsense-ros #1275: L515 Incorrect imu Header Timestamps

wangyuanbiubiubiu, 2020-07-10:
I want to read the IMU data and image data of L515 laser camera. I find that the timestamp of the image is the timestamp of the system, while the timestamp of the IMU is the timestamp of the device. How should I fix this problem?
Below is my Launch file Settings：
[image]
Below is my log：
[image]

wangyuanbiubiubiu, 2020-07-11:
I have now checked the type of timestamp and found the following problems：
[image]
Output in terminal
[image]

RealSenseSupport, 2020-07-14:
Hi @wangyuanbiubiubiu 

I'm not sure I understand what you mean by you "found the following problems:"...

...but have you tried setting the *global_time_enabled* parameters to _true_ on both camera/motion_module and camera/rgb_camera? This should get you both timestamps in the same domain.

Thanks

closed 2020-07-30 by RealSenseSupport