realsenseai/realsense-ros #1276: Odometry of T265 is not being published when starting the default launch file from the development branch.

alessandromorra, 2020-07-10:
OS: Ubuntu 18.04
Branch: `development` [default when installing]

I have tried to launch my T265 sensor via ROS like that:

```
roslaunch realsense2_camera rs_t265.launch 
```

Unfortunately, my `/camera/odom/sample` is not available:

```
$ rostopic echo /camera/odom/sample 
WARNING: topic [/camera/odom/sample] does not appear to be published yet
```

Only when I switch to this branch then it works: `2.2.14` [probably also others but I did not test them all]

I am not sure if this is a bug or a feature, just to let you guys know.

droter, 2020-08-27:
OS: Ubuntu 16.04
Branch: development [default when installing]

Also seeing the same thing.  IMU and Gyro are publishing but no odom.

FPSychotic, 2020-10-12:
same here...last version

nilseuropa, 2021-03-02:
same here as well

nilseuropa, 2021-03-02:
`      <arg name="enable_pose"              value="true"/>
`
is missing from the launch file

doronhi, 2021-03-03:
I am sorry for paying attention so late.
As the line "<arg name="enable_pose"         default="true"/>" is now in the rs_t265.launch file, does it work now? Can this issue be closed?

alessandromorra, 2021-03-04:
On version `2.2.22` it works. Form my side this issue can be closed.

nilseuropa, 2021-03-04:
I can confirm that it works.

closed 2021-05-03 by RealSenseSupport