realsenseai/realsense-ros #1279: Multiple Cameras Jetson Xavier USB SCP overflow

RGring, 2020-07-13:
Hi, 
I run the following setup:

- Jetson Xavier NX, L4T 32.4.3
- Stereo camera ZED2 (connected to USB3.0)
- Stereo camera Realsense D435i (connected to USB3.0)
- all in ROS

I have updated the firmware to 5.12.6.0. 

I reduced the fps to 15, but I need to keep the highest resolution for depth and rgb.

Unfortunately I still get the error:
**Hardware Notification:USB SCP overflow,1.59463e+12,Error,Hardware Error**

Any suggestions, how I could solve this?

MartyG-RealSense, 2020-07-13:
You could try disabling **infrared2**in  your roslaunch statement.  It saves on USB transportation and kernel work, according to Doronhi the RealSense ROS wrapper developer.  Doronhi adds "It will have no effect on the depth quality.  It only disables the infra2 images' transmission via the USB port.  They are still taken and the depth is calculated using them inside the device".

An example statement would be:

**roslaunch realsense2_camera rs_camera.launch enable_infra2:=false**

RGring, 2020-07-13:
Thanks for the reply. In the rs_camera.launch the enable_infra1 and enable_infra2 are set to false by default. So I had this already  applied. I only change the following parameters.

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="realsense"/>
    <arg name="enable_sync" value="true"/>
    <arg name="align_depth" value="true"/>

    <arg name="depth_width" value="1280"/>
    <arg name="depth_height" value="720"/>
    <arg name="infra_width" value="1280"/>
    <arg name="infra_height" value="720"/>

    <arg name="color_width" value="1920"/>
    <arg name="color_height" value="1080"/>
    
    <arg name="depth_fps" default="15"/>
    <arg name="color_fps" default="15"/>
  </include>

MartyG-RealSense, 2020-07-13:
Thanks for the information.  What is the ZED2 being used for in this project, please?  Also, which Librealsense version and ROS wrapper version are you using?

RGring, 2020-07-13:
I use the ROS Wrapper version 2.2.8, librealsense 2.35.2.

MartyG-RealSense, 2020-07-13:
In general you should try to match a particular ROS wrapper version as closely as possible with the SDK version recommended for that wrapper on the wrapper's Releases page.  Wrapper 2.2.8 was intended for use with the very old SDK **2.25.0**.

RGring, 2020-07-13:
Oh true! I overlooked the newer versions. I will try the most recent version (2.2.15) tomorrow and give feedback on the performance later.

RGring, 2020-07-15:
I update to 2.2.14 with librealsense 2.35.2 and the Hardware Error remains.

MartyG-RealSense, 2020-07-15:
Please try inserting this instruction into your roslaunch statement:

 **initial_reset:=true**

RGring, 2020-07-23:
I did that and it reduced the number of messages slightly, but they still remain.

MartyG-RealSense, 2020-07-23:
Though your camera is connected to a USB 3 port, could you check please that it is actually being identified as a USB 3 connection and not a slower USB 2 one, please?

closed 2020-08-08 by MartyG-RealSense