realsenseai/realsense-ros #1280: No "/camera/imu" messages are published after realsense update - D435i

shirbarzel, 2020-07-13:
I updated to:
[ INFO] [1594650833.778359642]: Built with LibRealSense v2.36.0
[ INFO] [1594650833.778443130]: Running with LibRealSense v2.36.0
[ INFO] [1594650834.486209056]: Device FW version: 05.12.06.00

Suddenly after running:

`roslaunch realsense2_camera rs_camera.launch align_depth:=true unite_imu_method:="linear_interpolation"`

But:

`rostopic hz /camera/imu
WARNING: topic [/camera/imu] does not appear to be published yet
`

Thanks

MartyG-RealSense, 2020-07-13:
Hi @shirbarzel   Which RealSense ROS wrapper are you using, please?  A new version **2.2.15** for use with SDK 2.36.0 was published today.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.15

shirbarzel, 2020-07-13:
> Hi @shirbarzel Which RealSense ROS wrapper version are you using, please? A new version **2.2.15** for use with SDK 2.36.0 was published today.
> 
> https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.15

[ INFO] [1594650833.778233626]: RealSense ROS v2.2.15

MartyG-RealSense, 2020-07-13:
Thanks for adding the ROS wrapper version information.  Can you try the IMU statement without the quotation marks please:

**roslaunch realsense2_camera rs_camera.launch  unite_imu_method:=linear_interpolation**

shirbarzel, 2020-07-13:
> Thanks for adding the ROS wrapper version information. Can you try the IMU statement without the quotation marks please:
> 
> **roslaunch realsense2_camera rs_camera.launch align_depth:=true unite_imu_method:=linear_interpolation**

Still not working with the new versions update.

BTW:
I have both versions on my platform, with the old versions everything works fine
(With this combination):
[ INFO] [1594652813.789921614]: RealSense ROS v2.2.13
[ INFO] [1594652813.790044591]: Running with LibRealSense v2.34.1

MartyG-RealSense, 2020-07-13:
I am on the Support team rather than development, so I will tag @doronhi the RealSense ROS wrapper developer into this discussion to seek his input.  I apologise for the wait in the meantime.

doronhi, 2020-07-13:
The imu is disabled by default in the version 2.2.15. There is a comment in the release notes but perhaps it’s note emphasized enough.
Try setting “enable_gyro:=true” and “enable_accel:=true”

shirbarzel, 2020-07-14:
Thanks @doronhi!

closed 2020-07-14 by shirbarzel