realsenseai/realsense-ros #1281: [Possible bug] L515 controls desktop orientation

nkhedekar, 2020-07-14:
On running the `realsense-viewer` and streaming the colored pointcloud of an L515, rotating the L515 sometimes causes the desktop orientation to change. 

Ubuntu: 18.04
librealsense: 2.35.2
firmware: 01.04.01.00

Here is the output of `rs-enumerate-devices`:

```bash
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	00000000f0140727
    Firmware Version              : 	01.04.01.00
    Recommended Firmware Version  : 	01.04.01.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video4
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	L500
    Asic Serial Number            : 	00000003a8a873dd
    Firmware Update Id            : 	00000003a8a873dd

Stream Profiles supported by L500 Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Infrared	  1024x768	@ 30Hz	   Y8
    Infrared	  768x1024	@ 30Hz	   Y8
    Infrared	  640x480	@ 30Hz	   Y8
    Infrared	  480x640	@ 30Hz	   Y8
    Depth	  1024x768	@ 30Hz	   Z16
    Depth	  768x1024	@ 30Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  480x640	@ 30Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1920x1080	@ 30Hz	   RGB8
    Color	  1920x1080	@ 30Hz	   Y16
    Color	  1920x1080	@ 30Hz	   BGRA8
    Color	  1920x1080	@ 30Hz	   RGBA8
    Color	  1920x1080	@ 30Hz	   BGR8
    Color	  1920x1080	@ 30Hz	   YUYV
    Color	  1920x1080	@ 15Hz	   RGB8
    Color	  1920x1080	@ 15Hz	   Y16
    Color	  1920x1080	@ 15Hz	   BGRA8
    Color	  1920x1080	@ 15Hz	   RGBA8
    Color	  1920x1080	@ 15Hz	   BGR8
    Color	  1920x1080	@ 15Hz	   YUYV
    Color	  1920x1080	@ 6Hz	   RGB8
    Color	  1920x1080	@ 6Hz	   Y16
    Color	  1920x1080	@ 6Hz	   BGRA8
    Color	  1920x1080	@ 6Hz	   RGBA8
    Color	  1920x1080	@ 6Hz	   BGR8
    Color	  1920x1080	@ 6Hz	   YUYV
    Color	  1280x720	@ 60Hz	   RGB8
    Color	  1280x720	@ 60Hz	   Y16
    Color	  1280x720	@ 60Hz	   BGRA8
    Color	  1280x720	@ 60Hz	   RGBA8
    Color	  1280x720	@ 60Hz	   BGR8
    Color	  1280x720	@ 60Hz	   YUYV
    Color	  1280x720	@ 30Hz	   RGB8
    Color	  1280x720	@ 30Hz	   Y16
    Color	  1280x720	@ 30Hz	   BGRA8
    Color	  1280x720	@ 30Hz	   RGBA8
    Color	  1280x720	@ 30Hz	   BGR8
    Color	  1280x720	@ 30Hz	   YUYV
    Color	  1280x720	@ 15Hz	   RGB8
    Color	  1280x720	@ 15Hz	   Y16
    Color	  1280x720	@ 15Hz	   BGRA8
    Color	  1280x720	@ 15Hz	   RGBA8
    Color	  1280x720	@ 15Hz	   BGR8
    Color	  1280x720	@ 15Hz	   YUYV
    Color	  1280x720	@ 6Hz	   RGB8
    Color	  1280x720	@ 6Hz	   Y16
    Color	  1280x720	@ 6Hz	   BGRA8
    Color	  1280x720	@ 6Hz	   RGBA8
    Color	  1280x720	@ 6Hz	   BGR8
    Color	  1280x720	@ 6Hz	   YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 400Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 100Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 100Hz	   MOTION_XYZ32F
```

jmiseikis, 2020-07-17:
I experienced the same today. Did you find any workaround or fix?

nkhedekar, 2020-07-20:
Nope, nothing yet

RealSenseSupport, 2020-07-22:
@nkhedekar @jmiseikis Seems it is the issue of Ubuntu itself. You can try "sudo apt-get remove iio-sensor-proxy" to fix that according to https://askubuntu.com/questions/968169/screen-randomly-rotates-on-ubuntu-17-10

nkhedekar, 2020-07-27:
Yep, that seems to be it!

closed 2020-07-27 by nkhedekar