realsenseai/realsense-ros #1282: [Possible bug] HID set_power 1 failed

nkhedekar, 2020-07-14:
There seems to be a sporadic error message on launching `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud` with an L515:

```bash
25/06 16:26:45,953 WARNING [140254882772736] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.7/0003:8086:0B64.0009/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable
```

Since the message seems sporadic, I do not have a method to definitively reproduce it. It may indicate a non-functioning IMU, however, I am able to stream the IMU values.

This issue was originally found while trying to fix #1246.

RealSenseSupport, 2020-12-09:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-12-31 by RealSenseSupport

saralopez7, 2022-09-28:
I'm having the same problem using a D455, ROS2 and launching: `ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=2 publish_odom_tf:=true`. I am using the camera both for IMU and Depth data.

The only way to solve it is to unplug and plug the camera in again. Any idea why this happens and how to solve it?