realsenseai/realsense-ros #1283: L515 can't run

comwise, 2020-07-14:
ubutun 16.04 kernal 

Linux jzrobot-x 4.15.0-15-generic #16~16.04.1-Ubuntu SMP Thu Apr 5 12:19:23 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

ros-kinetic-librealsense2                    2.35.2-2xenial-20200624-230144+0000                         amd64        Library for capturing data from the Intel(R) RealSense(TM) SR300, D400 Depth cameras and T2xx Tracking devices.

ros-kinetic-realsense2-camera                2.2.14-1xenial-20200624-232116+0000                         amd64        RealSense Camera package allowing access to Intel T265 Tracking module and SR300 and D400 3D cameras

ros-kinetic-realsense2-description           2.2.14-1xenial-20200618-130604+0000                         amd64        RealSense Camera description package for Intel 3D D400 cameras

I installed the dependency library by the following way

> export ROS_VER=kinetic 
>or
>export ROS_VER=melodic 
>sudo apt-get install ros-$ROS_VER-realsense2-camera`

But I get the following error

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path:
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename:
 * /camera/realsense2_camera/serial_no:
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method:
 * /camera/realsense2_camera/usb_port_id:
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [23250]
process[camera/realsense2_camera-2]: started with pid [23251]
[ INFO] [1594720112.692218955]: Initializing nodelet with 4 worker threads.
[ INFO] [1594720112.823973491]: RealSense ROS v2.2.14
[ INFO] [1594720112.824174319]: Running with LibRealSense v2.35.2
[ INFO] [1594720112.877733576]:
 14/07 17:48:32,914 ERROR [140546057164544] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 14/07 17:48:32,956 WARNING [140546090735360] (rs.cpp:282) null pointer passed for argument "device"
[ WARN] [1594720112.956931481]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1594720112.957015016]: The requested device with  is NOT found. Will Try again.

> 

i get the usb port by "lusb -t"

/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 4: Dev 2, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 2, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 2, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 2, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 2, If 4, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 2, If 5, Class=Video, Driver=uvcvideo, 5000M

can you tell me why?

RealSenseSupport, 2020-07-14:
Hi,

If you run realsense viewer instead of launching ROS does the camera work?

Thanks

comwise, 2020-07-15:
@RealSenseSupport  yes it‘s ok

RealSenseSupport, 2020-07-15:
In that case I suspect there is something wrong with your installation. I'd recommend completely removing all packages and re-installing librealsense/ros using the instructions here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

and here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

Or you can build ROS from source using the instructions here:

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

This will also get you the latest and more stable versions of ROS/LibRealSense than the method you used.

Thanks

comwise, 2020-07-16:
@RealSenseSupport  i download your sdk， then build your ros source code，but install sdk error

> Loading new librealsense2-dkms-1.3.12 DKMS files...
First Installation: checking all kernels...
Building only for 4.15.0-15-generic
Building initial module for 4.15.0-15-generic
ERROR (dkms apport): unable to determine source package for librealsense2-dkms
Error! Bad return status for module build on kernel: 4.15.0-15-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.12/build/make.log for more information.

Loading the modified modules into kernel... complete

> sdk version：
> librealsense2:amd64                            2.36.0-0~realsense0.3168
> librealsense2-dev:amd64                     2.36.0-0~realsense0.3168
> librealsense2-dkms                             1.3.12-0ubuntu1         
> librealsense2-gl:amd64                        2.36.0-0~realsense0.3168
> librealsense2-net:amd64                      2.36.0-0~realsense0.3168
> librealsense2-udev-rules:amd64           2.36.0-0~realsense0.3168
> librealsense2-utils:amd64                     2.36.0-0~realsense0.3168

I installed the early SDK version which is2.26.0 had no problems（ librealsense2-dkms_1.3.5-0ubuntu1_all.deb）

I find your sdk compatibility is so bad， Now my new camera l515 and d435 doesn't  work ，

can you  fix your compatibility problem quickly ？？！！！**

RealSenseSupport, 2020-07-16:
Hi @comwise 

Can you please provide some details on the exact steps you took from the links above? Also what HW/CPU are you running on?

Thanks

comwise, 2020-07-17:
@RealSenseSupport

cat /proc/cpuinfo
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 142
model name      : Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz
stepping        : 9
microcode       : 0xb4
cpu MHz         : 2500.001
cache size      : 3072 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 2
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 22
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves ibpb ibrs stibp dtherm arat pln pts hwp hwp_notify hwp_act_window hwp_epp
bugs            : cpu_meltdown spectre_v1 spectre_v2
bogomips        : 5424.00
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

DKMS make.log for librealsense2-dkms-1.3.12 for kernel 4.15.0-15-generic (x86_64)
2020年 07月 17日 星期五 11:32:00 CST
make: Entering directory '/usr/src/linux-headers-4.15.0-15-generic'
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.o
/var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.c:219:11: error: ‘V4L2_PIX_FMT_W10’ undeclared here (not in a function)
   .fcc  = V4L2_PIX_FMT_W10,
           ^
/var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.c:224:11: error: ‘V4L2_PIX_FMT_CONFIDENCE_MAP’ undeclared here (not in a function)
   .fcc  = V4L2_PIX_FMT_CONFIDENCE_MAP,
           ^
/var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.c:236:11: error: ‘V4L2_PIX_FMT_RW16’ undeclared here (not in a function)
   .fcc  = V4L2_PIX_FMT_RW16,
           ^
/var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.c:246:11: error: ‘V4L2_PIX_FMT_FG’ undeclared here (not in a function)
   .fcc  = V4L2_PIX_FMT_FG,
           ^
/var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.c:251:11: error: ‘V4L2_PIX_FMT_INZC’ undeclared here (not in a function)
   .fcc  = V4L2_PIX_FMT_INZC,
           ^
/var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.c:256:11: error: ‘V4L2_PIX_FMT_PAIR’ undeclared here (not in a function)
   .fcc  = V4L2_PIX_FMT_PAIR,
           ^
/var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.c:261:11: error: ‘V4L2_PIX_FMT_Z16H’ undeclared here (not in a function)
   .fcc  = V4L2_PIX_FMT_Z16H,
           ^
scripts/Makefile.build:332: recipe for target '/var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.o' failed
make[1]: *** [/var/lib/dkms/librealsense2-dkms/1.3.12/build/4.15.0-55/drivers/media/usb/uvc/uvc_driver.o] Error 1
Makefile:1728: recipe for target 'uvcvideo.ko' failed
make: *** [uvcvideo.ko] Error 2
make: Leaving directory '/usr/src/linux-headers-4.15.0-15-generic'

the kernel source code is dismatch , I want to know if this version(librealsense2-dkms_1.3.5-0ubuntu1_all.deb) matches the current SDK( 2.36.0), I can't always upgrade the kernel to fit the SDK, Because our custom doesn't access to the Internet

comwise, 2020-07-17:
install sdk by the following scripts 

echo -e "${BLUE}--- Processing realsense d435i runtime lib >>>"${NC}
sudo apt install -y libssl-dev \
    libusb-1.0-0-dev \
    pkg-config \
    libgtk-3-dev \
    libglfw3 \
    libglfw3-dev \
    libelf-dev \
    ros-kinetic-cv-bridge \
    ros-kinetic-image-transport \
    ros-kinetic-tf \
    ros-kinetic-diagnostic-updater \
    dkms || { echo -e "${RED}deps install realsense lib fail when runtime_env_install of jrosth${NC}"; exit 7; }
download_rs_pkgs=(
    librealsense2-udev-rules_2.36.0-0~realsense0.3168_amd64.deb
    librealsense2_2.36.0-0~realsense0.3168_amd64.deb
    librealsense2-dkms_1.3.12-0ubuntu1_all.deb
    librealsense2-gl_2.36.0-0~realsense0.3168_amd64.deb
    librealsense2-net_2.36.0-0~realsense0.3168_amd64.deb
    librealsense2-utils_2.36.0-0~realsense0.3168_amd64.deb
    librealsense2-dev_2.36.0-0~realsense0.3168_amd64.deb
    )
for pkg in ${download_rs_pkgs[@]}; do
    [ -f ${debs_dir}/${pkg} ] || ${download_tool} -P ${debs_dir} ${server}/third_parties/all_debs/${pkg}
    sudo dpkg -E -i ${debs_dir}/${pkg}
done

comwise, 2020-07-17:
librealsense2-dkms_1.3.x-0ubuntu1_all.deb

I checked your compile directory which is 

`/var/lib/dkms/librealsense2-dkms/1.3.12/build`

then I found something which had additional directories 4.15.0-55 from 1.3.11

`4.10.0  4.13.0  4.15.0  4.15.0-55  4.4.0  4.8.0  dkms.conf  make.log  patches`

**But i check 1.3.9  that this directory is also available from 1.3.9, but i can build kernel ok**

/var/lib/dkms/librealsense2-dkms/1.3.9/build$

`4.10.0  4.13.0  4.15.0  4.15.0-55  4.4.0  4.8.0  dkms.conf  drivers  patches
`

RealSenseSupport, 2020-08-14:
Hi,

Are you still experiencing issues with this? Have you tried starting fresh as recommended above? You can remove all RealSense™ SDK-related packages with:

dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

and then install the latest SDK as per here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

And finally the latest RealSense-ROS as per here:

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

Thanks

*If we don’t hear from you in 7 days, this issue will be closed.

comwise, 2020-08-17:
@RealSenseSupport your response is so quickly, Every time I didn't get the answer I wanted, your answers is so bad,  You can take customer problems seriously, such company replies really like shit

closed 2020-09-01 by RealSenseSupport