realsenseai/realsense-ros #1284: Intel Realsense: Time is out of Dual 32-bit range

jacky1089, 2020-07-14:
We are trying to get the Intel Realsense D435i to work on our RaspberryPi4  with the Raspbian OS and ROS Melodic.

After we configured our Raspberry Pi with Raspbian and installed ROS Melodic on it, we installed the realsense-ros package on our Raspberry Pi. When we connect our Realsense camera to the Raspberry and run the following command:

$ roslaunch realsense2_camera rs_camera.launch

We get the following error: error: Time is out of Dual 32-bit range

Is there anybody who has any ideas?

MartyG-RealSense, 2020-07-14:
@doronhi the RealSense ROS wrapper developed investigated this error on a previous occasion and concluded that it likely was not related to the RealSense ROS wrapper.  

https://github.com/IntelRealSense/realsense-ros/issues/539

A few approaches to resolving the problem are suggested in the link below.  For example, a RealSense user with a Pi 4 commented that their solution was to switch from Raspbian Buster to Ubuntu 18.04 and use a stable power connection.

https://github.com/IntelRealSense/realsense-ros/issues/1178

jacky1089, 2020-07-14:
We can't replace the operating system because of the problem with the realsense sensor . @MartyG-RealSense 

I looked at the code and found "enable sync=true," and the error disappeared. 

        if (_sync_frames)
        {
            t = ros::Time::now();
        }
        else
        {
            t = ros::Time(_ros_time_base.toSec()+ (/*ms*/ frame_time - /*ms*/ _camera_time_base) / /*ms to seconds*/ 1000);
        }

 The variable frame_time is too large to cause an error

MartyG-RealSense, 2020-07-14:
Can you confirm whether you still have problems, please?   Thanks!

jacky1089, 2020-07-14:
There are still mistakes. 
 After the installation is completed, run the default parameters and run roslaunch on rpi4 without making any settings. This error will appear.

MartyG-RealSense, 2020-07-15:
It is difficult to advise on a solution if the source of the problem is outside of RealSense.

Other than doing a fresh rebuild or changing OS, another suggestion made was to check whether the system clock is correct.  Can you check the clock on your Pi please and see if the time and date are correct?

closed 2020-08-02 by MartyG-RealSense