realsenseai/realsense-ros #1287: Jetson Xavier AGX/NX will always fail with d435, why?

AndreV84, 2020-07-14:
steps to reproduce
`docker run -it --rm --net=host -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE iad.ocir.io/idso6d7wodhe/jetson_nx/librealsense1ros /bin/bash -c '. /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch realsense2_camera rs_camera.launch'`
reference thread : https://forums.developer.nvidia.com/t/realsensed400-camera-can-not-work-on-jetson-xavier-nx/125138/18
reference Docker file:

`FROM nvcr.io/nvidia/l4t-base:r32.4.2
RUN apt-get update -y
RUN apt install -y wget software-properties-common apt-utils  curl git
RUN /bin/bash  -c 'echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list'
RUN  apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key
RUN apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate 
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-melodic-desktop clang-format ros-melodic-desktop-full python-pip python3-pip  python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential 
RUN apt update -y
#RUN rm -rf /var/lib/apt/lists
RUN  apt-get install -y librealsense2-dev librealsense2-utils
RUN apt install -y ros-melodic-ddynamic-reconfigure
RUN dpkg-reconfigure --frontend noninteractive tzdata
RUN rm -rf /var/lib/apt/lists
WORKDIR /home/nvidia/ws
RUN mkdir src && cd src &&  git clone https://github.com/intel-ros/realsense.git -b development
RUN rosdep init && rosdep update

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash;  cd /home/nvidia/ws; rosdep install --from-paths src --ignore-src; catkin_make'`

AndreV84, 2020-07-14:
will l515 work with it?
t265 seem worked;
the same steps on x86_64 worked also with d400

AndreV84, 2020-07-14:
INFO] [1594754681.293070147]: insert Depth to L500 Depth Sensor
 14/07 12:24:41,295 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 14/07 12:24:41,451 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1594754681.563714222]: SELECTED BASE:Depth, 0
[ INFO] [1594754681.564919130]: RealSense Node Is Up!
 14/07 12:24:41,764 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/07 12:24:41,814 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/07 12:24:41,865 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/07 12:24:41,915 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 14/07 12:24:41,966 WARNING [547063062912] (messenger-libusb.cpp:42) control_transfer returned error, inde

RealSenseSupport, 2020-07-15:
Hi @AndreV84 
Can you help clear a bit of confusion?  Is this thread a L515 based issue or D435 issue?  I'm reading the title and D435 is there which may be causing our confusion.

MartyG-RealSense, 2020-07-16:
Hi @RealSenseCustomerSupport   This was originally a D435 discussion that made reference in one of its comments to L515 and T265 (as @AndreV84 also has an L515 and T265 - they are not having problems with the T265 though).  It has now been split into two separate cases to make it easier to handle: D435 only on this one, and L515 only on the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1289

AndreV84, 2020-07-17:
there are issues implementing ros2-foxy-librealsense2

instruction https://github.com/intel/ros2_intel_realsense/tree/refactor#install-ros2_intel_realsense-from-source won't work

command sudo apt-get install ros-foxy-librealsense2 wouldn't find apt package 

building from sources would return multiple errors, e.g.

CMake Error at /opt/ros/foxy/share/std_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
/opt/ros/foxy/share/std_msgs/cmake/std_msgsConfig.cmake:41 (include)
CMakeLists.txt:20 (find_package)

doronhi, 2020-07-19:
The "control_transfer returned error" message in itself does not predict a real error. I see in your reference thread : https://forums.developer.nvidia.com/t/realsensed400-camera-can-not-work-on-jetson-xavier-nx/125138/18 that the issue is that the camera is not found in following runs. Is that true? If so it would be very helpful to state clearly somewhere, maybe even in the topic.
I don't have news about it yet. I know some work has been done on librealsense2's RSUSB backend and you should check out the latest version of it. As to errors that might be caused by the ROS wrapper itself, work is still need to be done. See also here: #1171 

As for the foxy version, it is not maintained in **this** repository. You should report the issue in [ros2_intel_realsense repository]( https://github.com/intel/ros2_intel_realsense) although I don't think foxy is supported by the wrapper, as ubuntu 20.04 is not yet officially supported by librealsense2.

AndreV84, 2020-07-19:
@doronhi Thank you for your response.

- the camera is always found in all runs, as it seems to me
What made you suppose that the camera was not found?

- they mentioned multiple times that there is released librealsense2 package for foxy
e.g. https://github.com/ros2-gbp/librealsense-release#librealsense-foxy---2340-1
Created the issue here:
https://github.com/intel/ros2_intel_realsense/issues/143
May I know what do you mean stating that 20.04 ubuntu is not officially supported by intel?
from sources, the librealsense sdk builds and works. Moreover, there are explicit instructions for foxy 20.04 installation :+1: https://github.com/intel/ros2_intel_realsense/tree/refactor#install-ros2_intel_realsense-from-source

doronhi, 2020-07-19:
Well, I guess I misunderstood your post [here](https://forums.developer.nvidia.com/t/realsensed400-camera-can-not-work-on-jetson-xavier-nx/125138/25) and it reminded me of other issues. If it is still relevant to this repository, could you clarify what the problem is then?

Librealsense2 is not officially supported for Ubuntu 20.04 as you can see in the list of supported systems of the [latest release](https://github.com/IntelRealSense/librealsense/releases/tag/v2.36.0). I heard it builds well on Ubuntu 20.04 but it may still lack some kernel patches and it is not officially tested by Intel. That's why I didn't suppose Intel officially supported a wrapper in an environment not officially supported for the library.
Other then that, I also noticed that librealsense2 was removed for the moment from the foxy and dashing repositories of ROS2 (https://github.com/ros/rosdistro/pull/25814).
All that may account to your current difficulties but in no way means that it couldn't work. I think you did correctly opening the issue at [intel/ros2_intel_realsense#143](https://github.com/intel/ros2_intel_realsense/issues/143).

AndreV84, 2020-07-19:
@doronhi Thank you for following up,
Could you help out resolving issues that emerge while implementing d400 ros wrapper on ubuntu 18.04, using armv8l & ros1 environment, also containerized?
The only observable issue is after the Node starts it throws errors continuously:
```
08/06 00:33:23,125 ERROR [546442297728] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
08/06 00:33:23,144 ERROR [546903667072] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
08/06 00:33:23,158 ERROR [546442297728] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
```
errors like that
errors like that
```
7 17:20:16,664 WARNING [547364323712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1594416016.768815959]: SELECTED BASE:Depth, 0
[ INFO] [1594416016.790701784]: RealSense Node Is Up!
10/07 17:20:16,792 WARNING [547364323712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
10/07 17:20:16,842 WARNING [547364323712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
10/07 17:20:16,893 WARNING [547364323712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
10/07 17:20:17,000 WARNING [547364323712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
10/07 17:20:17,051 WARNING [547364323712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
10/07 17:20:17,102 WARNING [547364323712] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

AndreV84, 2020-07-19:
Another issue:
I am trying  to do calibration steps under linux using the official instruction steps from
https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide
It won't work.
```
sudo apt-get install librscalibrationtool
Unable to locate package librscalibrationtool
```

AndreV84, 2020-07-19:
ubuntu 18.04 is not supported? supported but not for arm?

doronhi, 2020-07-20:
It will help if you split your questions to several issues. That way it will be clearer and different people could respond to different questions.

As for the "control_transfer returned error" message, you can see my answer [here](https://github.com/IntelRealSense/realsense-ros/issues/1292#issuecomment-660617207).

As for "Exception was thrown during user processing callback", this is a general question. Are you running realsense2_camera wrapper? if so it may help to have a full log up until the messages start appearing. Otherwise you should consult the [librealsense2's issues](https://github.com/IntelRealSense/librealsense/issues)

> ubuntu 18.04 is not supported? supported but not for arm?

Supported by whom? librealsense2 supports ubuntu18.04 and arm.

AndreV84, 2020-07-20:
Hello @doronhi ,
Thank you for following up!
I already dissected the issue cutting off all information about other models of realsense by request of the support guys.
Only D400 concern is being published at this particular github issue thread. Other issues are not addressed at all at other threads, I believe.
When you state that "librealsense2 supports ubuntu18.04 and arm." I understand that the answer doesn't address to any extent the issue of absense of the package "librscalibrationtool" in ubuntu 18.04 apt repositories. Did I get you right? Are you able to address this particular issue?
Are you able to address the issue of running d400 cameras within arm ubuntu 18.04 + ROS1 using ros wrapper?
The full log has been attached before, in parts, and completely via the [devtalk forum thread](https://forums.developer.nvidia.com/t/realsensed400-camera-can-not-work-on-jetson-xavier-nx/125138/25). The latter has full log of the starting command up to the emergence of the errors.
What solution do you propose?
post more logs?
anything else?

AndreV84, 2020-07-20:
Just to make it clear.
Steps to reproduce the issue:
1. build the dockerfile attached below
2. run the container with ros_wrapper arguments
Dockerfile
```
FROM nvcr.io/nvidia/l4t-base:r32.4.3
RUN apt-get update -y
RUN apt install -y wget software-properties-common apt-utils  curl git
RUN /bin/bash  -c 'echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list'
RUN  apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key
RUN apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate 
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-melodic-desktop clang-format ros-melodic-desktop-full python-pip python3-pip  python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential 
RUN apt update -y
#RUN rm -rf /var/lib/apt/lists
RUN  apt-get install -y librealsense2-dev librealsense2-utils
RUN apt install -y ros-melodic-ddynamic-reconfigure
RUN dpkg-reconfigure --frontend noninteractive tzdata
RUN rm -rf /var/lib/apt/lists
WORKDIR /home/nvidia/ws
RUN mkdir src && cd src &&  git clone https://github.com/intel-ros/realsense.git -b development
#  && git clone https://github.com/slamtec/rplidar_ros

RUN rosdep init && rosdep update

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash;  cd /home/nvidia/ws; rosdep install --from-paths src --ignore-src; catkin_make'
```
docker build . -t docker build . -t test_librealsense
docker run -it --rm --net=host -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE test_librealsense /bin/bash -c ‘. /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch realsense2_camera rs_camera.launch’

otherwise less updated version can be executed with one single command
`docker run -it --rm --net=host -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE iad.ocir.io/idso6d7wodhe/jetson_nx/librealsense1ros /bin/bash -c ‘. /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch realsense2_camera rs_camera.launch’
`

AndreV84, 2020-07-20:
now list of errors and discrepancies;
minor:
```
E: Unable to locate package ros-melodic-librealsense2
ERROR: the following rosdeps failed to install
  apt: command [apt-get install ros-melodic-librealsense2] failed
```
major:
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at realsense/realsense2_camera/CMakeLists.txt:37 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.36.0".

  The following configuration files were considered but not accepted:

    /usr/lib/aarch64-linux-gnu/cmake/realsense2/realsense2Config.cmake, version: 2.35.2

CMake Error at realsense/realsense2_camera/CMakeLists.txt:39 (message):
  

  

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

AndreV84, 2020-07-20:
so I am doing as per the error suggestion
wget https://github.com/IntelRealSense/librealsense/archive/v2.36.0.tar.gz 
tar -xvf v2.36.0.tar.gz
cd librealsense-2.36.0/
further I am trying to find the source code installation instructions in order to get the thing installed on arm device; The following reference seems to point out to an absent file
```
#### Configuring and building from the source code
While we strongly recommend to use DKMS package whenever possible, there are certain cases where installing and patching the system manually is necessary:
 - Using SDK with non-LTS Ubuntu kernel versions: **4.16 and 5.0**
 - Integration of user-specific patches/modules with `librealsense` SDK.
 - Adjusting the patches for alternative kernels/distributions.
there is no installation.md

The steps are described in [Linux manual installation guide](./installation.md)
```

AndreV84, 2020-07-20:
so I am just doing` mkdir build && cd build && cmake ..`
then make -j8 && make install
The updated Dockerfile:
`FROM nvcr.io/nvidia/l4t-base:r32.4.3
RUN apt-get update -y
RUN apt install -y wget software-properties-common apt-utils  curl git
RUN /bin/bash  -c 'echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list'
RUN  apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key
RUN apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate 
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-melodic-desktop clang-format ros-melodic-desktop-full python-pip python3-pip  python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential 
RUN apt update -y
#RUN rm -rf /var/lib/apt/lists
RUN  apt-get install -y librealsense2-dev librealsense2-utils
RUN apt install -y ros-melodic-ddynamic-reconfigure
RUN dpkg-reconfigure --frontend noninteractive tzdata
RUN rm -rf /var/lib/apt/lists
WORKDIR /home/nvidia/ws
RUN wget https://github.com/IntelRealSense/librealsense/archive/v2.36.0.tar.gz && tar -xvf v2.36.0.tar.gz
RUN cd librealsense-2.36.0 && ls && mkdir build && cd build && cmake .. && make -j8 && make install
RUN mkdir src && cd src &&  git clone https://github.com/intel-ros/realsense.git -b development
#  && git clone https://github.com/slamtec/rplidar_ros

RUN rosdep init && rosdep update

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash;  cd /home/nvidia/ws; rosdep install --from-paths src --ignore-src; catkin_make'`

Minor issues:
[ 67%] Linking CXX executable rs-motion
[ 67%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/upload.cpp.o
In file included from /usr/include/c++/7/backward/strstream:50:0,
                 from /home/nvidia/ws/librealsense-2.36.0/src/gl/upload.cpp:24:
/usr/include/c++/7/backward/backward_warning.h:32:2: warning: #warning This file includes at least one deprecated or antiquated header which may be removed without further notice at a future date. Please use a non-deprecated interface with equivalent functionality instead. For a listing of replacement headers and interfaces, consult the file backward_warning.h. To disable this warning use -Wno-deprecated. [-Wcpp]
 #warning \
  ^~~~~~~
[ 67%] Built target rs-motion
[ 71%] Linking CXX executable rs-pose-and-image
In file included from /home/nvidia/ws/librealsense-2.36.0/src/gl/camera-shader.cpp:17:0:
/home/nvidia/ws/librealsense-2.36.0/src/gl/../../common/res/sr300.h: In function 'void uncompress_sr305_obj(std::vector<rs2::float3>&, std::vector<rs2::float3>&, std::vector<short3>&)':
/home/nvidia/ws/librealsense-2.36.0/src/gl/../../common/res/sr300.h:11:77: warning: 'int LZ4_uncompress(const char*, char*, int)' is deprecated: use LZ4_decompress_fast() instead [-Wdeprecated-declarations]
     LZ4_uncompress((const char*)sr305_obj_data, uncompressed.data(), 0x43c5c);
                                                                             ^
In file included from /home/nvidia/ws/librealsense-2.36.0/src/gl/../../common/res/d435.h:6:0,
                 from /home/nvidia/ws/librealsense-2.36.0/src/gl/camera-shader.cpp:15:
/home/nvidia/ws/librealsense-2.36.0/third-party/realsense-file/lz4/lz4.h:446:68: note: declared here
 LZ4LIB_API LZ4_DEPRECATED("use LZ4_decompress_fast() instead") int LZ4_uncompress (const char* source, char* dest, int outputSize);
                                                                    ^~~~~~~~~~~~~~
In file included from /home/nvidia/ws/librealsense-2.36.0/src/gl/camera-shader.cpp:19:0:
/home/nvidia/ws/librealsense-2.36.0/src/gl/../../common/res/l500.h: In function 'void uncompress_L515_obj(std::vector<rs2::float3>&, std::vector<rs2::float3>&, std::vector<short3>&)':
/home/nvidia/ws/librealsense-2.36.0/src/gl/../../common/res/l500.h:11:76: warning: 'int LZ4_uncompress(const char*, char*, int)' is deprecated: use LZ4_decompress_fast() instead [-Wdeprecated-declarations]
     LZ4_uncompress((const char*)L515_obj_data, uncompressed.data(), 0x49e1e);
                                                                            ^
In file included from /home/nvidia/ws/librealsense-2.36.0/src/gl/../../common/res/d435.h:6:0,
                 from /home/nvidia/ws/librealsense-2.36.0/src/gl/camera-shader.cpp:15:
/home/nvidia/ws/librealsense-2.36.0/third-party/realsense-file/lz4/lz4.h:446:68: note: declared here
 LZ4LIB_API LZ4_DEPRECATED("use LZ4_decompress_fast() instead") int LZ4_uncompress (const char* source, char* dest, int outputSize);
                                                                    ^~~~~~~~~~~~~~
[ 71%] Built target rs-pose-and-image
[ 71%] Building CXX object src/gl/CMakeFiles/realsense2-gl.dir/colorizer-gl.cpp.o
In file included from /usr/include/c++/7/backward/strstream:50:0,
                 from /home/nvidia/ws/librealsense-2.36.0/src/gl/colorizer-gl.cpp:22:
[... cut ...]
/home/nvidia/ws/librealsense-2.36.0/third-party/realsense-file/boost/boost/function/function_base.hpp:318:13: warning: placement new constructing an object of type 'boost::detail::function::functor_manager_common<rosbag::TopicQuery>::functor_type {aka rosbag::TopicQuery}' and size '24' in a region of type 'char' and size '1' [-Wplacement-new=]
             new (reinterpret_cast<void*>(&out_buffer.data)) functor_type(*in_functor);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 83%] Linking CXX executable rs-benchmark

AndreV84, 2020-07-20:
from one non-wroking state it derived into another non-working state
```
docker run -it --rm --net=host -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE test_librealsense /bin/bash -c '. /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch realsense2_camera rs_camera.launch'
... logging to /root/.ros/log/0667ce6a-ca9d-11ea-9106-00044bcb9a37/roslaunch-jetson-agx001-54.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-agx001:42627/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 23 lines cut ...]
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [64]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0667ce6a-ca9d-11ea-9106-00044bcb9a37
process[rosout-1]: started with pid [75]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [82]
[ INFO] [1595258638.621448441]: Initializing nodelet with 8 worker threads.
process[camera/realsense2_camera-3]: started with pid [83]
[ INFO] [1595258639.586602669]: RealSense ROS v2.2.15
[ INFO] [1595258639.586750130]: Built with LibRealSense v2.36.0
[ INFO] [1595258639.586806868]: Running with LibRealSense v2.36.0
 20/07 08:23:59,607 WARNING [547301892480] (ds5-factory.cpp:1119) DS5 group_devices is empty.
[ WARN] [1595258639.608017894]: No RealSense devices were found!
 20/07 08:24:05,811 WARNING [547301892480] (ds5-factory.cpp:1119) DS5 group_devices is empty.
[ WARN] [1595258645.811834510]: No RealSense devices were found!
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit

```

AndreV84, 2020-07-20:
the issue is not new to me; I had to get it previously when building the librealsense-sdk from sources; It was mitigated with using the apt repository provided librealsense that you dropped from being published, as I understand

AndreV84, 2020-07-21:
with ROS2 the situation is different; librealsense builds and works from within docker container; ros wrapper fails;
```
FROM osrf/ros2:nightly-rmw
ENV RTI_NC_LICENSE_ACCEPTED yes

# bootstrap rosdep
RUN rosdep update

# install dependencies
RUN . /opt/ros/$ROS_DISTRO/setup.sh \
    && apt-get update \
    && rosdep install -y \
    --from-paths /opt/ros/$ROS_DISTRO/share \
    --ignore-src \
    --skip-keys " \
      " \
    && rm -rf /var/lib/apt/lists/*

# set up environment
ENV NDDSHOME /opt/rti.com/rti_connext_dds-5.3.1
ENV PATH "$NDDSHOME/bin":$PATH
ENV LD_LIBRARY_PATH "$NDDSHOME/lib/x64Linux3gcc5.4.0":$LD_LIBRARY_PATH
# bootstrap
#RUN  apt-get install -y ros-foxy-librealsense2

RUN apt-get update -y
RUN apt install -y wget curl git
RUN apt-get update -y
#RUN locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 && LANG=en_US.UTF-8
RUN apt update && apt install curl gnupg2 lsb-release
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc |  apt-key add - && sh -c 'echo "deb http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2>
RUN apt update -y
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-foxy-ros-base
RUN apt update -y
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-foxy-cv-bridge  ros-foxy-message-filters ros-foxy-image-transport
RUN DEBIAN_FRONTEND="noninteractive" apt install -y  libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
#RUN DEBIAN_FRONTEND="noninteractive" apt install -y  ros-foxy-realsense-camera-msgs ros-foxy-realsense-ros2-camera
RUN dpkg-reconfigure --frontend noninteractive tzdata
#RUN rm -rf /var/lib/apt/lists
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-foxy-desktop
WORKDIR /home/nvidia/ros2_ws
RUN  mkdir -p ~/src && cd ~/src && git clone https://github.com/intel/ros2_intel_realsense.git && cd ros2_intel_realsense && git checkout refactor
RUN  sh -c 'echo "deb [arch=amd64,arm64] http://repo.ros2.org/ubuntu/main `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc |  apt-key add -
RUN apt-get update -y
#RUN  apt-get install -y ros-foxy-librealsense2
RUN git clone https://github.com/IntelRealSense/librealsense.git && cd librealsense && mkdir build && cd build &&  cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=re>
RUN apt install -y python3-colcon-common-extensions
#RUN cd ~ &&  /bin/bash -c '. /opt/ros/foxy/setup.bash; colcon build --symlink-install --base-paths src/ros2_intel_realsense'
```

doronhi, 2020-07-21:
Thanks for your detailed log files. 
I can't comment on the ROS2 issue as I'm not familiar with it (and also I don't see the error message..)
Regarding the ROS1, the message "DS5 group_devices is empty" implies trying to use D435**i** with v4l2 but without the proper kernel patches. If you are indeed using a D435i device, I suggest you try building librealsense2 with RSUSB backend. That is, run `cmake ../ -DFORCE_RSUSB_BACKEND=true` and rebuild librealsense2 and realsense2_camera.
If you are using D435 and not D435i I can only speculate for now that the camera is not reflected correctly to the docker..

AndreV84, 2020-07-21:
I specifically connected d435 as of the two it is te only supported by nvidia isaac;
with ROS1 the issue is with the librealsense sdk built from sources; it won't see the camera; The issue is known; The workaround was to use apt repository build with librealsense sdk. It would recognize the cameras; A week ago the apt distribution started to throw issues that it hasn't had before. [ I have somewhere older versions where librealsense is installed into the docker from apt and recognizes devicers]. On the other hand, dockerized or not dockerized librealsense sdk built from sources as I recollect wouldn't see any device at arm. I can confirm it by trials, but it won't help you to provide a solution against it?
I also have the d435i that I can connect instead or in addition, but it probably won't help either?
With ROS2 the situation is that errors are shown on attemp to build with colcon --symlink;
I used to build for ROS2 with following arguments 
 `cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true && make -j4 && make install`

Errors are:
root@3ef304e9c37f:/home/nvidia/ros2_ws#  colcon build --symlink-install
Starting >>> librealsense2
Starting >>> realsense_msgs
--- stderr: realsense_msgs                                        
CMake Error at /opt/ros/foxy/share/rosidl_generator_c/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /opt/ros/foxy/share/rosidl_generator_c/cmake/rosidl_generator_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:11 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)

CMake Error at /opt/ros/foxy/share/rosidl_generator_c/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /opt/ros/foxy/share/rosidl_generator_c/cmake/rosidl_generator_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:11 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)

CMake Error at /opt/ros/foxy/share/rosidl_typesupport_connext_c/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /opt/ros/foxy/share/rosidl_typesupport_connext_c/cmake/rosidl_typesupport_connext_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)

CMake Error at /opt/ros/foxy/share/rosidl_typesupport_connext_c/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /opt/ros/foxy/share/rosidl_typesupport_connext_c/cmake/rosidl_typesupport_connext_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)

CMake Error at /opt/ros/foxy/share/rmw/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /opt/ros/foxy/share/rmw/cmake/rmwConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_typesupport_connext_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_connext_c/cmake/rosidl_typesupport_connext_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)

[... cut ...]
Call Stack (most recent call first):
  /opt/ros/foxy/share/std_msgs/cmake/std_msgsConfig.cmake:41 (include)
  CMakeLists.txt:20 (find_package)

---
Failed   <<< realsense_msgs [8.58s, exited with code 1]
Aborted  <<< librealsense2 [9.78s]                            

Summary: 0 packages finished [10.3s]
  1 package failed: realsense_msgs
  1 package aborted: librealsense2
  2 packages had stderr output: librealsense2 realsense_msgs
  3 packages not processed
root@3ef304e9c37f:/home/nvidia/ros2_ws#

AndreV84, 2020-07-21:
system wide [ not docker] arm; d435 connected [ not i]
 `git clone https://github.com/IntelRealSense/librealsense.git && cd librealsense && mkdir build && cd build &&  cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true && make -j8`
with the same arguments tried the docker implementation;
the camera seems detected; the full log is
```
docker run -it --rm --net=host -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE test_librealsense:latest /bin/bash -c '. /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch realsense2_camera rs_camera.launch'
... logging to /root/.ros/log/9f489d42-cb3a-11ea-b5cc-00044bcb9a37/roslaunch-jetson-agx001-54.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-agx001:39823/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 63 lines cut ...]
[ INFO] [1595326329.121856355]: RGB Camera was found.
[ INFO] [1595326329.122181040]: num_filters: 0
[ INFO] [1595326329.123624010]: Setting Dynamic reconfig parameters.
[ INFO] [1595326336.511473733]: Done Setting Dynamic reconfig parameters.
[ INFO] [1595326336.520806653]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1595326336.522562276]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1595326336.523612783]: setupPublishers...
[ INFO] [1595326336.538366658]: Expected frequency for depth = 30.00000
[ INFO] [1595326336.680924765]: Expected frequency for color = 30.00000
[ INFO] [1595326336.745698196]: setupStreams...
[ INFO] [1595326336.873165678]: insert Depth to Stereo Module
[ INFO] [1595326336.873334997]: insert Color to RGB Camera
 21/07 03:12:17,043 WARNING [547507646848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1595326337.267299316]: SELECTED BASE:Depth, 0
[ INFO] [1595326337.321760298]: RealSense Node Is Up!
 21/07 03:12:17,363 WARNING [547507646848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 21/07 03:12:17,802 WARNING [547507646848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/07 03:12:17,984 WARNING [547507646848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/07 03:12:18,287 WARNING [547507646848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/07 03:12:18,593 WARNING [547507646848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

```

AndreV84, 2020-07-21:
are you able to help out with the particular error below? on arm? within or outside of docker?  Neither? Both?
`(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61`

AndreV84, 2020-07-21:
is it considered a working state?

doronhi, 2020-07-21:
> is it considered a working state?

As mentioned, for now, if this message appears several times at the beginning and not continuously during the run, it is considered a working state. Do you see any issues with the streams?
As for the attempts to build ROS2, I can't help.

AndreV84, 2020-07-21:
Thank you for following up!
I shall figure out how to see streams at all as I am accessing the setup remotely.
I shall update you soon

AndreV84, 2020-07-21:
how do I calibrate the camera? remotely?
for x86_64 there are packages in apt repository; for arm there are not;

AndreV84, 2020-07-21:
how do I run self calibration for 435?
www.intelrealsense.com/self-calibration-for-depth-cameras/amp/
it seems different from https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

MartyG-RealSense, 2020-07-21:
@AndreV84  You can run self-calibration by going to the **More** option at the top of the RealSense Viewer's options side-panel and selecting **On-Chip Calibration** from the 'More' option's drop-down menu.

Intel have published a white-paper guide to self calibration.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

AndreV84, 2020-07-24:
could you comment on occupancy package, please?
Dockerfile
```
FROM nvcr.io/nvidia/l4t-base:r32.4.3
RUN apt-get update -y
RUN apt install -y wget software-properties-common apt-utils  curl git
RUN /bin/bash  -c 'echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list'
RUN  apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key
RUN apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate 
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-melodic-desktop clang-format ros-melodic-desktop-full python-pip python3-pip  python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential 
RUN apt update -y
#RUN rm -rf /var/lib/apt/lists
RUN  apt-get install -y librealsense2-dev librealsense2-utils
RUN apt install -y ros-melodic-ddynamic-reconfigure
RUN dpkg-reconfigure --frontend noninteractive tzdata
RUN rm -rf /var/lib/apt/lists
WORKDIR /home/nvidia/ws
RUN wget https://github.com/IntelRealSense/librealsense/archive/v2.36.0.tar.gz && tar -xvf v2.36.0.tar.gz
RUN cd librealsense-2.36.0 && ls && mkdir build && cd build && cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true && make -j8 && make install
RUN mkdir src && cd src &&  git clone https://github.com/intel-ros/realsense.git -b development 
#  && git clone https://github.com/slamtec/rplidar_ros
COPY realsense-ros/occupancy src/occupancy
RUN rosdep init && rosdep update

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash;  cd /home/nvidia/ws; rosdep install --from-paths src --ignore-src; catkin_make'
```

Full error
```
docker run -it --rm --net=host -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE occupancy:latest /bin/bash -c '. /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch occupancy occupancy_live_rviz.launch'
... logging to /root/.ros/log/8944c85e-cdba-11ea-9d76-00044bcb9a37/roslaunch-jetson-agx001-54.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-agx001:44051/

SUMMARY
========

PARAMETERS
 * /DepthOfInterestMax: 2.0
 * /DepthOfInterestMin: 0.2
 * /HeightOfInterestMax: 0.05
 * /HeightOfInterestMin: -0.05
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/depth_fps: 30
[... 246 lines cut ...]
[ INFO] [1595601182.256447687]: Device with physical ID 1-4-10 was found.
[ INFO] [1595601182.256505737]: Device with name Intel RealSense D435I was found.
[ INFO] [1595601182.257573136]: Device with port number 1-4 was found.
[ INFO] [1595601182.561576496]: Device with serial number  was found.

[ INFO] [1595601182.561728149]: Device with physical ID 2-1-2 was found.
[ INFO] [1595601182.561792727]: Device with name Intel RealSense D435 was found.
[ INFO] [1595601182.562865855]: Device with port number 2-1 was found.
[ INFO] [1595601182.597407208]: Device with serial number  was found.

[ INFO] [1595601182.597545902]: Device with physical ID 2-3-7 was found.
[ INFO] [1595601182.598060512]: Device with name Intel RealSense T265 was found.
[ INFO] [1595601182.599275565]: Device with port number 2-3 was found.
[ERROR] [1595601182.694879718]: The requested device with serial number T265_SN and device name containing t265 is NOT found. Will Try again.
 24/07 07:33:07,957 WARNING [547205673344] (types.cpp:78) set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
 24/07 07:33:07,957 ERROR [547205673344] (ds5-factory.cpp:823) RGB Extrinsic recovery routine failed
 24/07 07:33:07,978 WARNING [547205673344] (types.cpp:78) hwmon command 0x15( 20 0 0 0 ) failed (response 0= Success)
 24/07 07:33:07,978 WARNING [547205673344] (ds5-factory.cpp:669) Cannot read RGB calibration table
 24/07 07:33:07,978 WARNING [547205673344] (types.cpp:78) Calibration data invald, buffer too small : expected 16 , actual: 0
[ INFO] [1595601187.978966725]: Device with serial number  was found.

[ INFO] [1595601187.979229998]: Device with physical ID 1-4-10 was found.
[ INFO] [1595601187.979398644]: Device with name Intel RealSense D435I was found.

```

MartyG-RealSense, 2020-07-24:
@AndreV84   Is your question about the occupancy package directed at @doronhi please?

AndreV84, 2020-07-24:
@doronhi could you address the ROS1 occupancy package issue, please?
I am just also wondering what is a proper way to iementthe sample and upload the calibration file https://github.com/IntelRealSense/librealsense/blob/development/examples/tracking-and-depth/H_t265_d400.cfg . Is the sample one of these that are built by defalt with librealsense sdk? And the calibration file? Is it loaded by the sample automatically or will need to be loaded via realsense viewer? to one camera? two cameras?

AndreV84, 2020-07-24:
@doronhi 
with the same dockerfile but with added rtmabmap-ros package I am trying to get  the following comand
 docker run -it --rm --net=host -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE rtabmap:latest /bin/bash -c '. /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch realsense2_camera rs_rtabmap.launch
'

AndreV84, 2020-07-24:
it seems to get into a similar loop with the rtabmap as it does with occupancy

AndreV84, 2020-07-24:
rtabmap seem worked
occyuancy seems to have device detection loop

AndreV84, 2020-07-24:
[image]
what can be done with depth-tracking example that won't run on x86_64 with librealsense sdk installed from apt repository?

AndreV84, 2020-07-24:
built newer sdk from sources and installed latest firmware to cameras
 ```
./rs-tracking-and-depth 
2020-07-24 16:53:10,003 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:10,009 WARNING [default] Could not open device Unable to open device interface
2020-07-24 16:53:10,060 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:10,071 WARNING [default] Could not open device Unable to open device interface
2020-07-24 16:53:15,050 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:15,062 WARNING [default] Could not open device Unable to open device interface
2020-07-24 16:53:15,112 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:15,120 WARNING [default] Could not open device Unable to open device interface
2020-07-24 16:53:15,159 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:15,169 WARNING [default] Could not open device Unable to open device interface
```

MartyG-RealSense, 2020-07-24:
@AndreV84   Please post one question at a time and wait for a reply, otherwise it makes the discussion very hard to follow.  Thank you.

AndreV84, 2020-07-24:
Couldn't open ./H_t265_d400.cfg
but I am approaching the issue and the state changes;
with powered usb hubb the example would throw the erorr above

MartyG-RealSense, 2020-07-24:
Okay, now that I see you have mentioned that you are using a T265 in today's questions I will have to leave the discussion unfortunately, as someone else is responsible for supporting T265 related cases and I cannot work on them.  I do hope that you get the answers that you are seeking.

AndreV84, 2020-07-24:
yes, thank you for following up!
with d400 the issue of using ros wrapper on ROS1 has been sorted; the next objective was to run both t265&400

AndreV84, 2020-07-24:
shall I raise a separate thread for the latter concern?

MartyG-RealSense, 2020-07-24:
Yes, it would be great if you could close this case if you are satisfied with the ROS1 outcome and start a new case just for T265 and 400 together.  Thanks!

AndreV84, 2020-07-29:
the one that worked for ROS1 D400
FROM nvcr.io/nvidia/l4t-base:r32.4.3
RUN apt-get update -y
RUN apt install -y wget software-properties-common apt-utils  curl git
RUN /bin/bash  -c 'echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list'
RUN  apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key
RUN apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate 
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-melodic-desktop clang-format ros-melodic-desktop-full python-pip python3-pip  python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential 
RUN apt update -y
#RUN rm -rf /var/lib/apt/lists
RUN  apt-get install -y librealsense2-dev librealsense2-utils
RUN apt install -y ros-melodic-ddynamic-reconfigure
RUN dpkg-reconfigure --frontend noninteractive tzdata
RUN rm -rf /var/lib/apt/lists
WORKDIR /home/nvidia/ws
RUN wget https://github.com/IntelRealSense/librealsense/archive/v2.36.0.tar.gz && tar -xvf v2.36.0.tar.gz && cd librealsense-2.36.0/ && mkdir build && cd build &&  cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true && make -j4 && make install
RUN mkdir src && cd src &&  git clone https://github.com/intel-ros/realsense.git -b development
#  && git clone https://github.com/slamtec/rplidar_ros

RUN rosdep init && rosdep update

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash;  cd /home/nvidia/ws; rosdep install --from-paths src --ignore-src; catkin_make'

AndreV84, 2020-08-03:
can not get 435i runing
on AGX;
it is listed with lsusb,
but sdk won't see it working correctly
```
/librealsense-2.36.0/build/tools/fw-update$ ./rs-fw-update -l

connected devices:
 03/08 17:56:55,730 ERROR [547890889152] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
RealSense error calling rs2_create_device(info_list:0x5582c1db70, index:0):
    failed to set power state
```

AndreV84, 2020-08-03:
reopened for d400 series assistance continuation

closed 2020-08-03 by MartyG-RealSense

AndreV84, 2020-08-03:
it wouldn't get running even on system wide instalaltoin with multiple versions of sdk tested from 35 to 38

MartyG-RealSense, 2020-08-03:
Please do not carry on this ticket.  Please leave it closed and start a new case for your latest question.   Thanks!

AndreV84, 2020-08-04:
yes, eventually after hours of attempts we found out that with sudo exection rs-fw-update -l will work
Thanks

mzahana, 2021-03-09:
@AndreV84 sorry to ask in a closed issue, but I am looking for how to run D435i within a docker container in Xavier NX dev kit, and I just would like to know if there are any conclusions on how to do that. 

Thanks.

AndreV84, 2021-03-09:
@mzahana  what prevents you from implementing it?

mzahana, 2021-03-09:
@AndreV84 thanks for the reply. So, are there there specific instructions on how to setup librealsens and the ROS1 wrapper inside a container on Xavier NX, or the apt-get packages should work, or I need to build from source? Do I need to patch the kernel ?

AndreV84, 2021-03-09:
@mzahana  why do not you run ros melodic systemwide realsense node? once you get it working it would be less complicated to dockerize it though it is diffucult

MartyG-RealSense, 2021-03-09:
Hi @mzahana   The most recent knowledge that I have about using Docker with librealsense and Nvida Jetson was that some users had problems with **udevadm**.

https://github.com/IntelRealSense/librealsense/issues/7849

Overall though, @AndreV84 knows a lot more about Docker than I do.

AndreV84, 2021-03-09:
I will release nx2 melodic ros node docker image
but little bit later

mzahana, 2021-03-14:
@AndreV84 I was able to install librealsense2 and realsense-ros both using `apt-get`, I am able run the camera with ROS and get the color, depth imaages as well as the pointcloud, but I am getting the following warning in the terminal,
```bash
14/03 09:09:23,976 WARNING [547516039552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

Is this something to worry about? Or it requires further fixes?

Thanks

AndreV84, 2021-03-20:
@MartyG-RealSense how do we capture a single shoot from terminal? will it work using rs-capture? which arguments required?
@mindThomas you would need to build the sdk with rsusb/ also libusb-dev package would need to be installed

MartyG-RealSense, 2021-03-20:
@AndreV84   If you want a librealsense C++ example for saving a single frame, there is **rs-save-to-disk**.  It captures the first 30 frames and then saves the last of those frames as a PNG image.

https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk

mindThomas, 2021-03-21:
@AndreV84 Why I am tagged here?

[44 more replies]