realsenseai/realsense-ros #1288: Out of frame resources

RGring, 2020-07-15:
My setup:
Jetson Xavier NX, L4T 32.4.3
Stereo camera Realsense D435i (connected to USB3.0)
ros wrapper 2.2.14 
librealsense 2.35.2
Firmware 5.12.6.0

I start the camera with the following configuration.
```
  <arg name="fps" default="15"/>

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="realsense"/>
    <arg name="enable_sync" value="true"/>
    <arg name="align_depth" value="true"/>

    <arg name="depth_width" value="1280"/>
    <arg name="depth_height" value="720"/>
    <arg name="infra_width" value="1280"/>
    <arg name="infra_height" value="720"/>

    <arg name="color_width" value="1920"/>
    <arg name="color_height" value="1080"/>
    
    <arg name="depth_fps" default="$(arg fps)"/>
    <arg name="color_fps" default="$(arg fps)"/>
  </include>
```
When I subscribe to the RGB (color/image_raw) and depth image (aligned_depth_to_color/image_raw), I get the following error.
```
 15/07 16:28:30,192 WARNING [546106753408] (types.cpp:49) Out of frame resources!
 15/07 16:28:30,192 ERROR [546106753408] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
```

Any ideas to optimize on it? Is 15 fps the lowest possible framerate?

I appreciate any help.

MartyG-RealSense, 2020-07-15:
The lowest FPS speed that is supported by RealSense 400 Series cameras is 6 FPS.  Using 6 FPS can in itself introduce errors that do not occur at 15 FPS or higher, but you are welcome to try 6 FPS and see how your project reacts.

RGring, 2020-07-16:
Thanks for the support. 

Decreasing to fps=6 results in less of those error messages, but unfortunately further remains. Also in a later stage of the project, I would like to go for a fps=15 with the high resolution images.

I read something, that building librealsense with CUDA support could help. Is that still up to date?

MartyG-RealSense, 2020-07-16:
CUDA support is included in the librealsense Debian packages.  

Do you get a further reduction in errors if you include this term in your roslaunch instruction:

**initial_reset:=true**

RGring, 2020-07-16:
It was already set in my last runs! So unfortunately not.

MartyG-RealSense, 2020-07-16:
Have you tried the camera in a different USB 3.1 port on the NX?

RGring, 2020-07-23:
Yes, sure.

MartyG-RealSense, 2020-07-23:
Some RealSense users who have wanted to define a custom speed that has a low value have used some form of **timer** so that they only use a certain number of the frames, such as 1 in every 5 frames.  So this, for example, allows them to run at 30 FPS but simulate a speed of 6 FPS (30 / 6).

In ROS, a timer mechanism called a **ROS Rate** is provided for setting custom frequencies.

https://roboticsbackend.com/ros-rate-roscpy-roscpp/

MartyG-RealSense, 2020-08-08:
Case closed due to no further comments received.

closed 2020-08-08 by MartyG-RealSense

Alex-Beh, 2020-12-01:
@RGring Hello, I met the exactly same issue with you on nvidia xavier NX. Did you manage to solve the issue?

GOBish, 2021-08-01:
I am getting the same error on the Xavier NX, I tried the initial_reset:=true with no success. Any advice?

MartyG-RealSense, 2021-08-01:
Hi @GOBish  What method did you use to build librealsense on your Jetson NX, please?

GOBish, 2021-08-01:
Built from source

MartyG-RealSense, 2021-08-01:
Thank you very much.  And you built from source using Intel's official Jetson instructions and included the CMake build flag ** -DBUILD_WITH_CUDA=true** to enable CUDA support in your librealsense build?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

GOBish, 2021-08-01:
Yes

MartyG-RealSense, 2021-08-01:
Is the camera being detected by the launch process as being on a USB 3 or USB 2.1 connection?  You can find this out by looking at the **Device USB type** line of the roslaunch log.  For example:

[ INFO] [1625956241.779675219]: Device with physical ID 1-1.3-6 was found.
[ INFO] [1625956241.779883552]: Device with name Intel RealSense D435I was found.
[ INFO] [1625956241.786105844]: Device with port number 1-1.3 was found.
**[ INFO] [1625956241.786421521]: Device USB type: 2.1**

GOBish, 2021-08-01:
Hi @MartyG-RealSense , I looked through all the roslaunch logs and could not find that output anywhere in the log.

Some sample output:

[roslaunch][INFO] 2021-08-01 19:41:58,238: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2021-08-01 19:41:58,240: roslaunch starting with args ['/opt/ros/melodic/bin/roslaunch', 'realsense2_camera', 'rs_camera.launch']
[roslaunch][INFO] 2021-08-01 19:41:58,241: roslaunch env is {'JETSON_OPENCV_CUDA': 'YES', 'XDG_SESSION_TYPE': 'x11', 'JETSON_TYPE': 'Xavier NX (Developer Kit Version)', 'VIRTUALENVWRAPPER_SCRIPT': '/usr/local/bin/virtualenvwrapper.sh', 'SHELL': '/bin/bash', 'XDG_DATA_DIRS': '/usr/share/unity:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop', 'COMPIZ_CONFIG_PROFILE': 'ubuntu', 'UNITY_DEFAULT_PROFILE': 'unity', 'TEXTDOMAIN': 'im-config', 'XMODIFIERS': '@im=ibus', 'OPENBLAS_CORETYPE': 'ARMV*', 'INVOCATION_ID': 'b038ebbd84224e739efe67785170638c', 'WORKON_HOME': '/home/gbenel/.virtualenvs', 'XDG_RUNTIME_DIR': '/run/user/1000', 'PYTHONPATH': '/home/gbenel/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages:/home/gbenel/models/research:/home/gbenel/models/research/slim:/usr/local/lib/python2.7/pyrealsense2:/home/gbenel/Downloads/dlib-19.21', 'COMPIZ_BIN_PATH': '/usr/bin/', 'JETSON_JETPACK': 'UNKNOWN', 'ROS_ROOT': '/opt/ros/melodic/share/ros', 'JETSON_BOARD': 'P3509-000', 'QT_ACCESSIBILITY': '1', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', 'GTK_MODULES': 'gail:atk-bridge:unity-gtk-module', 'DESKTOP_SESSION': 'unity', 'GNOME_TERMINAL_SERVICE': ':1.109', 'JETSON_BOARDIDS': '', 'JETSON_OPENCV': '4.5.0', 'XDG_MENU_PREFIX': 'gnome-', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'JETSON_L4T_REVISION': '4.4', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'XDG_CURRENT_DESKTOP': 'Unity:Unity7:ubuntu', 'USER': 'gbenel', 'CMAKE_PREFIX_PATH': '/home/gbenel/catkin_ws/devel:/opt/ros/melodic', 'MANAGERPID': '7431', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'JOURNAL_STREAM': '8:69278', 'SESSION_MANAGER': 'local/gbenel:@/tmp/.ICE-unix/8802,unix/gbenel:/tmp/.ICE-unix/8802', 'SHLVL': '2', 'LD_LIBRARY_PATH': '/home/gbenel/catkin_ws/devel/lib:/opt/ros/melodic/lib:/usr/local/lib:/usr/local/cuda-10.2/lib64:/usr/local/cuda-10.2/extras/CUPTI/lib64:/usr/local/cuda/extras/CUPTI/lib64', 'GTK_CSD': '0', 'CLUTTER_IM_MODULE': 'xim', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'JETSON_TENSORRT': '7.1.3.0', 'LANG': 'en_US.UTF-8', 'JETSON_VISIONWORKS': '1.6.0.501', 'JETSON_CUDA_ARCH_BIN': '7.2', 'GDMSESSION': 'unity', 'PKG_CONFIG_PATH': '/home/gbenel/catkin_ws/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig', 'ROS_PACKAGE_PATH': '/home/gbenel/catkin_ws/src:/opt/ros/melodic/share', '_': '/opt/ros/melodic/bin/roslaunch', 'GTK_IM_MODULE': 'ibus', 'JETSON_VPI': '0.4.4', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-unity:/etc/xdg', 'JETSON_L4T_RELEASE': '32', 'ROS_MASTER_URI': 'http://localhost:11311', 'COLORTERM': 'truecolor', 'GNOME_TERMINAL_SCREEN': '/org/gnome/Terminal/screen/efc32061_38af_4b0a_b280_1b7d89ad1d75', 'QT4_IM_MODULE': 'xim', 'ROS_IP': '192.168.1.26', 'HOME': '/home/gbenel', 'DISPLAY': ':0', 'XDG_SESSION_DESKTOP': 'unity', 'ROS_LOG_FILENAME': '/home/gbenel/.ros/log/ec75b5e2-f31f-11eb-8b6c-706655b14c15/roslaunch-gbenel-23901.log', 'VIRTUALENVWRAPPER_PYTHON': '/usr/bin/python3', 'JETSON_CUDNN': '8.0.0.180', 'ROS_PYTHON_VERSION': '2', 'USERNAME': 'gbenel', 'CUDA_HOME': '/usr/local/cuda-10.2', 'VTE_VERSION': '5202', 'JETSON_SERIAL_NUMBER': '1422120012431', 'JETSON_SOC': 'tegra194', 'JETSON_MACHINE': 'NVIDIA Jetson Xavier NX (Developer Kit Version)', 'JETSON_VULKAN_INFO': '1.2.70', 'VIRTUALENVWRAPPER_HOOK_DIR': '/home/gbenel/.virtualenvs', 'JETSON_CODENAME': 'jakku', 'ROS_ETC_DIR': '/opt/ros/melodic/etc/ros', 'SSH_AGENT_LAUNCHER': 'gnome-keyring', 'ROS_DISTRO': 'melodic', 'UNITY_HAS_3D_SUPPORT': 'true', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/gbenel/catkin_ws/devel/shar [... cut ...] : 'P3668', 'JETSON_L4T': '32.4.4', 'TERM': 'xterm-256color', 'VIRTUALENVWRAPPER_WORKON_CD': '1', 'WINDOWPATH': '1', 'IM_CONFIG_PHASE': '2', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'TEXTDOMAINDIR': '/usr/share/locale/', 'JETSON_CUDA': '10.2.89', 'LD_PRELOAD': 'libgtk3-nocsd.so.0', 'VIRTUALENVWRAPPER_PROJECT_FILENAME': '.project', 'OLDPWD': '/home/gbenel/catkin_ws', 'PWD': '/home/gbenel/catkin_ws/src/realsense-ros/realsense2_camera', 'ROS_VERSION': '1', 'GNOME_SESSION_XDG_SESSION_PATH': ''}

MartyG-RealSense, 2021-08-02:
Hi @GOBish  The above log does not look like a log from a roslaunch.  I mean the log that is generated in the terminal during a ROS launch process, like the one in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1957#issuecomment-872505287

GOBish, 2021-08-02:
I tried using my D455 now instead of the D435 and I get the following output:

~/catkin_ws/src/realsense-ros/realsense2_camera$ roslaunch realsense2_camera rs_d400_and_t265.launch 
... logging to /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/roslaunch-gbenel-5111.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.26:34857/

SUMMARY
========

PARAMETERS
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/depth_fps: 30
 * /d400/realsense2_camera/depth_frame_id: d400_depth_frame
 * /d400/realsense2_camera/depth_height: 480
 * /d400/realsense2_camera/depth_optical_frame_id: d400_depth_optica...
 * /d400/realsense2_camera/depth_width: 640
 * /d400/realsense2_camera/device_type: d4.5
 * /d400/realsense2_camera/enable_accel: True
 * /d400/realsense2_camera/enable_color: True
 * /d400/realsense2_camera/enable_depth: True
 * /d400/realsense2_camera/enable_fisheye1: True
 * /d400/realsense2_camera/enable_fisheye2: True
 * /d400/realsense2_camera/enable_fisheye: True
 * /d400/realsense2_camera/enable_gyro: True
 * /d400/realsense2_camera/enable_infra1: True
 * /d400/realsense2_camera/enable_infra2: True
 * /d400/realsense2_camera/enable_pointcloud: False
 * /d400/realsense2_camera/enable_sync: False
 * /d400/realsense2_camera/filters: pointcloud
 * /d400/realsense2_camera/fisheye1_frame_id: d400_fisheye1_frame
 * /d400/realsense2_camera/fisheye1_optical_frame_id: d400_fisheye1_opt...
 * /d400/realsense2_camera/fisheye2_frame_id: d400_fisheye2_frame
 * /d400/realsense2_camera/fisheye2_optical_frame_id: d400_fisheye2_opt...
 * /d400/realsense2_camera/fisheye_fps: 30
 * /d400/realsense2_camera/fisheye_frame_id: d400_fisheye_frame
 * /d400/realsense2_camera/fisheye_height: 0
 * /d400/realsense2_camera/fisheye_optical_frame_id: d400_fisheye_opti...
 * /d400/realsense2_camera/fisheye_width: 0
 * /d400/realsense2_camera/gyro_fps: 0
 * /d400/realsense2_camera/gyro_frame_id: d400_gyro_frame
 * /d400/realsense2_camera/gyro_optical_frame_id: d400_gyro_optical...
 * /d400/realsense2_camera/imu_optical_frame_id: d400_imu_optical_...
 * /d400/realsense2_camera/infra1_frame_id: d400_infra1_frame
 * /d400/realsense2_camera/infra1_optical_frame_id: d400_infra1_optic...
 * /d400/realsense2_camera/infra2_frame_id: d400_infra2_frame
 * /d400/realsense2_camera/infra2_optical_frame_id: d400_infra2_optic...
 * /d400/realsense2_camera/infra_fps: 30
 * /d400/realsense2_camera/infra_height: 480
[... 149 lines cut ...]
[ INFO] [1627918334.212220871]: Device Product ID: 0x0B37
[ INFO] [1627918334.212311528]: Enable PointCloud: Off
[ INFO] [1627918334.212377544]: Align Depth: Off
[ INFO] [1627918334.212473129]: Sync Mode: Off
[ INFO] [1627918334.213272462]: Device Sensors: 
[ INFO] [1627918334.213935410]: Tracking Module was found.
[ INFO] [1627918334.214101715]: (Depth, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918334.214206323]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918334.214291700]: (Infrared, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918334.214373236]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918334.214465621]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918334.215097337]: num_filters: 0
[ INFO] [1627918334.215245018]: Setting Dynamic reconfig parameters.
[ INFO] [1627918334.273361049]: Done Setting Dynamic reconfig parameters.
[ INFO] [1627918334.274722082]: setupPublishers...
[ INFO] [1627918334.289905149]: setupStreams...
[ INFO] [1627918334.290366208]: insert Gyro to Tracking Module
[ INFO] [1627918334.290973124]: insert Accel to Tracking Module
[ INFO] [1627918334.291161605]: insert Pose to Tracking Module
[ INFO] [1627918334.333081123]: SELECTED BASE:Pose, 0
[ INFO] [1627918334.352386936]: RealSense Node Is Up!
[d400/realsense2_camera_manager-3] process has died [pid 5130, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/d400-realsense2_camera_manager-3.log].
log file: /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/d400-realsense2_camera_manager-3*.log
[d400/realsense2_camera-4] process has finished cleanly
log file: /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/d400-realsense2_camera-4*.log

GOBish, 2021-08-02:
putting the D435 back in, I get this output now:

~/catkin_ws/src/realsense-ros/realsense2_camera$ roslaunch realsense2_camera rs_d400_and_t265.launch 
... logging to /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/roslaunch-gbenel-5968.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.26:41113/

SUMMARY
========

PARAMETERS
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/depth_fps: 30
 * /d400/realsense2_camera/depth_frame_id: d400_depth_frame
 * /d400/realsense2_camera/depth_height: 480
 * /d400/realsense2_camera/depth_optical_frame_id: d400_depth_optica...
 * /d400/realsense2_camera/depth_width: 640
 * /d400/realsense2_camera/device_type: d4.5
 * /d400/realsense2_camera/enable_accel: True
 * /d400/realsense2_camera/enable_color: True
 * /d400/realsense2_camera/enable_depth: True
 * /d400/realsense2_camera/enable_fisheye1: True
 * /d400/realsense2_camera/enable_fisheye2: True
 * /d400/realsense2_camera/enable_fisheye: True
 * /d400/realsense2_camera/enable_gyro: True
 * /d400/realsense2_camera/enable_infra1: True
 * /d400/realsense2_camera/enable_infra2: True
 * /d400/realsense2_camera/enable_pointcloud: False
 * /d400/realsense2_camera/enable_sync: False
 * /d400/realsense2_camera/filters: pointcloud
 * /d400/realsense2_camera/fisheye1_frame_id: d400_fisheye1_frame
 * /d400/realsense2_camera/fisheye1_optical_frame_id: d400_fisheye1_opt...
 * /d400/realsense2_camera/fisheye2_frame_id: d400_fisheye2_frame
 * /d400/realsense2_camera/fisheye2_optical_frame_id: d400_fisheye2_opt...
 * /d400/realsense2_camera/fisheye_fps: 30
 * /d400/realsense2_camera/fisheye_frame_id: d400_fisheye_frame
 * /d400/realsense2_camera/fisheye_height: 0
 * /d400/realsense2_camera/fisheye_optical_frame_id: d400_fisheye_opti...
 * /d400/realsense2_camera/fisheye_width: 0
 * /d400/realsense2_camera/gyro_fps: 0
 * /d400/realsense2_camera/gyro_frame_id: d400_gyro_frame
 * /d400/realsense2_camera/gyro_optical_frame_id: d400_gyro_optical...
 * /d400/realsense2_camera/imu_optical_frame_id: d400_imu_optical_...
 * /d400/realsense2_camera/infra1_frame_id: d400_infra1_frame
 * /d400/realsense2_camera/infra1_optical_frame_id: d400_infra1_optic...
 * /d400/realsense2_camera/infra2_frame_id: d400_infra2_frame
 * /d400/realsense2_camera/infra2_optical_frame_id: d400_infra2_optic...
 * /d400/realsense2_camera/infra_fps: 30
 * /d400/realsense2_camera/infra_height: 480
[... 142 lines cut ...]
[ INFO] [1627918721.361899370]: Device Product ID: 0x0B37
[ INFO] [1627918721.361961578]: Enable PointCloud: Off
[ INFO] [1627918721.362020075]: Align Depth: Off
[ INFO] [1627918721.362073579]: Sync Mode: Off
[ INFO] [1627918721.362416781]: Device Sensors: 
[ INFO] [1627918721.363174546]: Tracking Module was found.
[ INFO] [1627918721.363348051]: (Depth, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918721.363452660]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918721.363522420]: (Infrared, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918721.363605781]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918721.363669397]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1627918721.364176249]: num_filters: 0
[ INFO] [1627918721.364276537]: Setting Dynamic reconfig parameters.
[ INFO] [1627918721.431919861]: Done Setting Dynamic reconfig parameters.
[ INFO] [1627918721.433380670]: setupPublishers...
[ INFO] [1627918721.448389056]: setupStreams...
[ INFO] [1627918721.448893892]: insert Gyro to Tracking Module
[ INFO] [1627918721.449466856]: insert Accel to Tracking Module
[ INFO] [1627918721.449609736]: insert Pose to Tracking Module
[ INFO] [1627918721.493905483]: SELECTED BASE:Pose, 0
[ INFO] [1627918721.514903028]: RealSense Node Is Up!
[d400/realsense2_camera_manager-3] process has died [pid 5987, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/d400-realsense2_camera_manager-3.log].
log file: /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/d400-realsense2_camera_manager-3*.log
[d400/realsense2_camera-4] process has finished cleanly
log file: /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/d400-realsense2_camera-4*.log

MartyG-RealSense, 2021-08-02:
Thanks for the log.  I cannot comment on the T265 as I do not handle that RealSense model.  Was **roslaunch realsense2_camera rs_d400_and_t265.launch** the launch instruction that you were using with D435, launching the D435 and T265 together and receiving the **Out of frame resources** error?

If you were using D435 and T265 together, do you experience errors if you instead launch T265 and D435 with **rs_multple_devices.launch** 

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

GOBish, 2021-08-02:
Yes, I was getting the out of frame resources error when using the D435 and T265 together, and also sometimes when doing just d435 alone.
with the d455 and rs_multiple_devices.launch:

roslaunch realsense2_camera rs_multiple_devices.launch 
... logging to /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/roslaunch-gbenel-8788.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.26:42483/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: True
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: True
 * /camera1/realsense2_camera/enable_fisheye2: True
 * /camera1/realsense2_camera/enable_fisheye: True
 * /camera1/realsense2_camera/enable_gyro: True
 * /camera1/realsense2_camera/enable_infra1: True
 * /camera1/realsense2_camera/enable_infra2: True
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: 
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye2_optical_frame_id: camera1_fisheye2_...
 * /camera1/realsense2_camera/fisheye_fps: 30
 * /camera1/realsense2_camera/fisheye_frame_id: camera1_fisheye_f...
 * /camera1/realsense2_camera/fisheye_height: 0
 * /camera1/realsense2_camera/fisheye_optical_frame_id: camera1_fisheye_o...
 * /camera1/realsense2_camera/fisheye_width: 0
 * /camera1/realsense2_camera/gyro_fps: 0
 * /camera1/realsense2_camera/gyro_frame_id: camera1_gyro_frame
 * /camera1/realsense2_camera/gyro_optical_frame_id: camera1_gyro_opti...
 * /camera1/realsense2_camera/imu_optical_frame_id: camera1_imu_optic...
 * /camera1/realsense2_camera/infra1_frame_id: camera1_infra1_frame
 * /camera1/realsense2_camera/infra1_optical_frame_id: camera1_infra1_op...
 * /camera1/realsense2_camera/infra2_frame_id: camera1_infra2_frame
 * /camera1/realsense2_camera/infra2_optical_frame_id: camera1_infra2_op...
 * /camera1/realsense2_camera/infra_fps: 30
[... 116 lines cut ...]
 02/08 11:48:04,818 ERROR [546794627456] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 02/08 11:48:04,821 WARNING [546821853568] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
terminate called after throwing an instance of 'rs2::error'
  what():  failed to set power state
[ INFO] [1627919285.014822033]: Device with serial number 044422250214 was found.

[ INFO] [1627919285.015045874]: Device with physical ID 2-3.4-19 was found.
[ INFO] [1627919285.015138419]: Device with name Intel RealSense D455 was found.
[ WARN] [1627919285.023794542]: Error extracting usb port from device with physical ID: 2-3.4-19
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[FATAL] [1627919285.024918262]: Unsupported device! Product ID: 0x0B5C
terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error'
  what():  stop_streaming() failed. UVC device is not streaming!
[camera2/realsense2_camera_manager-3] process has died [pid 8807, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/camera2-realsense2_camera_manager-3.log].
log file: /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/camera2-realsense2_camera_manager-3*.log
[... cut ...]
log file: /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/camera1-realsense2_camera_manager-1*.log
[camera1/realsense2_camera-2] process has finished cleanly
log file: /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/camera1-realsense2_camera-2*.log
[camera2/realsense2_camera-4] process has finished cleanly
log file: /home/gbenel/.ros/log/9f30c678-f3a6-11eb-bf91-706655b14c15/camera2-realsense2_camera-4*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

GOBish, 2021-08-02:
I tried entering the serial numbers of the camera directly in the launch file as mentioned in several other issues, but that did not solve it.

MartyG-RealSense, 2021-08-03:
I noticed that there is a mismatch between your ROS wrapper and librealsense versions in the log above.  It is important to match a specific ROS wrapper version as closely as possible with the librealsense version listed in its release notes.  If you are using librealsense **2.39.0** then the correct wrapper match for it is **2.2.18** rather than the 2.2.11 version that you are currently using.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.18

GOBish, 2021-08-03:
Thanks for picking that up. I reinstalled with the 2.2.18 wrapper and re-launched rs_multiple devices (and adding the serial numbers and names to the launch file) and get the following output:

~/catkin_ws/src/realsense-ros/realsense2_camera$ roslaunch realsense2_camera rs_multiple_devices.launch initial_reset:=true
... logging to /home/gbenel/.ros/log/d5a53fcc-f469-11eb-b65d-706655b14c15/roslaunch-gbenel-5497.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.26:37249/

SUMMARY
========

PARAMETERS
 * /d435/realsense2_camera/accel_fps: 0
 * /d435/realsense2_camera/accel_frame_id: d435_accel_frame
 * /d435/realsense2_camera/accel_optical_frame_id: d435_accel_optica...
 * /d435/realsense2_camera/align_depth: False
 * /d435/realsense2_camera/aligned_depth_to_color_frame_id: d435_aligned_dept...
 * /d435/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d435_aligned_dept...
 * /d435/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d435_aligned_dept...
 * /d435/realsense2_camera/aligned_depth_to_fisheye_frame_id: d435_aligned_dept...
 * /d435/realsense2_camera/aligned_depth_to_infra1_frame_id: d435_aligned_dept...
 * /d435/realsense2_camera/aligned_depth_to_infra2_frame_id: d435_aligned_dept...
 * /d435/realsense2_camera/allow_no_texture_points: False
 * /d435/realsense2_camera/base_frame_id: d435_link
 * /d435/realsense2_camera/calib_odom_file: 
 * /d435/realsense2_camera/clip_distance: -1.0
 * /d435/realsense2_camera/color_fps: 30
 * /d435/realsense2_camera/color_frame_id: d435_color_frame
 * /d435/realsense2_camera/color_height: 480
 * /d435/realsense2_camera/color_optical_frame_id: d435_color_optica...
 * /d435/realsense2_camera/color_width: 640
 * /d435/realsense2_camera/depth_fps: 30
 * /d435/realsense2_camera/depth_frame_id: d435_depth_frame
 * /d435/realsense2_camera/depth_height: 480
 * /d435/realsense2_camera/depth_optical_frame_id: d435_depth_optica...
 * /d435/realsense2_camera/depth_width: 640
 * /d435/realsense2_camera/device_type: 
 * /d435/realsense2_camera/enable_accel: False
 * /d435/realsense2_camera/enable_color: True
 * /d435/realsense2_camera/enable_depth: True
 * /d435/realsense2_camera/enable_fisheye1: False
 * /d435/realsense2_camera/enable_fisheye2: False
 * /d435/realsense2_camera/enable_fisheye: False
 * /d435/realsense2_camera/enable_gyro: False
 * /d435/realsense2_camera/enable_infra1: False
 * /d435/realsense2_camera/enable_infra2: False
 * /d435/realsense2_camera/enable_infra: False
 * /d435/realsense2_camera/enable_pointcloud: False
 * /d435/realsense2_camera/enable_pose: False
 * /d435/realsense2_camera/enable_sync: False
 * /d435/realsense2_camera/filters: 
 * /d435/realsense2_camera/fisheye1_frame_id: d435_fisheye1_frame
 * /d435/realsense2_camera/fisheye1_optical_frame_id: d435_fisheye1_opt...
 * /d435/realsense2_camera/fisheye2_frame_id: d435_fisheye2_frame
 * /d435/realsense2_camera/fisheye2_optical_frame_id: d435_fisheye2_opt...
 * /d435/realsense2_camera/fisheye_fps: 30
 * /d435/realsense2_camera/fisheye_frame_id: d435_fisheye_frame
 * /d435/realsense2_camera/fisheye_height: 0
 * /d435/realsense2_camera/fisheye_optical_frame_id: d435_fisheye_opti...
 * /d435/realsense2_camera/fisheye_width: 0
 * /d435/realsense2_camera/gyro_fps: 0
 * /d435/realsense2_camera/gyro_frame_id: d435_gyro_frame
 * /d435/realsense2_camera/gyro_optical_frame_id: d435_gyro_optical...
 * /d435/realsense2_camera/imu_optical_frame_id: d435_imu_optical_...
 * /d435/realsense2_camera/infra1_frame_id: d435_infra1_frame
 * /d435/realsense2_camera/infra1_optical_frame_id: d435_infra1_optic...
 * /d435/realsense2_camera/infra2_frame_id: d435_infra2_frame
 * /d435/realsense2_camera/infra2_optical_frame_id: d435_infra2_optic...
[... 211 lines cut ...]
[ INFO] [1628003058.661493372]: num_filters: 0
[ INFO] [1628003058.661618716]: Setting Dynamic reconfig parameters.
[ INFO] [1628003059.220882718]: Done Setting Dynamic reconfig parameters.
[ INFO] [1628003059.222421704]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1628003059.224145907]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1628003059.224275828]: setupPublishers...
[ INFO] [1628003059.236247076]: Expected frequency for depth = 30.00000
[ INFO] [1628003059.321113117]: Expected frequency for color = 30.00000
[ INFO] [1628003059.388532864]: setupStreams...
[ INFO] [1628003059.413212678]: insert Depth to Stereo Module
[ INFO] [1628003059.413416647]: insert Color to RGB Camera
 03/08 11:04:19,495 WARNING [547174719872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1628003059.546572899]: SELECTED BASE:Depth, 0
[ INFO] [1628003059.594286274]: RealSense Node Is Up!
 03/08 11:04:19,617 WARNING [547174719872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:04:19,667 WARNING [547174719872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:04:19,718 WARNING [547174719872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:04:19,899 WARNING [547174719872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 03/08 11:04:20,303 WARNING [547174719872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:04:20,354 WARNING [547174719872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:04:20,505 WARNING [547174719872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:04:20,556 WARNING [547174719872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-08-03:
Good to hear that you made progress.  Are the topics able to be visualized in RViz despite the **control_transfer returned** messages?

GOBish, 2021-08-03:
The topics seem to be mostly listed, but when I try to add them in RVIZ the data doesn't come through -- i.e. images from the rgb camera don't show up. The depth image comes up very distorted.

MartyG-RealSense, 2021-08-03:
It is a difficult case to answer as I am not involved in the T265 and so cannot discuss that aspect of your setup.

Do you still receive errors if you unplug the T265 and just run the D455 or the D435 on its own?

GOBish, 2021-08-03:
with t265 unplugged and running rs_camera.launch:
only get color and depth images in rviz, no infrared topics as I got before changing the wrapper. depth image is distorted.

~/catkin_ws/src/realsense-ros/realsense2_camera$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/gbenel/.ros/log/d5a53fcc-f469-11eb-b65d-706655b14c15/roslaunch-gbenel-1151.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.26:35345/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 66 lines cut ...]
[ INFO] [1628004768.403292157]: Done Setting Dynamic reconfig parameters.
[ INFO] [1628004768.404718471]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1628004768.406348177]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1628004768.406483378]: setupPublishers...
[ INFO] [1628004768.417801949]: Expected frequency for depth = 30.00000
[ INFO] [1628004768.501194818]: Expected frequency for color = 30.00000
[ INFO] [1628004768.569006401]: setupStreams...
[ INFO] [1628004768.593919813]: insert Depth to Stereo Module
[ INFO] [1628004768.594151302]: insert Color to RGB Camera
 03/08 11:32:48,675 WARNING [546836533632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1628004768.726564303]: SELECTED BASE:Depth, 0
[ INFO] [1628004768.775618354]: RealSense Node Is Up!
 03/08 11:32:48,796 WARNING [546836533632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:32:48,846 WARNING [546836533632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:32:48,897 WARNING [546836533632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:32:49,077 WARNING [546836533632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 03/08 11:32:49,683 WARNING [546836533632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:32:49,734 WARNING [546836533632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:32:49,885 WARNING [546836533632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 11:32:49,936 WARNING [546836533632] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-08-03:
The log says that llibrealsense and the wrapper are mismatched again.  It now has librealsense **2.45.0** and wrapper 2.2.18 (wen the correct match for 2.2.18 is 2.39.0).  

If you have upgraded librealsense to 2.45.0 then you need to build wrapper **2.3.0** so wrapper and SDK are correctly matched again.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.0

GOBish, 2021-08-03:
strange, I didn't change anything. I will biuld with wrapper 2.3.0 now and try again.

GOBish, 2021-08-03:
~/catkin_ws/src/realsense-ros/realsense2_camera$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/gbenel/.ros/log/074087e4-f474-11eb-9db9-706655b14c15/roslaunch-gbenel-18769.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.26:35643/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 76 lines cut ...]
[ INFO] [1628007086.509848464]: Done Setting Dynamic reconfig parameters.
[ INFO] [1628007086.511541042]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1628007086.513371220]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1628007086.513529332]: setupPublishers...
[ INFO] [1628007086.524622848]: Expected frequency for depth = 30.00000
[ INFO] [1628007086.612880224]: Expected frequency for color = 30.00000
[ INFO] [1628007086.685949903]: setupStreams...
[ INFO] [1628007086.710406826]: insert Depth to Stereo Module
[ INFO] [1628007086.710602794]: insert Color to RGB Camera
 03/08 12:11:26,794 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1628007086.845836349]: SELECTED BASE:Depth, 0
[ INFO] [1628007086.901810682]: RealSense Node Is Up!
[ WARN] [1628007086.915233225]: 
 03/08 12:11:26,916 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 12:11:26,967 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 12:11:27,192 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 12:11:27,242 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 03/08 12:11:27,799 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 12:11:27,850 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 12:11:28,001 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/08 12:11:28,052 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

GOBish, 2021-08-03:
I get this error often as well:
03/08 12:16:28,416 ERROR [546744271232] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 03/08 12:16:29,139 ERROR [546719093120] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130

GOBish, 2021-08-03:
my ultimate goal is to do VSLAM with rtab map on the jetson, and it feels like I am nowhere close.

Rtabmap using the tutorial here http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping is not able to read the topics at all, and crashes.

MartyG-RealSense, 2021-08-03:
You have a correct librealense and wrapper pairing now (2.45.0 and 2.3.0) and the camera is on a USB 3 connection.

There is clearly a serious communication problem with the device if the **control_transfer returned** message is generating continuously instead of being produced a small number of times and then stopping.

Do you have the camera attached directly to the Jetson or plugged into a hub?

GOBish, 2021-08-03:
directly.

After a couple of seconds the control_transfer returned error stops.

MartyG-RealSense, 2021-08-03:
Okay, that sounds healthy.   The topics should be publishing.  Let's check the depth topic without using RViz to see whether it is generating frames.  Please enter the command below into the ROS terminal after the launch has completed.

**rostopic echo /camera/depth/image_rect_raw**

GOBish, 2021-08-03:
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 189, 2, 189, 2, 189, 2, 189, 2, 189, 2, 189, 2, 189, 2, 189, 2, 189, 2, 189, 2, 189, 2, 188, 2, 188, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 190, 1, 190, 1, 190, 1, 190, 1, 190, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 141, 1, 141, 1, 141, 1, 141, 1, 141, 1, 141, 1, 141, 1, 141, 1, 141, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 133, 1, 133, 1, 133, 1, 133, 1, 133, 1, 133, 1, 133, 1, 133, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 75, 1, 75, 1, 74, 1, 74, 1, 0, 0, 73, 1, 73, 1, 72, 1, 72, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 15, 3, 15, 3, 15, 3, 15, 3, 15, 3, 15, 3, 15, 3, 14, 3, 14, 3, 14, 3, 12, 3, 10, 3, 6, 3, 2, 3, 1, 3, 255, 2, 254, 2, 251, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 [... cut ...] , 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
---

MartyG-RealSense, 2021-08-04:
Let's see if identifying the camera by the ID of the USB port that it is in makes any difference.  The **Device with port number** line of your launch log finds the camera on the USB port with ID **2.3.2**.  So please plug the D435 camera into the same USB port that you used for your previous test above and use the roslaunch instruction below.

roslaunch realsense2_camera rs_camera.launch usb_port_id:=2-3-2

GOBish, 2021-08-04:
Using usb_port_d:=2-3-2 produced
 
[ INFO] [1628087113.522289383]: Initializing nodelet with 6 worker threads.
[ INFO] [1628087113.892618274]: RealSense ROS v2.3.0
[ INFO] [1628087113.892741123]: Built with LibRealSense v2.45.0
[ INFO] [1628087113.892829795]: Running with LibRealSense v2.45.0
[ INFO] [1628087113.984497883]:  
[ INFO] [1628087114.308376356]: Device with serial number 819312072122 was found.

[ INFO] [1628087114.308525029]: Device with physical ID 2-3.2-5 was found.
[ INFO] [1628087114.308622662]: Device with name Intel RealSense D435 was found.
[ INFO] [1628087114.310319124]: Device with port number 2-3.2 was found.
[ERROR] [1628087114.312452069]: The requested device with usb port id 2-3-2 is NOT found. Will Try again.

but using usb_port_d:=2-3.2 produced 

~/catkin_ws/src/realsense-ros/realsense2_camera$ roslaunch realsense2_camera rs_camera.launch usb_port_id:=2-3.2
... logging to /home/gbenel/.ros/log/7089f48a-f48b-11eb-af8a-706655b14c15/roslaunch-gbenel-25785.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.26:44697/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
[... 81 lines cut ...]
[ INFO] [1628087169.609175282]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1628087169.609304979]: num_filters: 0
[ INFO] [1628087169.609396276]: Setting Dynamic reconfig parameters.
[ INFO] [1628087170.192621506]: Done Setting Dynamic reconfig parameters.
[ INFO] [1628087170.194467569]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1628087170.196114750]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1628087170.196254207]: setupPublishers...
[ INFO] [1628087170.206906545]: Expected frequency for depth = 30.00000
[ INFO] [1628087170.295385568]: Expected frequency for color = 30.00000
[ INFO] [1628087170.367817714]: setupStreams...
[ INFO] [1628087170.392118894]: insert Depth to Stereo Module
[ INFO] [1628087170.392289040]: insert Color to RGB Camera
[ INFO] [1628087170.482891695]: SELECTED BASE:Depth, 0
[ INFO] [1628087170.537599255]: RealSense Node Is Up!
[ WARN] [1628087170.603362998]: 
 04/08 10:26:10,605 WARNING [546974183808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/08 10:26:10,655 WARNING [546974183808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/08 10:26:10,877 WARNING [546974183808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/08 10:26:10,927 WARNING [546974183808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/08 10:26:11,079 WARNING [546974183808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 04/08 10:26:11,281 WARNING [546974183808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/08 10:26:11,332 WARNING [546974183808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/08 10:26:11,483 WARNING [546974183808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/08 10:26:11,533 WARNING [546974183808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

GOBish, 2021-08-04:
when I launch rs_d400_and_t265 and then look at the topics, I get this:

rostopic hz /d400/color/image_raw
subscribed to [/d400/color/image_raw]
average rate: 37.615
	min: 0.001s max: 0.118s std dev: 0.04730s window: 15
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
...

right after I run the rostopic command I get this error, which causes the no new messages:

04/08 11:27:12,806 ERROR [547121779072] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

which is probably one of the things causing rtab-map to fail as well

MartyG-RealSense, 2021-08-04:
I have searched through many cases that feature multiple-generating **(uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132** messages but found few references that looked helpful.  The resources that I would consider most helpful are linked below.

https://github.com/IntelRealSense/realsense-ros/issues/1487#issuecomment-723591832
https://github.com/IntelRealSense/realsense-ros/issues/1164#issuecomment-620903027

GOBish, 2021-08-05:
hmm, I can't seem to figure this one out.

MartyG-RealSense, 2021-08-05:
@GOBish   The firmware that you currently have installed in the camera is **5.12.8.200**, according to your launch log.  Now that you are using SDK 2.45.0 it may be worth upgrading the firmware to **5.12.13.50**, the firmware recommended for 2.45.0.

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

GOBish, 2021-08-05:
updated firmware. Topics are still empty.

~/catkin_ws/src/realsense-ros/realsense2_camera$ roslaunch realsense2_camera rs_d400_and_t265.launch 
... logging to /home/gbenel/.ros/log/5565da3e-f535-11eb-8c42-706655b14c15/roslaunch-gbenel-1965.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.26:38457/

SUMMARY
========

PARAMETERS
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/confidence_fps: 30
 * /d400/realsense2_camera/confidence_height: 480
 * /d400/realsense2_camera/confidence_width: 640
 * /d400/realsense2_camera/depth_fps: 30
 * /d400/realsense2_camera/depth_frame_id: d400_depth_frame
 * /d400/realsense2_camera/depth_height: 480
 * /d400/realsense2_camera/depth_optical_frame_id: d400_depth_optica...
 * /d400/realsense2_camera/depth_width: 640
 * /d400/realsense2_camera/device_type: D435
 * /d400/realsense2_camera/enable_accel: False
 * /d400/realsense2_camera/enable_color: True
 * /d400/realsense2_camera/enable_confidence: True
 * /d400/realsense2_camera/enable_depth: True
 * /d400/realsense2_camera/enable_fisheye1: False
 * /d400/realsense2_camera/enable_fisheye2: False
 * /d400/realsense2_camera/enable_fisheye: False
 * /d400/realsense2_camera/enable_gyro: False
 * /d400/realsense2_camera/enable_infra1: False
 * /d400/realsense2_camera/enable_infra2: False
 * /d400/realsense2_camera/enable_infra: False
 * /d400/realsense2_camera/enable_pointcloud: False
 * /d400/realsense2_camera/enable_pose: False
 * /d400/realsense2_camera/enable_sync: False
 * /d400/realsense2_camera/filters: pointcloud
 * /d400/realsense2_camera/fisheye1_frame_id: d400_fisheye1_frame
 * /d400/realsense2_camera/fisheye1_optical_frame_id: d400_fisheye1_opt...
 * /d400/realsense2_camera/fisheye2_frame_id: d400_fisheye2_frame
 * /d400/realsense2_camera/fisheye2_optical_frame_id: d400_fisheye2_opt...
 * /d400/realsense2_camera/fisheye_fps: 30
 * /d400/realsense2_camera/fisheye_frame_id: d400_fisheye_frame
 * /d400/realsense2_camera/fisheye_height: 0
 * /d400/realsense2_camera/fisheye_optical_frame_id: d400_fisheye_opti...
 * /d400/realsense2_camera/fisheye_width: 0
 * /d400/realsense2_camera/gyro_fps: 0
 * /d400/realsense2_camera/gyro_frame_id: d400_gyro_frame
 * /d400/realsense2_camera/gyro_optical_frame_id: d400_gyro_optical...
 * /d400/realsense2_camera/imu_optical_frame_id: d400_imu_optical_...
[... 227 lines cut ...]
[ INFO] [1628187104.753194450]: Expected frequency for color = 30.00000
[ INFO] [1628187104.820047529]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1628187104.888291370]: setupStreams...
[ INFO] [1628187104.913008888]: insert Depth to Stereo Module
[ INFO] [1628187104.913170105]: insert Color to RGB Camera
[ INFO] [1628187104.998739316]: SELECTED BASE:Depth, 0
[ INFO] [1628187105.053169907]: RealSense Node Is Up!
[ WARN] [1628187105.120369133]: 
 05/08 14:11:45,122 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:11:45,172 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:11:45,223 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:11:45,274 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:11:45,325 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:11:45,375 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:11:45,426 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:11:45,477 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:11:45,527 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 05/08 14:13:10,543 ERROR [546828165504] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 05/08 14:13:10,944 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:17:54,183 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 05/08 14:17:54,233 WARNING [547172102528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

GOBish, 2021-08-05:
rostopic hz /camera/depth/image_rect_raw
subscribed to [/camera/depth/image_rect_raw]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages

GOBish, 2021-08-05:
:~/catkin_ws/src/realsense-ros/realsense2_camera$ rostopic hz /d400/color/image_raw

subscribed to [/d400/color/image_raw]
no new messages
no new messages
no new messages
no new messages

GOBish, 2021-08-05:
Here is the launch file:

<launch>
  <arg name="device_type_camera1"    		default="T265"/>
  <arg name="device_type_camera2"    		default="D435"/>		
  <arg name="serial_no_camera1"    			default="909212110257"/>
  <arg name="serial_no_camera2"    			default="819312072122"/>
  <arg name="camera1"              			default="t265"/>
  <arg name="camera2"              			default="d400"/>
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="tf_prefix_camera2"         default="$(arg camera2)"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="enable_fisheye"            default="false"/>
  <arg name="color_width"               default="640"/>
  <arg name="color_height"              default="480"/>
  <arg name="depth_width"               default="640"/>
  <arg name="depth_height"              default="480"/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>

  <group ns="$(arg camera1)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="device_type"           value="$(arg device_type_camera1)"/>
      <arg name="serial_no"             value="$(arg serial_no_camera1)"/>
      <arg name="tf_prefix"         		value="$(arg tf_prefix_camera1)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
      <arg name="enable_fisheye1"       value="$(arg enable_fisheye)"/>
      <arg name="enable_fisheye2"       value="$(arg enable_fisheye)"/>
      <arg name="topic_odom_in"         value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"       value="$(arg calib_odom_file)"/>
      <arg name="enable_pose"           value="true"/>
    </include>
  </group>

  <group ns="$(arg camera2)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="device_type"           value="$(arg device_type_camera2)"/>
      <arg name="serial_no"             value="$(arg serial_no_camera2)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_camera2)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
      <arg name="align_depth"           value="true"/>
      <arg name="filters"               value="pointcloud"/>
      <arg name="color_width"           value="$(arg color_width)"/>
      <arg name="color_height"          value="$(arg color_height)"/>
      <arg name="depth_width"           value="$(arg depth_width)"/>
      <arg name="depth_height"          value="$(arg depth_height)"/>
      <arg name="clip_distance"         value="$(arg clip_distance)"/>

      
    </include>
  </group>
  <node pkg="tf" type="static_transform_publisher" name="t265_to_d400" args="0 0 0 0 0 0 /$(arg tf_prefix_camera1)_link /$(arg tf_prefix_camera2)_link 100"/>

</launch>

GOBish, 2021-08-05:
and I still get the watchdog error:

05/08 14:25:29,363 ERROR [547021099392] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

MartyG-RealSense, 2021-08-06:
@GOBish   Have you tried reflashing the kernel of your Jetson NX, please?

https://developer.ridgerun.com/wiki/index.php?title=Jetson_Xavier_NX/Development/Building_the_Kernel_from_Source#Flash_Kernel_Image_only