realsenseai/realsense-ros #1289: L515 ros wrapper will run on nvidia Jetson but will throw many errors and warnings

AndreV84, 2020-07-15:
how to improve the situation?
`FROM nvcr.io/nvidia/l4t-base:r32.4.2
RUN apt-get update -y
RUN apt install -y wget software-properties-common apt-utils curl git
RUN /bin/bash -c 'echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list'
RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key
RUN apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-melodic-desktop clang-format ros-melodic-desktop-full python-pip python3-pip python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential
RUN apt update -y
#RUN rm -rf /var/lib/apt/lists
RUN apt-get install -y librealsense2-dev librealsense2-utils
RUN apt install -y ros-melodic-ddynamic-reconfigure
RUN dpkg-reconfigure --frontend noninteractive tzdata
RUN rm -rf /var/lib/apt/lists
WORKDIR /home/nvidia/ws
RUN mkdir src && cd src && git clone https://github.com/intel-ros/realsense.git -b development
RUN rosdep init && rosdep update

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash; cd /home/nvidia/ws; rosdep install --from-paths src --ignore-src; catkin_make'``

Any idea what might be wrong, considering everything is installed from apt & firmware is the latest?

AndreV84, 2020-07-16:
trying again with
```
FROM nvcr.io/nvidia/l4t-base:r32.4.2
RUN apt-get update -y
RUN apt install -y wget software-properties-common apt-utils  curl git
RUN /bin/bash  -c 'echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list'
RUN  apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key
RUN apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-melodic-desktop clang-format ros-melodic-desktop-full python-pip python3-pip  python-rosdep python-rosinstall python$
RUN apt update -y
#RUN rm -rf /var/lib/apt/lists
RUN  apt-get install -y librealsense2-dev librealsense2-utils
RUN apt install -y ros-melodic-ddynamic-reconfigure
RUN dpkg-reconfigure --frontend noninteractive tzdata
RUN rm -rf /var/lib/apt/lists
WORKDIR /home/nvidia/ws
RUN mkdir src && cd src &&  git clone https://github.com/intel-ros/realsense.git -b development
#  && git clone https://github.com/slamtec/rplidar_ros

RUN rosdep init && rosdep update

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash;  cd /home/nvidia/ws; rosdep install --from-paths src --ignore-src; catkin_make'
```

AndreV84, 2020-07-21:
updated Docker file; still same errors; librealsense sdk seems to work; but the wrapper build fails
FROM osrf/ros2:nightly-rmw
ENV RTI_NC_LICENSE_ACCEPTED yes

# bootstrap rosdep
RUN rosdep update

# install dependencies
RUN . /opt/ros/$ROS_DISTRO/setup.sh \
    && apt-get update \
    && rosdep install -y \
    --from-paths /opt/ros/$ROS_DISTRO/share \
    --ignore-src \
    --skip-keys " \
      " \
    && rm -rf /var/lib/apt/lists/*

# set up environment
ENV NDDSHOME /opt/rti.com/rti_connext_dds-5.3.1
ENV PATH "$NDDSHOME/bin":$PATH
ENV LD_LIBRARY_PATH "$NDDSHOME/lib/x64Linux3gcc5.4.0":$LD_LIBRARY_PATH
# bootstrap
#RUN  apt-get install -y ros-foxy-librealsense2

RUN apt-get update -y
RUN apt install -y wget curl git
RUN apt-get update -y
#RUN locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 && LANG=en_US.UTF-8
RUN apt update && apt install curl gnupg2 lsb-release
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc |  apt-key add - && sh -c 'echo "deb http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2>
RUN apt update -y
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-foxy-ros-base
RUN apt update -y
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-foxy-cv-bridge  ros-foxy-message-filters ros-foxy-image-transport
RUN DEBIAN_FRONTEND="noninteractive" apt install -y  libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
#RUN DEBIAN_FRONTEND="noninteractive" apt install -y  ros-foxy-realsense-camera-msgs ros-foxy-realsense-ros2-camera
RUN dpkg-reconfigure --frontend noninteractive tzdata
#RUN rm -rf /var/lib/apt/lists
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-foxy-desktop
WORKDIR /home/nvidia/ros2_ws
RUN  mkdir -p ~/src && cd ~/src && git clone https://github.com/intel/ros2_intel_realsense.git && cd ros2_intel_realsense && git checkout refactor
RUN  sh -c 'echo "deb [arch=amd64,arm64] http://repo.ros2.org/ubuntu/main `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc |  apt-key add -
RUN apt-get update -y
#RUN  apt-get install -y ros-foxy-librealsense2
RUN git clone https://github.com/IntelRealSense/librealsense.git && cd librealsense && mkdir build && cd build &&  cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=re>
RUN apt install -y python3-colcon-common-extensions
#RUN cd ~ &&  /bin/bash -c '. /opt/ros/foxy/setup.bash; colcon build --symlink-install --base-paths src/ros2_intel_realsense'

RealSenseSupport, 2020-08-20:
Hi @AndreV84 

Are you still having trouble with this on latest? We are able to install and run the SDK and ROS on Jetson w/ an L515 successfully.

On Jetson we'd recommend starting clean...:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

...installing the SDK first using the instructions here...:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

...and then building realsense-ros around the above installation in a catkin workspace using the instructions here:

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

Thanks

AndreV84, 2020-08-20:
Thank you for following up!
I believe that my last efforts to get realsense sdk to work on ROS2 failed, but on ROS 1 succeeded.
Would you be able to provide support for ROS2 implementation, please?

AndreV84, 2020-08-20:
When you are stating  "We are able to install and run the SDK and ROS on Jetson w/ an L515 successfully." could you clarify if you are able to implement it on ROS1? ROS2? both? Neither of the two?

RealSenseSupport, 2020-08-20:
We currently only (officially) support melodic and kinetic.

doronhi, 2020-08-26:
Work is being done on the ROS2 implementation. We don't have a date yet but we expect a working version sometimes around october 2020.

RealSenseSupport, 2021-03-24:
Hi,

Will you be needing further help with this? ROS2 wrapper is currently avail @ https://github.com/IntelRealSense/realsense-ros/releases

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

closed 2021-03-24 by AndreV84