realsenseai/realsense-ros #1291: roslaunch rs_rgbd.launch not working

hongluck, 2020-07-16:
Hello guys I have an issue with rs_rgbd.launch

For realsense-viewer and other roslaunch such as rs_camera.launch, roslaunch realsense2_camra rs_camera filters:=pointcloud etc

the node is properly working but when if comes to rs_rgbd.launch it gives a strange message

roslaunch realsense2_camera rs_rgbd.launch 
... logging to /home/hong/.ros/log/7994cfbc-c75c-11ea-bc8c-b025aa2e8947/roslaunch-hong-ubuntu-pc-16256.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 655, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 589, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 207, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: rgbd_launch
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/hong/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share

I need PointCloud2 message please help.

Just reinstalled realsense package today and having issues with specific node

MartyG-RealSense, 2020-07-16:
Hi @hongluck  Can you tell me please the information below:

-  Which RealSense camera you are using

-  Which computer or computing device you are using (PC, Jetson, Raspberry Pi, etc).

-  Which operating system are you using (Windows, Linux, Raspbian, etc)

Thanks!

hongluck, 2020-07-16:
> Hi @hongluck Can you tell me please the information below:
> 
> * Which RealSense camera you are using 
> * Which computer or computing device you are using (PC, Jetson, Raspberry Pi, etc).
> * Which operating system are you using (Windows, Linux, Raspbian, etc)
> 
> Thanks!

I am using D435 camera, PC, Linux Ubuntu thank you

hongluck, 2020-07-16:
resolved with 

sudo apt install ros-kinetic-rgbd-launch

closed 2020-07-16 by hongluck

MartyG-RealSense, 2020-07-16:
Great news - I'm very pleased that you found a solution.  Thanks so much for the update!