realsenseai/realsense-ros #1292: roslaunch realsense2_camera  not working

taegyuahn, 2020-07-17:
I use ubuntu 18.04 LTS on PC, ROS melodic and my device is LIDAR L515.

I followed these installation guide 

(Method2) https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

and my realsense-ros version is 2.2.15 , librealsense 2.36.0
`realsense-viewer` is working but if I launch `roslaunch realsense2_camera rs_camera.launch` 
the terminal shows like this.
[image]

I tried to install realsense-ros version 2.2.14 and reinstalled ros melodic 
but roslauch doesn't works.
Anyone can help me?

taegyuahn, 2020-07-17:
I solved this issue after installing libuvc in this page https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
but another issue happened...

[image]

doronhi, 2020-07-19:
I could not reproduce. I am very sorry as I believe it reflects a real issue.
The message of "control_transfer returned error" does not in itself implies a problem. It it known to appear a few times at the beginning of the node's run, as is the case that you show here, but unless it appears rapidly and constantly it seems to be dealt with by the driver. It is disturbing and is still under investigation

manu-adina, 2020-07-22:
> I solved this issue after installing libuvc in this page https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
> but another issue happened...
> 
> [image]

I'm facing the exact same issue. I'm using VMware with Ubuntu 18.04, the device is working in the RealSense viewer just fine, but nothing on Rviz. I'm getting the same succession of warnings as shown in the screenshot as well.

RealSenseSupport, 2020-09-11:
Hi @taegyuahn 

Is there anything else needed on this?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-18 by RealSenseSupport