realsenseai/realsense-ros #1293: l515 fails to start

ramilmsh, 2020-07-17:
I've built libreasense (f7cdf6e8961e1709f6d864bdb33095c00a671ca7), realsense_ros (a25ea06f3006753f5ce428822d0fac43500a5908) and ddynamic_reconfigure (0481332a3e4dceff458c202657f64c6b44224eb5) with catkin build on xenial

jmiseikis, 2020-07-17:
You can use the following launch file to start it. The main thing is to set correct resolutions for depth and color cams in the launch file.

`<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="l515"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="1280"/>
  <arg name="fisheye_height"      default="720"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="1024"/>
  <arg name="depth_height"        default="768"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="1280"/>
  <arg name="infra_height"       default="720"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="1920"/>
  <arg name="color_height"        default="1080"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>`

jmiseikis, 2020-07-17:
You can use the following launch file to start it. The main thing is to set correct resolutions for depth and color cams in the launch file.
```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="l515"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="1280"/>
  <arg name="fisheye_height"      default="720"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="1024"/>
  <arg name="depth_height"        default="768"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="1280"/>
  <arg name="infra_height"       default="720"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="1920"/>
  <arg name="color_height"        default="1080"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>
```

nkhedekar, 2020-07-20:
Just as an additional point of reference, have a look at #1246

RealSenseSupport, 2020-07-24:
Hi @ramilmsh - Were you able to get things going?

*If we don’t hear from you in 7 days, this issue will be closed.

ramilmsh, 2020-07-26:
@RealSenseCustomerSupport no, I have not. I keep getting the same error. I will be happy to provide any additional information about my set up

doronhi, 2020-07-27:
If the camera is connected as a usb2.0 device it may explain the error. Can you make sure you are connected as a USB3 device? Check with realsense-viewer if you can.

ramilmsh, 2020-07-28:
@doronhi yeah, it's working over usb 3.1, according to realsense viewer

randomblbl, 2020-07-28:
I have the same problem, L515 is connected as USB3.2 device. Is there any solution?

doterkuile, 2020-07-29:
I also encountered the same problem and I think it may be caused by some new commits in librealsense ( [81cffa8](https://github.com/IntelRealSense/librealsense/commit/81cffa839148fa620ebace13292ebc54c4cf5c6c) & [10e402a](https://github.com/IntelRealSense/librealsense/commit/10e402a09d62f3368af2ba289887e0b47828c220) ). In this commit the  enum RS2_OPTION_SENSOR_MODE gets an extra option QVGA. I am not sure how but somewhere the enum options get mixed up and i think this causes the error in the librealsense. I found a fix by changing line 1258 of [model-views.cpp](https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp) from

`                        if (res >= RS2_SENSOR_MODE_VGA && res < RS2_SENSOR_MODE_COUNT)
`
to:

`                        if (res >= RS2_SENSOR_MODE_XGA && res < RS2_SENSOR_MODE_COUNT)
`

By doing this it  *rs_camera.launch* worked for me although I cannot explain exactly why and it is only a quick fix, but maybe this helps figuring out the exact issue.

randomblbl, 2020-07-29:
> I also encountered the same problem and I think it may be caused by some new commits in librealsense ( [...]

It's not working. The enum is defined as follow. It does not make sense change RS2_SENSOR_MODE_VGA to RS2_SENSOR_MODE_XGA in  line 1258 of [model-views.cpp](https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp)

  typedef enum rs2_sensor_mode
    {
        RS2_SENSOR_MODE_VGA,
        RS2_SENSOR_MODE_XGA,
        RS2_SENSOR_MODE_QVGA,
        RS2_SENSOR_MODE_COUNT /**< Number of enumeration values. Not a valid input: intended to be used in for-loops. */
    } rs2_sensor_mode;

doterkuile, 2020-07-30:
Ah yes, sorry I forgot to mention to change the order of the enum as well:

typedef enum rs2_sensor_mode
{
RS2_SENSOR_MODE_QVGA,
RS2_SENSOR_MODE_VGA,
RS2_SENSOR_MODE_XGA,
RS2_SENSOR_MODE_COUNT /**< Number of enumeration values. Not a valid input: intended to be used in for-loops. */
} rs2_sensor_mode;

randomblbl, 2020-07-30:
> Ah yes, sorry I forgot to mention to change the order of the enum as well:
>
> typedef enum rs2_sensor_mode
> {
> RS2_SENSOR_MODE_QVGA,
> RS2_SENSOR_MODE_VGA,
> RS2_SENSOR_MODE_XGA,
> RS2_SENSOR_MODE_COUNT /**< [...]

It works for me. Thanks for sharing the solution.

To summarize @doronhi ' s solution:
- In "librealsense/include/librealsense2/h/rs_option.h", change definition of  rs2_sensor_mode (starting line 151) to 
typedef enum rs2_sensor_mode
    {
        RS2_SENSOR_MODE_QVGA,
        RS2_SENSOR_MODE_VGA,
        RS2_SENSOR_MODE_XGA,
        RS2_SENSOR_MODE_COUNT /**< Number of enumeration values. Not a valid input: intended to be used in for-loops. */
    } rs2_sensor_mode;

- In "librealsense/common/model-views.cpp", change line 1258, from 
       if (res >= RS2_SENSOR_MODE_**VGA** && res < RS2_SENSOR_MODE_COUNT)
  to 
       if (res >= RS2_SENSOR_MODE_**XGA** && res < RS2_SENSOR_MODE_COUNT)

Then, in "librealsense/build" folder run 
 - make && sudo make install

The problem will be resolved.

ramilmsh, 2020-08-03:
@doronhi I will change it locally for testing purposes, but, would you mind making a PR?

doronhi, 2020-08-03:
First of all, thanks for the credit but it was not my solution. Thank @doterkuile and @randomblbl for their work and investigation :)
The problem I am having is that I cannot reproduce this with a USB3.2 connection.
What firmware is installed on your L515? You mentioned Ubuntu16.04 but on what machine and what kernel (`uname -r`)?

Nir-Az, 2020-08-03:
Hi all,
I am not sure how this flow raised this exception but it seems to be related to QVGA resolution like mentioned above.
QVGA resolution was added for future use and it is not supported currently in libRealSense (The reason it was added as the last Enum value before count value was for not breaking the API).

**a relevant bug was fixed recently (for another reason but fixed a lot of issues with the RS Enums) and I believe it is the root of the issue.** 
Please see [c97f43fe](https://github.com/IntelRealSense/librealsense/commit/c97f43fee977d254e001000adac3a8d2a35373d1)

This fix will be included in the next release.

If you can try undoing the changes you made and apply the commit change I attached it will help us understand if the issue is fixed or something else is hiding over there.

Thanks.

blakely3dnm, 2020-08-04:
Will the QVGA option increase the confidence at a given range?

doterkuile, 2020-08-04:
@doronhi 

I use Ubuntu 18.04 on a desktop with kernel 4.15.0-72-generic. I have updated the firmware of my l515 to the latest version 01.04.01.02 using it with a usb3.2 connection. Hope this helps!

closed 2020-12-31 by RealSenseSupport