realsenseai/realsense-ros #1294: L515 Short-range mode on ROS

jmiseikis, 2020-07-17:
Hi, I found short range mode for L515 using realsense-viewer very useful for object detection in 20-50 cm range from the camera. Does anyone know how I can the short-range mode in ROS node?

Thank you in advance!

nkhedekar, 2020-07-20:
You can open the dynamic reconfigure and select the mode in the drop down menu

RealSenseSupport, 2020-07-24:
Hi @jmiseikis - were you able to see the mode?

*If we don’t hear from you in 7 days, this issue will be closed.

jmiseikis, 2020-07-27:
@nkhedekar Thank you!

@RealSenseSupport problem solved, please feel free to close the issue.

closed 2020-07-30 by RealSenseSupport

Drift-Of-Little-Forest, 2022-10-31:
我通过python代码启动深度相机，出现的深度画面是远程模式(Max range),可是我的项目需要使用相机的短程模式，如何在python中通过代码来设置L515的短程模式（short range）,有实例代码吗？请帮助我