realsenseai/realsense-ros #1298: L515 Portrait Resolution (480 x 640) Pointcloud is Rotated and Mirrored

pauldeee, 2020-07-21:
Hi,

The L515 is listed as supporting several depth resolutions:
`Depth	  1024x768	@ 30Hz	   Z16
Depth	  768x1024	@ 30Hz	   Z16
Depth	  640x480	        @ 30Hz	   Z16
Depth	  480x640	        @ 30Hz	   Z16`

When I attempt to use the "portrait" resolution 480 x 640 the point cloud appears to be rotated 90 degrees and mirrored. Is this the intended behavior?

In my launch file I specify:
`<arg name="depth_width" value="480"/>
<arg name="depth_height" value="640"/>`

Might there be a different way to use the portrait resolustions that I am unaware of?

Regards

blakely3dnm, 2020-07-23:
I see the same thing in realsense-viewer so I wonder if this is a firmware issue, as there is no way to unmirror a pointcloud by rotation or translation (without doing things in the 4th dimension)

pauldeee, 2020-08-03:
The gentlest of bumps.

RealSenseSupport, 2020-08-17:
Hi,

Yes this is expected behavior. Incidentally the plan is to deprecate these "flipped" resolutions so we'd recommend using the standard (landscape) ones .

Thanks

blakely3dnm, 2020-08-18:
I was thinking it would be nice to have this setting to change from portrait to landscape orientations to better suit the environment as a robot is moving through a scene, without the need for actually rotating the sensor. I can see how perhaps the mirroring could be corrected downstream of realsense-ros but it would be unfortunate if the option to change orientation is done away with completely.

closed 2020-09-26 by RealSenseSupport