realsenseai/realsense-ros #1299: How to modify this package to controll SR305 sensor's option?

weseu, 2020-07-22:
Hi. 
I want to controll the auto exposure on/off at different enviroment. so I need to add funtion in this package.

Ps. I try to create a new code to controll sensor's option. But, I will not be able to open rs-camera.launch,because I need to open device it will conflict.

I try to add function in "realsense_node_factory.cpp" and "base_realsense_node.cpp", but it just run once.

and then I try to know how did you let it always publisher, but I didn't finish it.....please give me some help..

a. I find "nodelet.launch.xml",it use nodelet.....But I don't know "load realsense2_camera/RealSenseNodeFactory $(arg manager)" "which mean what??
[image]

MartyG-RealSense, 2020-07-22:
The link below provides code for turning auto exposure on / off with a **dynamic_reconfigure** rosrun instruction.  Is that of any help to you, please?

https://github.com/IntelRealSense/realsense-ros/issues/318#issuecomment-377218773

MartyG-RealSense, 2020-07-22:
There is a means for the 400 Series cameras of turning auto-exposure on / off with a **dynamic reconfigure** rosrun instruction during runtime.

https://github.com/IntelRealSense/realsense-ros/issues/318#issuecomment-377218773

For the SR305 it may not be as straightforward, as you have discovered.  Even in the earlier **realsense_camera** node - http://wiki.ros.org/realsense_camera -  for the original SR300 camera, people were discussing how to control exposure.  A RealSense user ended up creating their own PR to add dynamic reconfigure control of exposure for the SR300.

https://github.com/IntelRealSense/realsense-ros/pull/214

https://github.com/IntelRealSense/realsense-ros/pull/214/commits/a43218d05f9315ca284d9d8a95647feff3f98d07

So the ideal situation would be to develop a means if possible to use dynamic_reconfigure in realsense2_camera to control the SR305's on-off state during runtime, like the solution above for the earlier realsense_camera.

On the subject of **RealSenseNodeFactory**, I believe that it refers to the mechanism for launching the **realsense2_camera** node.

closed 2020-08-08 by MartyG-RealSense