realsenseai/realsense-ros #1300: Intel D4XX controlling the Laser  Power problem

dingbaiyin, 2020-07-22:
The main purpose is to adjust the power of the laster to the maximum before each picture is collected, and adjust the power to 0 after acquisition. After changing the control of Laster power, the program error "terminate called after throwing an instance of'rs2::error' was encountered when calling the controlling Laster power program
   what(): null pointer passed for argument "device"
It may be that there is a problem with my program, but I don't know how to eliminate it. Below is my relevant code pictures, please help me analyze where the problem is。
[image]
[image]
[image]
[image]

dingbaiyin, 2020-07-22:
The  error is reported when the program starts to call the ControlLasterPowerTo_Max() function or ControlLasterPowerTo_Min().Program error message  as the picture shows,
[image]

MartyG-RealSense, 2020-07-22:
Hi @dingbaiyin   If I were writing such a program myself then I would centralise all the camera configuration processes such as the **ctx** and **device** instructions at the top of one script.  In your current setup, you define the configuration once at the top of **cam_process.cpp** and then repeat the configuration instructions again in **ControlLasterPowerTo_Min** and  **ControlLasterPowerTo_Max**.  

Camera config instructions tend to be defined **before** the pipeline starts.  The program may not be agree with these setup instructions being called once the pipeline has started.

closed 2020-08-08 by MartyG-RealSense