realsenseai/realsense-ros #1301: Realsense camera launch fails after some time

akashatharv, 2020-07-22:
Hi

I am using a realsense d435 camera on a Raspberry Pi4 with Raspbian buster 10 and ROS melodic. I am running the launch file rs_rgbd.launch for getting and visualizing the data, however after all the nodes have ran for about 30-45 seconds the terminal with the launch file stops working and my Pi becomes unresponsive after that.  Is there a reason why that's happening ?

MartyG-RealSense, 2020-07-22:
Hi @akashatharv   Can you provide details of any error messages that you have in the ROS log, please?   Thanks!

MartyG-RealSense, 2020-07-22:
If the **control_transfer returned** error is continuously generating then it indicates a communication problem with the device. 

https://github.com/IntelRealSense/realsense-ros/issues/1089#issuecomment-597074650

Do the messages start appearing in the log immediately after launch or 30 to 45 seconds into the streaming session, please?

akashatharv, 2020-07-22:
They only appear at the beginning

MartyG-RealSense, 2020-07-22:
Okay, thank you.   So it is the red errors that occur when streaming fails?

akashatharv, 2020-07-22:
Yes

MartyG-RealSense, 2020-07-22:
Which Librealsense version are you using, please?

akashatharv, 2020-07-22:
I am using v2.34.0
Can you please also tell me what these errors mean 
ERROR [2499032080] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 131
 ERROR [2563761168] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

MartyG-RealSense, 2020-07-22:
I guessed that you might have been using SDK 2.34.0, as 2.34.0 was particularly affected by the **get_device_time_ms()** error, both in Librealsense and the RealSense ROS wrapper.   Would it be possible to upgrade to a newer Librealsense and matching ROS wrapper version?

akashatharv, 2020-07-22:
Okay let me try that. I will update the issue accordingly. Thank you for your help. Have a nice day!!

MartyG-RealSense, 2020-07-22:
I look forward to your update.   Good luck!   :)

akashatharv, 2020-07-22:
I updated both of them and now I am not getting any errors or warnings and the launch file is running all the nodes reliably for now. Thank you for your help!!

closed 2020-07-22 by akashatharv

MartyG-RealSense, 2020-07-22:
I'm very pleased that the recommendation worked for you - thanks so much for the update!