realsenseai/realsense-ros #1302: [Question] Fade colors

RGring, 2020-07-23:
Yesterday, I went out in the field and used the camera actively. Looking at the data, the RGB image doesn't look very satisfying. The colors are very fade and the image doesn't look as sharp as I have hoped for. Is there anything i can tune to improve the colors and sharpness? The object distance is in a range of 1.0 - 1.5 m.

My setup:
Jetson Xavier NX, L4T 32.4.3
Stereo camera Realsense D435i (connected to USB3.0)
ros wrapper 2.2.14
librealsense 2.35.2
Firmware 5.12.6.0

Example image Realsense:
[image]
Image from different camera, same object:
[image]

I appreciate any help and thanks in advance!

MartyG-RealSense, 2020-07-23:
This looks like it can be improved with the **Auto White Balance** RGB setting.  You can test this easily in the RealSense Viewer.

Under the **Controls** section of the RGB options in the Viewer side-panel, disable the **Auto White Balance** option.  Then reduce the slider from the center position to fade the colors and increase from the center position to strengthen the colors so that they look more like the bright green grass image.

RGring, 2020-07-28:
Thanks for the reply. I have more issues regarding this use case.
1. The camera can be somewhere at 0.7-1.2m above ground and it seems that objects < 10mm are not present in the depth map, because the noise is in the same range.
2. The RGB image quality is quite blurry, resulting in the loss of relevant leave feature. The blurr increases towards the edge of the image.

I tried tweaking the parameters, but without remarkable success. Though, maybe I missed something relevant for the use case: "small objects at fixed distance of 0.8m". 
I appreciate any hints!

MartyG-RealSense, 2020-07-28:
1.  Aligning depth to color can help to separate foreground detail from background detail.  

2.  Increasing the RGB stream's FPS speed to 60 FPS may help to reduce RGB image blur if you are currently using 30 FPS.  This is because the RGB has a 'rolling' shutter that is slower than the depth's 'global' shutter.  On the new **RealSense D455**camera model, both RGB and depth have a fast global shutter.

MartyG-RealSense, 2020-08-08:
Hi @RGring   Do you require further assistance with this case please, or can it be closed?   Thanks!