realsenseai/realsense-ros #1303: L515 Pointcloud is black in Rviz, trying to view it in colour with

manu-adina, 2020-07-24:
I'm unable to view a coloured point cloud.

[image]

It appears to be black despite having it set to RGB8. I'm able to view an aligned coloured pointcloud in the realsense-viewer just fine but not in Rviz. I'm running ros the following way,

`roslaunch realsense2_camera rs_515.launch filters:=pointcloud`

my rs_515.launch is a copy of the default rs_camera with different resolution.

Here are the logs when I launch the realsense-ros 

[ INFO] [1595558241.193088050]: Initializing nodelet with 2 worker threads.                                                                                                                                 
[ INFO] [1595558241.377167540]: RealSense ROS v2.2.14                                                                                                                                                       
[ INFO] [1595558241.377234884]: Running with LibRealSense v2.35.2                                                                                                                                           
[ INFO] [1595558241.386038685]:                                                                                                                                                                             
[ INFO] [1595558241.927828998]: Device with serial number 00000000f0172048 was found.                                                                                                                       
                                                                                                                                                                                                            
[ INFO] [1595558241.927959340]: Device with physical ID 4-1-20 was found.                                                                                                                                   
[ INFO] [1595558241.927992603]: Device with name Intel RealSense L515 was found.                                                                                                                            
[ INFO] [1595558241.929076108]: Device with port number 4-1 was found.                                                                                                                                      
[ INFO] [1595558241.931826148]: getParameters...                                                                                                                                                            
[ INFO] [1595558241.985458680]: setupDevice...                                                                                                                                                              
[ INFO] [1595558241.985667345]: JSON file is not provided                                                                                                                                                   
[ INFO] [1595558241.985695810]: ROS Node Namespace: l515                                                                                                                                                    
[ INFO] [1595558241.985720906]: Device Name: Intel RealSense L515                                                                                                                                           
[ INFO] [1595558241.985731075]: Device Serial No: 00000000f0172048                                                                                                                                          
[ INFO] [1595558241.985738206]: Device physical port: 4-1-20                                                                                                                                                
[ INFO] [1595558241.985745094]: Device FW version: 01.04.01.02                                                                                                                                              
[ INFO] [1595558241.985751455]: Device Product ID: 0x0B64                                                                                                                                                   
[ INFO] [1595558241.985783040]: Enable PointCloud: On                                                                                                                                                       
[ INFO] [1595558241.985792898]: Align Depth: On                                                                                                                                                             
[ INFO] [1595558241.985799303]: Sync Mode: On                                                                                                                                                               
[ INFO] [1595558241.985838445]: Device Sensors:                                                                                                                                                             
[ INFO] [1595558242.002706613]: L500 Depth Sensor was found.                                                                                                                                                
[ INFO] [1595558242.005217920]: RGB Camera was found.
[ INFO] [1595558242.005330072]: Motion Module was found.
[ INFO] [1595558242.005393164]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1595558242.005418299]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1595558242.005427079]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1595558242.005447532]: Add Filter: pointcloud
[ INFO] [1595558242.005891954]: num_filters: 1
[ INFO] [1595558242.005922109]: Setting Dynamic reconfig parameters.
[ INFO] [1595558242.313090457]: Done Setting Dynamic reconfig parameters.
[ INFO] [1595558242.315139749]: depth stream is enabled - width: 1024, height: 768, fps: 30, Format: Z16
[ INFO] [1595558242.315718815]: infra stream is enabled - width: 1024, height: 768, fps: 30, Format: Y8
[ INFO] [1595558242.318213469]: color stream is enabled - width: 1920, height: 1080, fps: 30, Format: RGB8
[ INFO] [1595558242.322156619]: setupPublishers...
[ INFO] [1595558242.325553053]: Expected frequency for depth = 30.00000
[ INFO] [1595558242.358672315]: Expected frequency for infra = 30.00000
[ INFO] [1595558242.375907032]: Expected frequency for aligned_depth_to_infra = 30.00000
[ INFO] [1595558242.390715207]: Expected frequency for color = 30.00000
[ INFO] [1595558242.404948569]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1595558242.421344911]: setupStreams...
 23/07 19:37:22,434 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1595558242.495654860]: insert Depth to L500 Depth Sensor
[ INFO] [1595558242.495899779]: insert Color to RGB Camera
[ INFO] [1595558242.495963709]: insert Infrared to L500 Depth Sensor
 23/07 19:37:22,603 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1595558242.727735932]: SELECTED BASE:Depth, 0
 23/07 19:37:22,728 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/07 19:37:22,779 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1595558242.830793331]: RealSense Node Is Up!
 23/07 19:37:22,931 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/07 19:37:22,982 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/07 19:37:23,033 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/07 19:37:23,084 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/07 19:37:23,134 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/07 19:37:23,186 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/07 19:37:23,236 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/07 19:37:23,287 WARNING [139643051820800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

I'm using VMware with Ubuntu 18.04, ROS melodic, with the latest realsense-ros and firmware

RealSenseSupport, 2020-07-24:
Hi @manu-adina 

We are looking into this but unable to reproduce. We have a few questions:

-You mentioned that you're using a copy of rs_camera launch file with just the resolution changed...but from log you shared I can see other changes as well. Can you please share the exact launch file you are using to help in potentially reproducing this.

-You mentioned that you're using the latest version of realsense-ros....but it doesn't appear so from the log file. The latest realsense-ros is 2.2.15 and the latest librealsense is 2.36.0. Can you try updating to those versions and see if you're still experiencing this?

-Also I'm a little concerned about the control_transfer errors you're seeing. What platform are you running this on? X64, ARM, etc? And is the device plugged into a USB3/2. Some details regarding the host and usb etc would be appreciated.

-Also can you provide the firmware version that's installed on your L515?

Thanks

wk738126046, 2020-07-25:
sorry ,I suffer from the same error. the system version that realsense-ros is 2.2.15 and the librealsense is 2.36.0.
first, rs-enumerate-devices is normal. then, realsense-viewer is normal. but I run rs_rgbd.launch and display same error.
is it not 640*480 on ROS?????? when I run rs-color ,it appears "The connected device doesn't support color streaming!".
so how to fix it ???????? thanks ~~~

manu-adina, 2020-07-27:
Thanks for you response @RealSenseSupport,

1) Someone posted a launch file for L515 that was suppose to work, I've used that launch file. It still wasn't working for me, but with changed resolution it did seem to work. The launch file is below,

```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="l515"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="1280"/>
  <arg name="fisheye_height"      default="720"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="1024"/>
  <arg name="depth_height"        default="768"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="1024"/>
  <arg name="infra_height"       default="768"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="1920"/>
  <arg name="color_height"        default="1080"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="200"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>
[... 30 lines cut ...]
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>
```

2) I downloaded realsense-ros through a distribution manager, so it is behind the source version for sure. I can try installing the latest source version to see if it does anything. 

3) I'm using USB 3.2 port, with x86_64 platform. The host is a Windows machine that's running Ubuntu 18.04 through VMware. 

4) Firmware version on L515 is 01.04.01.02.  The librealsense is 2.36 which I also got through apt-get. It is weird that it says in logs that `Running with LibRealSense v2.35.2` when in fact I shouldn't be. The realsense-viewer says that it's v2.36 and when I try apt-get it says `librealsense2-dev is already the newest version (2.36.0-0~realsense0.3169)`.

manu-adina, 2020-07-27:
> sorry ,I suffer from the same error. the system version is ubuntu16 and realsense-ros is 2.2.15 and the librealsense is 2.36.0.
> first, rs-enumerate-devices is normal. then, realsense-viewer is normal. but I run rs_rgbd.launch and display same error.
> is it not 640*480 on ROS?????? when I run rs-color ,it appears "The connected device doesn't support color streaming!".
> so how to fix it ???????? thanks ~~~
> #1271

@wk738126046 I'm not too sure if you're suppose to use RGBD launch file (rs_rgbd.launch). From my understanding RGBD is for getting depth through the use of multiple cameras. The other RealSense products use multiple cameras to obtain depth, however this product L515 is using Lidar to do it.

doronhi, 2020-07-27:
@manu-adina ,
The log show the version of librealsense2 that realsense2_camera was compiled with. This is fixed in version 2.2.15.
As to the black pointcloud, can you confirm that the color image is indeed published? Can you add it to an "Image" window on RViz?

manu-adina, 2020-07-29:
@doronhi 
The image is posting just fine and I'm able to see it in the window as shown,
[image]
I'll try to install realsense-ros from source, maybe that'll help

kristfrizh, 2020-08-12:
I have also having these problem with D435 

anyone has fixing it?

I found some notes said that in the launch files we need to change the pointcloud value to true but it still black like that

doronhi, 2020-08-12:
Could the solution in #1327 work here? 

> ... the style should be changed to points

manu-adina, 2020-08-14:
@doronhi 
I tried changing the style from the default squares to points, and it doesn't seem to do anything for me. When I switch to points, nothing is displayed in Rviz. I don't get any errors in either the Rviz or realsense-ros side when I do it.

BenjaminMichaelGreen, 2020-10-24:
I had the same issue. Not only were the points black and white, but the RVIZ controls for the point style and size were completely broken. This problem was resolved for me by adding "export SVGA_VGPU10=0" to my bashrc. If you want to try it, run the following commands and try again.
```console
echo "export SVGA_VGPU10=0" >> ~/.bashrc
source ~/.bashrc
```
Hope it works!

manu-adina, 2020-10-26:
> I had the same issue. Not only were the points black and white, but the RVIZ controls for the point style and size were completely broken. This problem was resolved for me by adding "export SVGA_VGPU10=0" to my bashrc. If you want to try it, run the following commands and try again.
> 
> ```
> echo "export SVGA_VGPU10=0" >> ~/.bashrc
> source ~/.bashrc
> ```
> 
> Hope it works!

@BenjaminMichaelGreen Thanks, that solves the issue!

closed 2020-10-26 by manu-adina