realsenseai/realsense-ros #1304: About algorithm

intel-user, 2020-07-25:
Hi,
Can you tell me know what's algorithm this project used?
Thank you for anyhelp.

MartyG-RealSense, 2020-07-25:
Could you provide more information about what you need to know, please?

The RealSense ROS wrapper provides the RealSense SDK (librealsense) with compatibility with ROS through a 'wrapper' ('realsense2_camera').   It requires librealsense to be installed for it to work.

intel-user, 2020-07-25:
Hi,
I want to know which algorithm this project used? for example ORB-SLAM, online SLAM, full SLAM v.v.
Thank you.

MartyG-RealSense, 2020-07-25:
The ROS wrapper does not have a built-in SLAM algorithm.  It can be used with ROS navigation programs such as RTABMAP_ROS and ORB-SLAM2 though.

For example, the link below has an official Intel tutorial for using a RealSense D435i with RTABMAP_ROS.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

intel-user, 2020-07-26:
Hi, I have a question.
What's purpose of the IR projector in the depth algorithm and how does it works?
Thank you so much.

MartyG-RealSense, 2020-07-26:
The projector projects a semi-random pattern of dots onto the scene that the camera is observing.  This pattern can be used by the camera to analyse the surfaces that the dots are on for depth detail.  It uses the dots as a "texture source" on surfaces that have little or no texture (especially flat, low-detail surfaces such as walls, doors, tables, chairs, etc).

closed 2020-08-08 by MartyG-RealSense